raspberrypi/linux #4663: brcmfmac: getting error messages instead of info

alexander-kurichenko, 2021-11-01:
Hello,
First of all, thank you for making a good product!

Here's my issue:
if you run dmesg, you can get following error messages:
```
root@raspberrypi:~# dmesg -l err
[    8.971676] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.221870] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.229584] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.244068] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
(skipped)
````
Expected behaviour: these messages should have "notice" or "info" loglevel,  not "error". I don't see any errors there. Everything looks good.

Affected HW: Raspberry Pi 4B
Why it's important? When people troubleshoot different kinds of issues, they try to gather all errors in logs, but these messages will confuse newcomers because they think it might be an error.

Is it possible to change loglevel for similar messages?
Thank you in advance

pelwell, 2021-11-02:
It's an unwanted side effect of enabling the BRCMDBG config setting  (which we want to enable) - it also redefines the brcmf_info logging function to be brcmf_err. This behaviour is unchanged from the upstream kernel, and so far it hasn't annoyed me sufficiently to want to change it. However, I'll take comments here as an informal vote if people want `info` messages to remain at `info` level.

pelwell, 2021-11-02:
No, it's no good - now you've pointed it out it is annoying.

See 182c986a862f.

```
pi@raspberrypi:~$ dmesg -l err
pi@raspberrypi:~$
```
Much better.

alexander-kurichenko, 2021-11-02:
Now everything looks perfect!
@pelwell, thank you for help.
Closing issue

closed 2021-11-02 by alexander-kurichenko