raspberrypi/linux #4665: Kernel Build from source (Cross Compiling) - results in Kernel Panic

WanaGo, 2021-11-03:
Hello,

We have been trying to rebuild the Kernel for the Raspberry Pi's (say Pi4 for arguments sake), and every time we package the Kernel and then unpackage it on a PI (ill explain soon), as soon as it is unpackaged no commands work anymore, ie 'ls' or any basic command like that, will not work. Rebooting the Pi results in a Kernel panic.

We have narrowed it down to not actually be the kernel img itself, but the modules, but we are at the end of our knowledge and could really use some help to figuring out what is going on. We tested this by coping the files from the /boot/ folder to the Pi, leaving the modules on the ext4 partition completely unchanged. The OS booted, but things didnt work perfectly as you might expect as the modules didnt match, but it booted, showing it wasnt the kernel img file as such causing the issue. We think at least. Correct me if wrong.

So this is a cut down version of what we were doing, as we were adding in our own files, and using Menuconfig to enable the modules etc, and then doing the build and the modules_install etc.  I decided to try doing it without adding in our modules at all, and the only thing I did in menuconfig was to change the name to have a -BASE at the end, so we could see what version this was. And it resulted in the same panic happening.

We are doing this Cross Compiled on PC to save a huge amount of time. No errors seem to happen during the process, but we just cannot get the Pi to boot after extracting everything.

This example is using Windows 10 with WSL Ubuntu as the Cross Compiling system, which seems to complete the process without errors.
Using this as the base of the process:
https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel

Note - this process used to work, but after about August 2020 it stopped working and we have been unable to provide updates to our customers due to this issue.

Here are the steps I have taken, as well as the output of the extraction of the tar.gz file on the Pi, after it had been packaged on the PC.

Starting with Fresh Image, using Raspberry Pi Imager.
[image]

**We used the archive of the same kernel version** in this case (also have tried the latest), and the result is the same.

```
cd ~/
wget https://github.com/raspberrypi/linux/archive/refs/tags/raspberrypi-kernel_1.20210527-1.zip
unzip raspberrypi-kernel_1.20210527-1.zip
mv linux-raspberrypi-kernel_1.20210527-1/ linux/

mkdir ~/output/
mkdir ~/output/boot/
mkdir ~/output/boot/overlays/

cd ~/linux

KERNEL=kernel7l
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- menuconfig
make -j25 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs

sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=~/output/ modules_install

cp arch/arm/boot/dts/*.dtb ~/output/boot/
cp arch/arm/boot/dts/overlays/*.dtb* ~/output/boot/overlays/
cp arch/arm/boot/dts/overlays/README ~/output/boot/overlays/
cp arch/arm/boot/zImage ~/output/boot/$KERNEL.img

cd
tar -czvf output_X-YY-ZZ-BASE.tar.gz -C output/ boot etc lib
```

At this point I have the fresh image up and running on the Pi, I have SSH enabled, Wifi connected, and have copied this tar.gz file over to the Pi.

```
    ┌──────────────────────────────────────────────────────────────────────┐
    │                 • MobaXterm Personal Edition v21.4 •                 │
    │               (SSH client, X server and network tools)               │
    │                                                                      │
    │ ➤ SSH session to pi@192.168.178.57                                   │
    │   • Direct SSH      :  ✔                                             │
    │   • SSH compression :  ✔                                             │
    │   • SSH-browser     :  ✔                                             │
    │   • X11-forwarding  :  ✔  (remote display is forwarded through SSH)  │
    │                                                                      │
    │ ➤ For more info, ctrl+click on help or visit our website.            │
    └──────────────────────────────────────────────────────────────────────┘

Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat May  8 03:07:29 2021

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ cd Downloads/
pi@raspberrypi:~/Downloads $ sudo tar --no-same-owner -xzvf output_X-YY-ZZ-BASE.tar.gz -C /
```

The full output log of this command is attached in the txt file.
You can see the final command is an 'ls' which results in:

[BASE.txt](https://github.com/raspberrypi/linux/files/7471106/BASE.txt)

```
pi@raspberrypi:~/Downloads $ ls
-bash: /usr/bin/ls: No such file or directory
```

I have no idea what part I am doing wrong or has changed. Can anyone help?

the --no-same-owner was added in after we had problems with extraction at one point, but we have also tried it without it, and as root, with no luck.

This is just an example of doing the bare minimum and it still failing, we havent actually customised anything at this point.

WanaGo, 2021-11-04:
Starting again with a fresh imaged card for the Pi4.
Cleaning my CC output folder and linux folder, and this time loading the latest kernel repo and building that. No menuconfig done at all, I didnt even do that step, simply building the default. 
Packaged up the build to a tar.gz.
New OS on the RPi4, did update/upgrade/full-upgrade as well as the firmware update etc, so that was the latest it could be. Rebooted.
Unpackaged the tar.gz, and its dead again.

What am I doing wrong?

On PC for Cross Compiling I did the following:

```
git clone --depth=1 https://github.com/raspberrypi/linux
mkdir ~/output/
mkdir ~/output/boot/
mkdir ~/output/boot/overlays/
cd ~/linux

KERNEL=kernel7l
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig
make -j25 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs

sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=~/output/ modules_install

cp arch/arm/boot/dts/*.dtb ~/output/boot/
cp arch/arm/boot/dts/overlays/*.dtb* ~/output/boot/overlays/
cp arch/arm/boot/dts/overlays/README ~/output/boot/overlays/
cp arch/arm/boot/zImage ~/output/boot/$KERNEL.img

cd ~/
sudo su
tar -czvf Output-LATEST.tar.gz -C output/ boot lib
```

Then on the fully updated Pi4, here is the log:
[Latest-Kernel.txt](https://github.com/raspberrypi/linux/files/7472184/Latest-Kernel.txt)

Again, as you can see, **ls** doesn't even work as a command once that extraction has happened. the OS is dead from this point forward.

pelwell, 2021-11-04:
I think your problem started when the OS was changed to make `/lib` a symbolic link (to `/usr/lib`, I think). This then causes havoc when extracting your tar archive because `tar` wants to replace the symbolic link with the new `/lib` directory, which gives the appearance of deleting the original contents. If your library path includes `/lib` but not the directory it used to link to then libraries will appear to disappear and most programs will no longer run (confusingly the error will say `No such file or directory`, but it's referring to the shared libraries.

Fortunately there is a fix - use `tar`'s `--keep-directory-symlink` option (which I think of as `--dont-delete-my-stuff`),
i.e.:
```
sudo tar --no-same-owner --keep-directory-symlink -xzvf whatever.tar.gz -C /
```

WanaGo, 2021-11-04:
o...m....g....
Thank you so much!!!!

I was just about to try my 50th attempt at something slightly different, so changed and did exactly what you said, and basic commands work after extraction!!

Legend, thank you so much.

Time for more TESTING!

closed 2021-11-04 by WanaGo