raspberrypi/linux #4667: [KMS] Loading video modules manually turns off the HDMI video

vanfanel, 2021-11-04:
Hi,

I am building an embedded system on the Pi4 where I use a small script with busybox mdev to bring things up.
Everything seems to work except for video, and this is the mdev init script:
```

# Mount the filesystems it needs
mount -t sysfs none /sys
mount -t proc proc /proc

# Mount a tmpfs that will containg the dev dir
mount -t tmpfs mdev /dev

# Tell the kernel what to run when there's a hotplug 
echo /bin/mdev > /proc/sys/kernel/hotplug

# Tell mdev to scan sys and create the /dev nodes according to it's findings.
busybox mdev -s

# Load the needed modules for the connected hw automatically
find /sys/devices -name modalias -type f -print0 | xargs -0 sort -u | xargs modprobe -ab 2>/dev/null
```

The problem is that, as soon as the vc4 module is loaded, the screen goes off.
(I have tried to load it manually after system boot, and `modprobe vc4` does indeed work, but HDMI video turns off on the instant the module is loaded).

Things to note:

-I boot the system in full KMS mode, with:
`dtoverlay=vc4-kms-v3d`
-Audio is not enabled, so `dtparam=audio=on` is commented out.
-I initialize the physical video mode with `video="HDMI-A-1:1280x720@60"` in `cmdline.txt`
HDMI video works as I said until I modprobe the vc4 module.

Any idea on why does it happen? It's the last stepping stone towards properly using mdev on embedded Linux on the Pi4.

6by9, 2021-11-04:
As part of vc4_drm_bind it notifies the firmware that it has finished with the simple-framebuffer, and that the firmware should back away from any further usage of the display pipeline.
https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_drv.c#L334

I suspect that you haven't loaded some other dependency of vc4, therefore it is prevented from fully probing at that point (probably deferring). At a guess it'd be the HDMI I2C controllers (i2c_brcmstb), or the DVP reset driver (defconfig defines as a builtin - CONFIG_CLK_BCM2711_DVP).

vanfanel, 2021-11-04:
@6by9 Tried manually loading in order, with the same results:
```

~# modprobe i2c_brcmstb
~# lsmod
Module                  Size  Used by
brcmfmac              323584  0
brcmutil               24576  1 brcmfmac
cfg80211              872448  1 brcmfmac
rfkill                 36864  1 cfg80211
hid_logitech_hidpp     45056  0
hid_logitech_dj        28672  0
i2c_brcmstb            16384  0
~# modprobe vc4
```
....and the HDMI still goes off.
CONFIG_CLK_BCM2711_DVP is a built-in, yes.
Any other ideas? Seems like I am not the first person to find this today:
https://forums.raspberrypi.com/viewtopic.php?p=1933404#p1933404

dridri, 2021-11-04:
Hello, I have the same problem indeed.
It's probably worth it to say that using the fkms overlay works, and modprobing vc4 automatically loads its dependencies.

eritondev-stack, 2024-04-08:
> Hi. Solved the problem. These are the modules I have to load manually so TTY console (and all graphical apps running on KMS/DRM render contexts) work:
> 
> ```
> modprobe i2c_brcmstb
> modprobe vc4
> modprobe v3d
> modprobe snd_soc_hdmi_codec
> ```
> 
> So, booting without ANY service is possible. No UDEV is needed at all, and for libudev, libudev-zero can be used: https://github.com/illiliti/libudev-zero
> 
> Not even MDEV as hotplug manager is needed. All that is needed is to load those modules on a custom init script.
> 
> Closing this, as it's solved.
> 
> @dridri: take a look, this works, ask whatever you need, I'll be glad to help.

This is work for me. thanks!!!

vanfanel, 2025-02-21:
> Can someone please help me, i'm struggling with this with the same black screen issue. I selected the VC4, V3D, Simple framebuffer drivers in kernel, but when i try to load with either fkms or kms, i get black screen only. What did you select when compiling the kernel? Many thansk

I totally lost interest on the Raspberry Pi, but I advise you to use modprobed-db (https://github.com/graysky2/modprobed-db.git) to detect the modules used by a default Raspberry Pi OS running kernel, build a kernel based on that, and slowly removing un-needed modules one after each another. That's the proper way to build an smaller kernel for any machine.