raspberrypi/linux #4670: Broken EDID loading on TC358743

mdevaev, 2021-11-05:
**Describe the bug**
The [latest version](https://github.com/raspberrypi/linux/commit/bad3872df40dd9a4ba7ff239c17288a7a84a80f9) of 5.10 kernel does not allow loading the EDID into the TC358743 capture device. Apparently, some ednpoints have changed in the kernel judging by dmesg:
```
[root@pikvm ~]# dmesg | grep tc3
[    9.845115] tc358743 10-000f: Consider updating driver tc358743 to match on endpoints
[   10.064257] tc358743 10-000f: tc358743 found @ 0x1e (i2c-22-mux (chan_id 1))
```
At the same time, video capture works if the source resolution corresponds to 640x480.

Perhaps the problem may be related to #4669 but I'm not sure because they look quite different.

**To reproduce**
Take a fresh kernel build and try to load the EDID using v4l2-ctl. The utility will output absolutely nothing - the EDID will not be loaded.

**Expected behaviour**
The EDID must be loaded into the device and v4l2-ctl should report about it.

**Actual behaviour**
Loading the EDID does not work.

**System**
* `Raspberry Pi 4 B+ 4G`
* Arch Linux ARM / PiKVM OS
* `vcgencmd version`:
  ```
  Nov  2 2021 13:22:15
  Copyright (c) 2012 Broadcom
  version e50fe24ee2b6974f3ba6615ba0f1d8f45c485f69 (clean) (release) (start)
  ```
* `Linux pikvm 5.10.76-4-raspberrypi4-ARCH #1 SMP Fri Nov 5 08:11:58 MSK 2021 armv7l GNU/Linux`

mdevaev, 2021-11-05:
@6by9 sorry to bother you, I've brought you a new nasty problem with tc358743. In addition, it is possible that it is indirectly related to kernel oops in another issue.

6by9, 2021-11-05:
```Consider updating driver tc358743 to match on endpoints``` is an upstream thing that has changed from 5.9 onwards https://github.com/torvalds/linux/commit/e80cdf0a3843fe570c891cb976feb0c136b19fb7. Seeing as tc358743 is an upstream driver, it would have been nice if they'd fixed it at the same time.

More likely #4641 and #4645.
As long as the flag for enabling Media Controller is not set in either module parameter or DT, then there should have been no change. I'll run up a board when I get a chance and see if there is anything obvious going on.

6by9, 2021-11-05:
Just checked with an Auvidea B102 on a CM4 - no issue setting the EDID on 5.10.76, ``` v4l2-ctl --set-dv-bt-timing query```and it's then reporting the format of 1280x720, and is then fine streaming video (used GStreamer ```gst-launch-1.0 -vvv -e v4l2src ! v4l2convert ! kmssink```)

mdevaev, 2021-11-05:
It's strange, my user reproduces it. Maybe it's something floating... I'll try to figure out details.

Could you something else please? Use `ustreamer` and OMX encoding with B102. This causes an immediate kernel crash on 720p like on #4669

```
git clone https://github.com/pikvm/ustreamer
cd ustreamer
make WITH_OMX=1
```
Add `cma=128M` to cmdline.txt and `gpu_mem=256` to config.txt and try this command:
```
ustreamer --h264-sink=foo --dv-timings --persistent -m uyvy
```

6by9, 2021-11-05:
Get them to check with ```v4l2-ctl -D``` that they haven't enabled the Media Controller side of bcm2835-unicam.
Unfortunately it doesn't decode the flag, but 
```	Capabilities     : 0x85a00001```
is without Media Controller, and 
```	Capabilities     : 0xa5a00001```
is with Media Controller. (In this case I get ```VIDIOC_S_EDID: failed: Inappropriate ioctl for device``` on trying to set the EDID. You need to direct it to  ```-d /dev/v4l-subdev0``` with Media Controller - see https://forums.raspberrypi.com/viewtopic.php?t=322076 for more details).

mdevaev, 2021-11-05:
Thak you, I will. I also edited #4669. This is definitely related to OMX and TC358743.

6by9, 2021-11-05:
I've used rpi-update to grab the latest released kernel build 
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.76-v7l+ #1477 SMP Tue Nov 2 13:20:53 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:~ $ vcgencmd version
Nov  2 2021 13:22:45 
Copyright (c) 2012 Broadcom
version e50fe24ee2b6974f3ba6615ba0f1d8f45c485f69 (clean) (release) (start_db)
```
ustreamer starts fine on that. It does appear to take a LARGE amount of CPU time though - htop is reporting 3 cores at 70% usage on a CM4.

mdevaev, 2021-11-05:
What commit is your kernel built from? Maybe we have something else that might be different?

mdevaev, 2021-11-05:
OMG I made a typo with options. Please try OMX JPEG:

`ustreamer --encoder=omx --dv-timings --persistent -m uyvy`

mdevaev, 2022-03-15:
I haven't seen this problem again, probably some underground knocks related to memory corruption. Irrelevant due to M2M.

closed 2022-03-15 by mdevaev