raspberrypi/linux #4671: New release tag is a duplicate

notcarbide, 2021-11-05:
The latest release (1.20211029) and the tag before it (1.20211007) are both at commit abff5012a. If intentional, feel free to close.

Thank you!

closed 2021-11-05 by notcarbide

popcornmix, 2021-11-05:
I suspect it is deliberate.
The two most recent commits here: https://github.com/raspberrypi/rpi-firmware/commits/stable
match the tag dates, but the second one didn't modify the kernel, only the firmware.

notcarbide, 2021-11-05:
Thank you for responding! I thought something like that was the case after looking through older releases. While not common, I noticed that this has happened before multiple times and releases did not seem kernel specific. (Hence why I closed the issue).

Have a great weekend!