raspberrypi/linux #4678: Unhandled prefetch abort

g1bs0nsg, 2021-11-05:
**Describe the bug**
Running chromium in kiosk mode on a raspberry pi 3B+, I use it to display a montage of my security cameras from Zoneminder.  Have been using this for well over a year with no issues.  Installed latest updates for Raspbian Buster last week, since then the browser will randomly give an Aw, Snap! error page.  Sometimes it will run for hours with no issue, other times it will crash like this 10 times in less than a minute.  When this happens if I look in kern.log or dmesg I see the following:

    Nov  5 11:13:54 raspberrypi kernel: [ 2197.733928] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03615428
    Nov  5 11:13:54 raspberrypi kernel: [ 2197.823719] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03615428
    Nov  5 11:33:40 raspberrypi kernel: [ 3384.120853] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03645428
    Nov  5 11:33:40 raspberrypi kernel: [ 3384.190499] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03645428
    Nov  5 11:53:53 raspberrypi kernel: [ 4597.378654] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03685428
    Nov  5 11:53:54 raspberrypi kernel: [ 4597.447498] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03685428
    Nov  5 12:31:27 raspberrypi kernel: [ 6851.345091] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x035e5428
    Nov  5 12:31:27 raspberrypi kernel: [ 6851.420274] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x035e5428
    Nov  5 13:04:05 raspberrypi kernel: [ 8809.009029] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03625428
    Nov  5 13:04:05 raspberrypi kernel: [ 8809.107525] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x03625428

**To reproduce**
/usr/bin/chromium-browser --disable-infobars --kiosk 'http://testwebpage.com' > /dev/null 2>&1 &

**Expected behaviour**
Display the webpage without issue for as long as it is running.

**Actual behaviour**
Random Aww, Snap! errors

**System**
 https://pastebin.com/tam64BWb

**Logs**
Output of chrome_debug.log when a crash occurs:

https://pastebin.com/fSqznwys

**Additional context**
After this started, I upgraded to bullseye hoping that might resolve the issue, but it made no difference.  Current version of chromium installed is:

Chromium 92.0.4515.98 Built on Raspbian , running on Raspbian 11

Also happened when I removed chromium-browser and installed chromium/stable 90.0.4430.212-1

I noticed in the chrome_debug.log it always seems to happen after those entries related to my Google Mini devices... I tried disabling the Load Media Router Component Extension, and Cast Media Route Provider thinking those might have something to do with, but it made no difference.

XmlmXmlmX, 2022-02-04:
I have a very similar error when calling `chromium-browser  --noerrdialogs --disable-infobars --kiosk http://openhabian:8080`.

```
[63257.742786] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x01b1317c
[0204/153816.511979:ERROR:elf_dynamic_array_reader.h(61)] tag not found
Trace/breakpoint trap
```

Have you already found a solution @g1bs0nsg?

It feels like my SD is corrupt or something like this.

g1bs0nsg, 2022-02-04:
At some point the chromium build in the repository updated to chromium-browser/stable,now 95.0.4638.78-rpt1 armhf and the problem went away for me, I'm no longer getting the unhandled prefect abort crashes after that update.  I posted the same question to the chromium devs and they advised it was likely a distribution issue and to report to the distro maintainers instead and closed it out.

XmlmXmlmX, 2022-02-07:
I'm using chromium-browser version `95.0.4638.78-rpt6`. Still doesn't working. Same issue.

XECDesign, 2022-02-09:
What are simple steps to reproduce this which start with 'Flash the latest official raspberry pi os desktop image' and only involve publicly accessible URLs?