raspberrypi/linux #4679: Rpi3 and cpu ofline on the fly

antonellocaroli, 2021-11-06:
I don't know if it's a misconfiguration of my kernel or if it's just not possible to ofline a cpu on the fly with the command:

echo 0 > /sys/devices/system/cpu/_cpuX_/online

pelwell, 2021-11-08:
I think the offlining support requires `CONFIG_HOTPLUG_CPU` to be enabled:
```
config HOTPLUG_CPU
	bool "Support for hot-pluggable CPUs"
	depends on SMP
	select GENERIC_IRQ_MIGRATION
	help
	  Say Y here to experiment with turning CPUs off and on.  CPUs
	  can be controlled through /sys/devices/system/cpu.
```
The Raspberry Pi kernel defconfigs don't enable this setting:
```
$ grep HOTPLUG .config
# CONFIG_HOTPLUG_CPU is not set
```
What is your reason for wanting to offline a CPU? It don't think it would save any power, if that is what you are hoping to do.

antonellocaroli, 2021-11-08:
yes, hotplug was the first thing I enabled... but it doesn't seem to work on raspberry...

pelwell, 2021-11-08:
That would have been a helpful thing to include in your issue report.

1. What other things have you changed?
2. Do any messages appear in the kernel log (`dmesg`) when you try to offline a core? Note that AFAIK you can't offline cpu0 on the arm architecture.

antonellocaroli, 2021-11-09:
1. no
2. no special message.

The kernel simply does not create the paths
 /sys/devices/system/cpu/cpuX/online
or
 /sys/devices/system/cpu/cpuX/ofline

pelwell, 2021-11-09:
The BCM283x architecture support does not define `.cpu_die` and `.cpu_kill` methods because there is no way to power off or reset individual cores, and the ARM SMP code includes:
```
int platform_can_cpu_hotplug(void)
{
#ifdef CONFIG_HOTPLUG_CPU
	if (smp_ops.cpu_kill)
		return 1;
#endif

	return 0;
}
```

_What is your reason for wanting to offline a CPU? It don't think it would save any power, if that is what you are hoping to do._

lategoodbye, 2021-11-09:
There is a related old pull request #1994

antonellocaroli, 2021-11-09:
the aim was to bring a cpu ofline, then bring it back online and dedicate it to an application, this done on the fly.

Of course you can use maxcpu and/or isolcpu, in the cmdline, both of which work.

But the question was whether it was possible to do this on the fly. So far it doesn't seem possible....

pelwell, 2021-11-09:
It's not possible without software development.