raspberrypi/linux #4681: [Bisected] RPI4: Android: System completely hangs during boot

rsglobal, 2021-11-08:
**Describe the bug**
Mainline kernel v5.10.y (v5.15.y is also affected) causes system to freeze during Android boot.

**To reproduce**
- Download GloDroid sources as described here:
https://github.com/GloDroid/glodroid_manifest/tree/v0.6.1

- Update the kernel
```bash
cd kernel/glodroid-broadcom
git remote add rpi https://github.com/raspberrypi/linux.git
git remote update
git checkout rpi/rpi-5.10.y
git cherry-pick 95caee789a62f09043882ec5ffff02c2111d9224 # To remove redundant Android.mk file
```

- Build the project for rpi4, flash the board

**Expected behaviour**
Android launcher loaded and working

**Actual behaviour**
System hangs during bootanimation, serial console does not respond

**System**
RPI4B 2gb

**Investigation results**
The point where issue starts to appear bisected down-to the stable branch commit: (@v5.10):
[0674fa99a7d8dc](https://github.com/raspberrypi/linux/commit/0674fa99a7d8dc) ("fdt: Properly handle "no-map" field in the memory region")

Reverting this commit on v5.10.y
```
git revert 54d3a5761951c # to get clean revert of the commit below
git revert 0674fa99a7d8dc
```
fixes the issue.

JamesH65, 2021-11-08:
We do not have an Android port, and do not support third party Android distributions. I suggest approaching the maintainers of this particular distribution for help.

rsglobal, 2021-11-08:
I am the maintainer. And this port is based on pure rpi linux without any additional commits on top.

rsglobal, 2021-11-08:
I do not expect you will download/build the Android, but you can assist me so we will find the root cause together.

JamesH65, 2021-11-08:
Assistance for a non-Pi OS will be limited as we are very busy maintaining our own systems. Sorry.

rsglobal, 2021-11-08:
No problem.

pelwell, 2021-11-08:
The bootloader configuration blob exported by the kernel is declared in Device Tree and marked with the `no-map` attribute. The first step to understanding the issue would be to remove that attribute (or the entire `nvram@0` node) and see if that prevents the hang.

rsglobal, 2021-11-08:
Thanks, I'll do that and get back to you.

rsglobal, 2021-11-08:
Ok, looks like this line injected by mainline ATF is causing the issues:

```
                atf@0 {
                        reg = <0x00000000 0x00000000 0x00080000>;
                        no-map;
                };
```

Removed ATF from config.txt and issue has gone.

pelwell, 2021-11-09:
Is this resolved to your satisfaction? The change in `no-map` handling is due to an upstream commit, and I don't think there's a suggestion that it is being misused downstream.

rsglobal, 2021-11-09:
Is there any official RPI upstreaming-focused team which I can report this issue.

pelwell, 2021-11-09:
@vianpl added the `nvram@0` node and its `no-map` option, and he's the maintainer of the bcm2835 architecture upstream, but he's also very busy. An email to the [linux-rpi-kernel mailing list](https://lists.infradead.org/mailman/listinfo/linux-rpi-kernel) (which vianpl will receive) is probably the best place to start.

rsglobal, 2021-11-09:
Thank you for support.

closed 2021-11-09 by rsglobal

pelwell, 2021-11-09:
In case I get a chance to have a poke around, is the `atf@0` being created in the `reserved-memory` node?

pelwell, 2021-11-09:
What is your kernel load address? And am I right in thinking you are running the 64-bit kernel?

rsglobal, 2021-11-09:
```
MMC read: dev # 0, block # 329728, count 65536 ... 65536 blocks read: OK
## Booting Android Image at 0x10008000 ...
Kernel load addr 0x00080000 size 14548 KiB
Kernel command line: androidboot.hardware=rpi4 coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 vc_mem.mem_base=0x3ec00000 vc_me
m.mem_size=0x40000000 console=ttyAMA0,115200 buildvariant=userdebug
RAM disk load addr 0x03300000 size 11040 KiB
   Uncompressing Kernel Image
   Using Device Tree in place at 000000002eff3300, end 000000002f00bfff
```

pelwell, 2021-11-09:
The 64-bit kernel likes to be at `text_offset` from a 2MB boundary, where `text_offset` in our kernel builds is 0, so a load address of 0x200000 should be correct.

rsglobal, 2021-11-09:
```
    0.000000] [Firmware Bug]: Kernel image misaligned at boot, please fix your bootloader!
```
Has gone after I set 0x200000. Thanks for suggesting this.
Still it doesn't solved the issue.

But upgrading the ATF from https://github.com/ARM-software/arm-trusted-firmware/commit/de9d0d7c7ffc592e34f9c8c38c1971556d4d4de8 to https://github.com/ARM-software/arm-trusted-firmware/commit/683bb4d7bdfd42e6e026902c43797f132b2a75d5 soled.

Thanks again and sorry for consuming your time.