raspberrypi/linux #4683: PSI support

ferferga, 2021-11-08:
**Describe the bug**
``/cat/proc/pressure/*`` is not available using the latest raspberrypi-kernel package for Raspbian bullseye. Tools like oomd can't be used.

**Expected behaviour**
Linux kernel from 4.20 and up include PSI enabled, it's up to the maintainers of the distro to disable it. It should be built into the kernel (I tried setting ``psi=1`` in cmdline.txt, but it`s still not available).

**Actual behaviour**
PSI can't be used without building the kernel.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi 4B+ 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Oct 29 2021 10:47:49
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start_cd)
```
* Which kernel version (`uname -a`)?
```
Linux ferferga-server 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```

pelwell, 2021-11-09:
> Linux kernel from 4.20 and up include PSI enabled, it's up to the maintainers of the distro to disable it.

Then why does the Kconfig entry say:
```
config PSI
        bool "Pressure stall information tracking"
        help
          Collect metrics that indicate how overcommitted the CPU, memory,
          and IO capacity are in the system.
...
          Say N if unsure.
```
?
Other distributions are available that may or may not have that config setting enabled, or you can build your own kernel - see https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building.

ferferga, 2021-11-09:
@pelwell This was my source:

https://facebookmicrosites.github.io/psi/docs/overview#prerequisites

> PSI is included in Linux kernel versions 4.20 and up. If building your own kernel, make sure CONFIG_PSI=y is set in the build configuration.

Your distribution might build the kernel with PSI support but disable the feature per default (CONFIG_PSI_DEFAULT_DISABLED), in which case you need to pass psi=1 on the kernel command line during boot.
>

Yes, I know I can build my own version of the kernel to include it. My point is why it's not in the shipped in the "public" version. I'm not an expert of the feature (I want to learn about it in fact) but it doesn't seem like something that could be controversial or detrimental to the user in any way. In fact, in my opinion, this feature seems completely targeted towards low-end devices like the Pi, to help aid in the debug of memory pressure problems.

If it's the case that the feature is detrimental for the user in some way, I guess I have no other option than build the kernel myself, but I would like to know first the reasoning behind not shipping the kernel with it (so I'm aware of its limitations and maybe run it off a testing sdcard instead to play with it, instead of my main image).

Best regards.

pelwell, 2021-11-09:
Raspberry Pis, especially the earlier ones, have limited resources compared to a general purpose computer. In particular they have limited RAM, and the relatively slow storage (SD cards) makes swapping unattractive. Every built-in feature that we add increases the kernel size and reduces the free memory for applications, and with millions of users in education who just want to run Python and Scratch we have to weigh up _(advantages of the feature) * (number of users)_ against _(costs of enabling the feature) * (number of non-users)_.

graysky2, 2021-11-13:
@ferferga - Maybe you should build the current stable kernel with this enabled (as the only change `CONFIG_PSI=y`) and compare the size of the kernel package you built to the one provided.  How much space overhead does it add?

ferferga, 2022-01-02:
Sorry for my lack of response for the past weeks, been really busy with studies and until now I didn't had enough time to dedicate to tinkering.
I take this opportunity to wish you both a happy new year!

@graysky2 Thank you very much for actually taking the time to build the kernel with the feature enabled. This is great news, since 66 kilobytes is negligible with today's standards with how cheap storage is nowadays.

@pelwell This is exactly why I think this is a really nice thing to have: As I explained in my original message, my idea was to use [oomd](https://github.com/facebookincubator/oomd) to act in the userspace before the kernel had to to kill hungry processes. Having high memory usage in the Pi means the kernel starts to swap to the SD, which is known for everybody how bad that was. Before having oomd, all sign of swapping rendered my Pi unrecoverable without an unplug and replug to the PSU. Now high memory usage scenearios also make stuff a bit sluggish, but oomd acts quickly, preventing my Pi to be dead.

This is way off topic, but in my honest opinion, I think oomd should be one of the packages included by default in all Raspbian OS distros. OOMDs are less common as we have now devices with 2 - 8 GB, but even with those, some intensive tasks like video  processing or forgetting to call ``free()`` while doing some basic tinkering with C (which is something that could happen to any noobish student in C like me 😉, meaning a complete reboot and losing precious work).

My main use case for this is to be used alongside oomd to avoid those scenearios (as explained above) but, beyond giving more reliability to the boards, I don't think is *useless* for education either. Your example made me realize that PSI would've been really useful in some of my "complexity of algorithms" (not sure what's the correct name for that specific lesson in the Analysis and design of algorithms field in English, sorry) classes, since PSI provides info about IO and latency as well, which are really useful for benchmark binaries compiled using GCC's LTO or the NEON instruction set. In my opinion, the advantages PSI  *potentially* has outweights a lot the disadvantages, specially at the cost of 66 kilobytes. Honestly I think there are other packages that comes preinstalled with Raspberry Pi OS that are less worthy than the (potential) mix of oomd + PSI + students learning low-level optimization with PSI.

ferferga, 2022-01-02:
Sorry for my lack of response for the past weeks, been really busy with studies and until now I didn't had enough time to dedicate to tinkering.
I take this opportunity to wish you both a happy new year!

@graysky2 Thank you very much for actually taking the time to build the kernel with the feature enabled. I ended up compiling the kernel myself, but forgot to measure the difference. This is great news, since 66 kilobytes is negligible with today's standards with how cheap storage is nowadays.

@pelwell This is exactly why I think this is a really nice thing to have: As I explained in my original message, my idea was to use [oomd](https://github.com/facebookincubator/oomd) to act in the userspace before the kernel had to to kill hungry processes. Having high memory usage in the Pi means the kernel starts to swap to the SD, which is known for everybody how bad that was. I ended up compiling the kernel with support for it enabled and, before having oomd, all sign of swapping rendered my Pi unrecoverable without an unplug and replug to the PSU. Now high memory usage scenearios also make stuff a bit sluggish, but oomd acts quickly, preventing my Pi to be dead.

This is way off topic, but in my honest opinion, I think oomd should be one of the packages included by default in all Raspbian OS distros. OOMDs are less common as we have now devices with 2 - 8 GB, but even with those, some intensive tasks like video  processing or forgetting to call ``free()`` while doing some basic tinkering with C (which is something that could happen to any noobish student in C like me 😉, meaning a complete reboot and losing precious work).

My main use case for this is to be used alongside oomd to avoid those scenearios (as explained above) but, beyond giving more reliability to the boards, I don't think is *useless* for education either. Your example made me realize that PSI would've been really useful in some of my "complexity of algorithms" (not sure what's the correct name for that specific lesson in the Analysis and design of algorithms field in English, sorry) classes, since PSI provides info about IO and latency as well, which are really useful for benchmark binaries compiled using GCC's LTO or the NEON instruction set. In my opinion, the advantages PSI  *potentially* has outweights a lot the disadvantages, specially at the cost of 66 kilobytes. Honestly I think there are other packages that comes preinstalled with Raspberry Pi OS that are less worthy than the (potential) mix of oomd + PSI + students learning low-level optimization with PSI.

FallenChromium, 2022-01-26:
I just wanted to pop in to say that enabling PSI would allow to run low-memory killer of Android on the Pi's kernel, which means things like WayDroid or redroid (Android in docker, basically) can run. It is a small usecase suited mostly to the RPi 4 (maybe RPi 3, too), but it's a great helper in low-cost Android development. Being able to use `oomd` with Raspberry Pi 4 seems like a necessity, because it's being marketed as an actual "computer replacement". Maybe we could enable it in bcm2711 config, or on aarch64 config only?

pelwell, 2023-07-17:
Added by fcfe018feab97279f946bb7fc9049c28268a1a8f. Note that cmdline.txt requires `psi=1` to enable.

closed 2023-07-17 by pelwell