raspberrypi/linux #4686: Official Raspberry DSI 7" Display not working

mat-l, 2021-11-10:
**Describe the bug**
After switching on the RPi 3b the rainbow color screen is loading and the Raspberry Pi Logo is shown. Afterwards the display switches off and stays off. Starting with a hdmi cable is working.

**To reproduce**
I have restartet the system several times, but the display not not staying active.

**Expected behaviour**
Well, the display should work as in RPiOS Buster

**Actual behaviour**
DSI Display not working

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

6by9, 2021-11-10:
This was reported via the [forums too](https://forums.raspberrypi.com/viewtopic.php?p=1936141#p1936141), so I've been looking already.

This appears to be an I2C issue on Pi3 (and possibly earlier, but not Pi4) that causes the touchpanel regulator to lock up.

Adding "dtparam=i2c_vc_baudrate=50000" to /boot/config.txt seems to fix it for now, and I'll look at what's actually going on.

6by9, 2021-11-10:
It blanks on Pi4.

At the point you see "starting file system check" or similar, the Linux kernel resets the ENTIRE display pipeline, including disabling the panel. (It should kill the power to the panel to stop the blanking being visible, but that would appear not to be working either).
On Pi3 the reinitialisation isn't working as the controller locks up, but it is on Pi4. This controller has been the bane of my life for the last 4 months or so.

jimmyeao, 2021-11-11:
Adding "dtparam=i2c_vc_baudrate=50000" to /boot/config.txt  fixes the display - I can move the mouse with my finger, but it does not appear to pick up a tap as a mouse click.

SpintroniK, 2021-11-11:
> This was reported via the [forums too](https://forums.raspberrypi.com/viewtopic.php?p=1936141#p1936141), so I've been looking already.
>
> This appears to be an I2C issue on Pi3 (and possibly earlier, [...]

That kind of solved the problem for me, but not entirely, as the screen still goes gray for a second or two before it becomes normal again.
It seems that adding lcd_rotate=2 at the end of /boot/config.txt works before the screen goes gray, but after the setting is not applied anymore, and the screen displays things upside down.
I experimented a bit, and it seems that running raspi-config and choosing the legacy gl driver fixes the issues I mentioned above.
Here's the content of /boot/config.txt:
```
 For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
framebuffer_width=1000
framebuffer_height=600

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=85

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtparam=i2c_vc_baudrate=50000
lcd_rotate=2
```
Note that, if I remove the line `dtparam=i2c_vc_baudrate=50000`, it works as well, so I'm not sure the problem is only due to the I2C baud rate.

I did a few more tests, here are my conclusions:
- Full kms driver + `dtparam=i2c_vc_baudrate=50000` in `/boot/config.txt` => works but the screen still goes gray for a couple of the seconds before things get back to normal, and the `lcd_rotate=2` has no effect, so everything is upside down.
- Legagy GL driver with or without `dtparam=i2c_vc_baudrate=50000` in `/boot/config.txt` => everything works as expected.

JamesH65, 2021-11-12:
During boot, when KMS takes over, the display will be restarted, which is the grey bit.  lcd_rotate does not have any effect in KMS, you need to use xrandr or the screen configuration utility.

The legacy driver would be expected to work - we have not changed that at all.

SpintroniK, 2021-11-12:
Indeed, it works. Just to make sure other people don't get confused: the screen configuration utility doesn't show up in legacy mode, you have to be in KMS to see it.
Another thing is that, although the screen can be inverted using the screen config utility, that only rotates the image, so when I touch somewhere on the screen, it is inverted. Any easy way to solve that?
Unfortunately, I see that the screen resolution is now limited to 800x480. I had changed that to 1000x600 using framebuffer_width and framebuffer_height in config.txt, but this has no effect now. Is it possible to do something similar in KMS?

tigernero79, 2021-11-12:
https://forums.raspberrypi.com/viewtopic.php?p=1937380#p1937380

ralphrmartin, 2021-11-13:
I have this problem too.
Interestingly, for the 64-bit version of bullseye on the same system, the screen display appears to work normally, BUT does not respond to touch at all (cannot drag cursor etc).

jkbgbr, 2021-11-14:
I have the same problem under Bullseye 32-bit with a Pi 2 Model B v1.1
I tried to solve mostly from hardware side, as the pi wasn't used for a few years now. I'm releived to know that this is probably a software issue and I don't have to throw away the hardware. I'll fire up a Buster to see if this still holds.

6by9, 2021-11-14:
I've already posted the workaround at https://github.com/raspberrypi/linux/issues/4686#issuecomment-965294637
"Me too" posts add nothing.

Add ```dtparam=i2c_vc_baudrate=50000``` to config.txt for the time being. The full reasons are being investigated.

vc4-kms-v3d TOTALLY replaces the firmware display stack, so nothing that the firmware sets up is applicable.

jimmyeao, 2021-11-14:
That doesn’t fix the no touch issue

Jimmy White

I've already posted the workaround at #4686 (comment)<https://github.com/raspberrypi/linux/issues/4686#issuecomment-965294637>

UliPrantz, 2021-11-14:
If the display driver turns out to be the problem, then it would be worth to also fix #3777 and probably have a look at the already fixed touchscreen driver provided by @ardera in this [repo](https://github.com/ardera/raspberrypi-fast-ts). Unfortunately the driver needs some fixes for the current Raspbian version since it doesn't seem to work out of the box anymore (at least not for me) but maybe @ardera knows a fix?

I also can confirm the issue with the mouse just moving around and that "left click" isn't working on the Raspberry Pi Model B+ as mentioned by @jimmyeao before.

jkbgbr, 2021-11-15:
> I've already posted the workaround at [#4686 (comment)](https://github.com/raspberrypi/linux/issues/4686#issuecomment-965294637) "Me too" posts add nothing.
>
> Add `dtparam=i2c_vc_baudrate=50000` to [...]

neither does providing a workarund not mentioning my version of the hardware. I was just trying to provide you with additional information to help resolve the issue. you are welcome.

noisysoil, 2021-11-15:
Pitching in with some more info:
**Pi hardware**
```
Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000a42ea780
Model           : Raspberry Pi 3 Model B Rev 1.2
```

**Display**
Official Touchscreen display, version without backlight brightness control (v1.0 as far as I can gather).

**config.txt**
```
arm_64bit=1
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=vc4-kms-v3d
hdmi_blanking=2
...etc with default clean Bullseye install values ...
```

**Kernel**
```
5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```

Adding `dtparam=i2c_vc_baudrate=50000` did not work for me; there is the initial boot text, but then I had the grey screen with dim vertical stripes as observed by others.

However, I noticed that my kernel log (with dmesg) was spammed with these messages:
```
[  162.797207] Registered IR keymap rc-cec
[  162.797397] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[  162.797566] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input1359
[  162.808109] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[  162.808327] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[  162.852993] hdmi-audio-codec hdmi-audio-codec.1359.auto: ASoC: error at snd_soc_component_set_jack on hdmi-audio-codec.1359.auto: -95
[  162.905250] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[  162.906285] Registered IR keymap rc-cec
[  162.906487] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[  162.906663] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input1360
[  162.913289] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[  162.913687] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[  162.965036] hdmi-audio-codec hdmi-audio-codec.1360.auto: ASoC: error at snd_soc_component_set_jack on hdmi-audio-codec.1360.auto: -95
[  163.057891] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[  163.058928] Registered IR keymap rc-cec
[  163.059119] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[  163.059293] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input1361
[  163.073126] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[  163.073408] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[  163.108716] hdmi-audio-codec hdmi-audio-codec.1361.auto: ASoC: error at snd_soc_component_set_jack on hdmi-audio-codec.1361.auto: -95
```

After a headscratch. I tried this change; then the display started working (regardless of `hdmi_blanking=2` enabled or not):
```
dtparam=audio=off
dtparam=i2c_vc_baudrate=50000
```
The strange this is that I then re-enabled `dtparam=audio=on` and the display continued to work after reboots / power-off-cycles (unplugged for >10 seconds).

6by9, 2021-11-15:
You do need to power cycle the screen after adding ```dtparam=i2c_vc_baudrate=50000```. A simple reboot is not sufficient.
The issue is that the controller on the panel totally locks up for reasons only partially understood, and can only be recovered by a power cycle.

noisysoil, 2021-11-15:
> You do need to power cycle the screen after adding `dtparam=i2c_vc_baudrate=50000`. A simple reboot is not sufficient. The issue is that the controller on the panel totally locks up for reasons only partially understood, and can only be recovered by a power cycle.

Thanks, that's good to know - I did notice that reboots didn't work (in fact, my device locked up when I tried).

ralphrmartin, 2021-11-15:
Just to make it clear, power cycling does not solve the "no touch" issue.

jimmyeao, 2021-11-15:
> Have you run sudo apt update sudo apt dist-upgrade since install, or allowed updates to be applied as part of the initial wizard?
>
> Having flashed a clean image I did find that the touch wasn't being actioned by the desktop on a Pi3. Updated, and with the workaround almost all seems good - the one niggle is that with my screen configured to be inverted, the touch isn't. xinput --map-to-device "generic ft5x06 (79)" DSI-1 deals with that, but it should have been done automatically.

I just commented out the kms driver in config.txt until it gets fixed. That gets it working again

Jimmy White

tigernero79, 2021-11-15:
> Have you run `sudo apt update` `sudo apt dist-upgrade` since install, or allowed updates to be applied as part of the initial wizard if you've just done a fresh install?
> 
> Having flashed a clean image I did find that the touch wasn't being actioned by the desktop on a Pi3. Updated, and with the workaround almost all seems good - the one niggle is that with my screen configured to be inverted, the touch isn't. `xinput --map-to-device "generic ft5x06 (79)" DSI-1` deals with that, but it should have been done automatically.

with the parameters "dtparam=i2c_vc_baudrate=50000" and option lcd_rotate = 2 
to me everything seems to work even touch...  

i could not find more setting from gui to rotate it and i put LCD rotate item back

UliPrantz, 2021-11-15:
Could somebody provide a hash from the [Hexxeh/rpi-firmware](https://github.com/Hexxeh/rpi-firmware/commits/master) repo which could be used for `sudo rpi-update hash`. 

A version where the OpenGL (Full KMS) driver works and the touchscreen for RPi 3. This version could be used as a workaround as long as this issue isn't fixed. Since for some application the OpenGL (Full KMS) driver is mandatory.

tigernero79, 2021-11-16:
I don't know if this is what you are looking for.  

i got hash by typing: "cat /boot/.firmware_revision" on my pi3A tell me if it's ok.  

value = "f2b42c69f3568a8446b0fa39a0437cbd09adc60d"

ralphrmartin, 2021-11-16:
> Have you run `sudo apt update` `sudo apt dist-upgrade` since install, or allowed updates to be applied as part of the initial wizard if you've just done a fresh install?

How? Without touch, there is no way to open a terminal window for keyboard input, or keyboard to type on.

noisysoil, 2021-11-16:
> > Have you run `sudo apt update` `sudo apt dist-upgrade` since install, or allowed updates to be applied as part of the initial wizard if you've just done a fresh install?
> 
> How? Without touch, there is no way to open a terminal window for keyboard input.

How about enabling ssh and log in from there with a known device / host name (e.g. `mypi.local`); if installing from fresh, the official Raspberry Pi Imager has a handy menu (CTRL + SHIFT + X) that allows you to pre-set hostname and ssh.

sergkonn, 2021-11-16:
In my case (RPi 3b 1.2 + DSI 7" + RPi Camera2) even switching off KMS doesn't fully fix the problem. With the old graphic stack everything works fine for 2-3 hours, but then the touchscreen randomly stops to work while the X server itself continues to work normally (i.e. it is fully functional if external USB keyboard and mouse are connected). And restart of the X server leads to that the new xorg process is stuck in D state and an kernel warning/oops is generated (please see attached [raspberry-display-warning.txt](https://github.com/raspberrypi/linux/files/7544754/raspberry-display-warning.txt)). Also after that in dmesg there are messages like:
```
[67411.663550] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 4294967186 (-34)                                                                                                
[67412.703008] hwmon hwmon1: Failed to get throttled (-110)
[67413.742542] graphics fb0: Set display number call failed. Old GPU firmware?
[67414.782024] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -110
[67416.861339] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-110)
[67416.861349] bcm2708_fb_pan_display(0,0) returns=-110
```
and only full power cycle helps to recover the system.

If the X server is restarted before the touchscreen stopped to work then the restart works completely fine.

`dtparam=i2c_vc_baudrate=50000` doesn't help to prevent this issue.

In dmesg I also see from time to time random messages like `i2c-bcm2835 3f205000.i2c: Got unexpected interrupt (from firmware?)`, but the touchscreen doesn't stop to work immediately after them, so they could be unrelated (that system has also bme680 sensor connected vi GPIO I2C).

sergkonn, 2021-11-16:
BTW, `dtparam=i2c_vc_baudrate=50000` doesn't help me to start the system in KMS mode, the screen remains grey and dmesg is full of the messages related to hdmi-audio-codec like @noisysoil mentioned above, pulseaudio using > 50% cpu, etc. Powercycled the system multiple times, have all the latest bullseye updates.

noisysoil, 2021-11-16:
> `dtparam=i2c_vc_baudrate=50000` doesn't help to prevent this issue.
>
> In dmesg I also see from time to time random messages like `i2c-bcm2835 3f205000.i2c: Got unexpected interrupt (from firmware?)`, [...]

Interesting - have you tried with the following and then power-cycling Pi after the first reboot attempt:
```
dtparam=audio=off
dtparam=i2c_vc_baudrate=50000
```

ralphrmartin, 2021-11-16:
> Have you run `sudo apt update` `sudo apt dist-upgrade` since install?

Having done that, the touchscreen is responding erratically: sometimes the touch responds, and sometimes it does not.

sergkonn, 2021-11-17:
> Interesting - have you tried with the following and then power-cycling Pi after the first reboot attempt:
> ```
> dtparam=audio=off
> dtparam=i2c_vc_baudrate=50000
> ```
I've managed to run my RPi3 with KMS enabled. Looks like the problem was caused by `enable_uart=1` in the config. With enabled UART the screen remained gray and there were hdmi related errors in dmesg even with `dtparam=audio=off` and with UART disabled RPI3 works fine in KMS mode even with `dtparam=audio=on`.
AFAIK `enable_uart=1` adjusts VC core clock, so may be it nullifies `i2c_vc_baudrate=50000` effect.

Looks like this problem is not entirely DSI display specific, but something is wrong on RPi3 with the firmware which goes with bullseye.

P.S. To rule out hardware problems I put back to my RPi3 an sdcard with buster with all latest updates and everything works fine, no touchscreen issues after running for more than 10 hours.

w8iss, 2021-11-19:
Since I am not going to be worrying about the touchscreen that much yet, I have done the mentioned 'fixes' and have gotten my 
screen working on a model 3 A+ other than the upside down screen issue.

Did try it with a brand new distro load and it worked with no issues with my model 4 B+.

Hope this helps some.

xbgmsharp, 2021-11-19:
I have exactly the same issue, RPI3 with Official Raspberry DSI 7".
Adding `dtparam=i2c_vc_baudrate=50000` to `/boot/config.txt` seems to fix it for now, however the displays is upside down.

Having `rotate_lcd=2` in the config file doesn't work in Bullseye. yet..(i hope)

Stuck sticking with Buster until it's fixed.

tigernero79, 2021-11-19:
For me option lcd_rotate = 2 works on pi3a and bullseye

rnayabed, 2021-11-20:
You guys SERIOUSLY shouldn't have removed the `fkms` from raspi-config. `kms` is still too new and unstable for usage. It doesnt work with any of the HDMI touch screens I have

6by9, 2021-11-20:
>  It doesnt work with any of the HDMI touch screens I have

Irrelevant to this issue then which is related to the 7" DSI panel.
If you have an issue with an HDMI touchscreen then open a new issue.

rnayabed, 2021-11-20:
> > It doesnt work with any of the HDMI touch screens I have
> 
> Irrelevant to this issue then which is related to the 7" DSI panel. If you have an issue with an HDMI touchscreen then open a new issue.

Sure, also opened another issue in raspi-config for the request to add `FKMS` option back

rnayabed, 2021-11-20:
https://github.com/raspberrypi/linux/issues/4719

tigernero79, 2021-11-20:
If I redo the clean installation and update with sudo apt update and sudo apt full-upgrade the problem reappears on the display. black screen and white lines. forcing me to put the "" strange line back because then putting it and commenting on it the system then goes

UliPrantz, 2021-11-22:
Is there anyone actively working on this issue? Seems like this issue is pretty crucial for anyone who uses the official Raspberry Pi Touchscreen as their only display and input device which in my opinion is a pretty common use case. Would be nice to have some updates or at least have clarity if anything is done about this at all.

JamesH65, 2021-11-22:
Yes, it is being worked on.

pelwell, 2021-11-22:
As stated on a duplicate thread (https://github.com/raspberrypi/firmware/issues/1658):
> There is a known issue on the Pi3 (and possibly earlier) that is still being investigated.
Adding dtparam=i2c_vc_baudrate=50000 to config.txt is the current workaround without killing 3D acceleration.

6by9, 2021-11-22:
And already stated earlier in this thread https://github.com/raspberrypi/linux/issues/4686#issuecomment-965294637

> This appears to be an I2C issue on Pi3 (and possibly earlier, but not Pi4) that causes the touchpanel regulator to lock up.
> Adding "dtparam=i2c_vc_baudrate=50000" to /boot/config.txt seems to fix it for now, and I'll look at what's actually going on.

There are a number of issues that have been noted since the Bullseye release. Triaging and then fixing them all takes a little time.

gwest7, 2021-11-25:
As have been mentioned by a few users `dtparam=i2c_vc_baudrate=50000` helped, but the cursor will follow my touches while taps don't result in a click. I do not have rotation issues.

I have noticed though that my very first tap (after boot) does result in a click, but nothing after that.

a-camacho, 2021-11-29:
Hi everyone.

Fresh install of Raspbian (Bullseye) and touchscreen not working (works then black screen).
Used `dtparam=i2c_vc_baudrate=50000` workaround, and screen works now.

But have no touch functionnality. Is that normal ?
Is the best solution for now to get back to Buster (for use screen + touch) ?

Thanks a lot.

6by9, 2021-11-29:
Touch should be fixed by #4736 that was merged on Friday.
I'm still investigating the full reasoning behind the Pi3 I2C issues to the display regulator.

mr-gosh, 2021-12-10:
> Legacy GL driver with or without `dtparam=i2c_vc_baudrate=50000` in `/boot/config.txt` => everything works as expected.

How to activate this driver? can't find an option for that in raspi-config...

fabriziorizzo, 2021-12-11:
These fixes did not work for me on a fresh install with 2021-10-30-raspios-bullseye-armhf-lite.img on a rpi2b.
Any other ideas?

mr-gosh, 2021-12-15:
this works for me:

```
#XPT2046 Touch Screen
dtparam=i2c_arm=on
dtparam=i2c_vc_baudrate=50000
dtparam=spi=on
dtoverlay=ads7846,penirq=25,speed=10000,penirq_pull=2,xohms=150
lcd_rotate=2
```

obrien-james, 2021-12-17:
> this works for me in my config.txt:
> 
> ```
> #XPT2046 Touch Screen
> dtparam=i2c_arm=on
> dtparam=i2c_vc_baudrate=50000
> dtparam=spi=on
> dtoverlay=ads7846,penirq=25,speed=10000,penirq_pull=2,xohms=150
> lcd_rotate=2
> ```

This worked for me, then I used the screen configuration program on the pi desktop to rotate the screen correctly.

mat-l, 2021-12-26:
@6by9 I have added the line "dtparam=i2c_vc_baudrate=50000" and the display is working, but touch what do I have to add to an existing installation that touch is working fine.
Currently the display is detecting the touch event, but it seems that it is keeping the touch event, as if I am clicking a second time, it is always drawing the rectangle in the desktop. Furthermore I cann not click on menu buttons or others. Please see attached screenshot:
[image]

fkjohn, 2021-12-29:
I just came across the same issue with the latest Bullseye OS on a RasPi 3A+ and the official 7" Touchscreen and want to share the solution that worked for me.

Disable GL Driver: 
1. _sudo raspi-config_ 
2. Go to _6 Advanced Options_
3. Go to _A2 GL Driver_
4. Check _G1 Legacy_

Afterwards, X Server was not working for me. 
After looking in the error logs, _sudo apt-get install xserver-xorg-video-fbturbo_ solved the issue for me.

Finally, the touchscreen display and touch input are working.

JamesH65, 2021-12-30:
> I just came across the same issue with the latest Bullseye OS on a RasPi 3A+ and the official 7" Touchscreen and want to share the solution that worked for me.
>
> Disable GL Driver:
>
> 1. _sudo [...]

Although that will work, you will be using the deprecated graphics libraries. You need to be fully up to date (new Bullseye image, plus apt update./full-upgrade) for KMS to work, IIRC there might still be some edge cases that fail to start up the DSI.  These should be fixed early in the New Year, so you should move back to KMS as soon as possible.

iqgrande, 2021-12-30:
> Although that will work, you will be using the deprecated graphics libraries. You need to be fully up to date (new Bullseye image, plus apt update./full-upgrade) for KMS to work, IIRC there might [...]

This message makes it seem like things are fixed with a few edge cases lurking. So, I just wanted to mention my circumstances to make sure I'm not missing anything. I have a Raspberry Pi 3 Model B Rev 1.2 that is running Raspberry Pi OS Bullseye (updated seconds before I crafted this comment for reference). My `/boot/config.txt` is below, which is the default one with commented-out items removed for brevity and 2 extra lines at the bottom. I have a list below with some notes and corresponding questions. Thank you for any clarification that you can provide. Also, please note that I have been very pleased with the Raspberry Pi ecosystem and support; thank you for everything you do. 

1. The second-to-last item (`dtparam=i2c_vc_baudrate=50000`) still appears to be needed for things to work. Will this always be needed or will updates in 2022 remove this requirement?
2. The last item (`lcd_rotate=2`) appears to work until the first time the splash screen fades to gray. Then, afterwards, when the display comes back because of `dtparam=i2c_vc_baudrate=50000` it is upside down. The old driver and `lcd_rotate=2` would have the screen right-side up without any X11 tomfoolery. Will the updates in 2022 also make it to where things are not upside down? Please note that the X11 option is not one I'm keen to use because I may want this to boot to a terminal rather than X11 and I will want the screen to be rotated appropriately there, too (and I believe the X11 "fixes" will not address this). 

```
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_overscan=1
[cm4]
otg_mode=1
[all]
[pi4]
arm_boost=1
[all]
dtparam=i2c_vc_baudrate=50000
lcd_rotate=2
```

randomscot22, 2022-01-04:
> @6by9 I have added the line "dtparam=i2c_vc_baudrate=50000" and the display is working, but touch what do I have to add to an existing installation that touch is working fine.
> Currently the display [...]

Out of interest, on boot if you click the top left RPI icon on the desktop do you get the menu, but then the screen is unresponsive ? I’m seeing that behaviour on a Pi 4B (as well as the grey box drag as you noted)..

mcnahum, 2022-01-18:
"dtparam=i2c_vc_baudrate=50000" & using the tool for configuring the screen worked for me ( fresh install)
but the issue remaining is the resolution who is now 800x480 and was before 1024x614 ... any way to fix that part?

SpudGunMan, 2022-01-29:
just downloaded jan 28th build and found this bug still exists

randomscot22, 2022-01-29:
Quick update - I downloaded imager a few days ago (ca. 25 Jan’22) and created a new clean Desktop image for my Pi 4B with 7” Official Touchscreen. The latest build appears to have fixed things for me (screen and touch works). I didn’t do any edits to config files.

the78mole, 2022-02-07:
Same problems here, tried every suggestion in this and other threads (updateing, upgrading, 32-Bit, 64-Bit, legacy, gl,... different options in config.txt), installed fresh system on SD-Card about 10 times, but still display is not working. Touch does somehow, because after screen goes to sleep, a touch wakes it up. Hope it will be resolved soon. I'm running a Pi 3B Rev 1.2 with the original 7" Display Rev 1.1 from2015. It worked about 1 or 2 years ago and now I wanted to setup a kiosk system for my smart home...

lrmoreno007, 2022-02-08:
Try this in the first line of your Kivy project
```
import os
os.environ['KIVY_GRAPHICS'] = 'gles'
os.environ['KIVY_WINDOW'] = 'sdl2'
os.environ['SDL_VIDEO_X11_VISUALID'] = ''
os.environ['DISPLAY'] = ':0.0'
```

This line will make the trick: `os.environ['SDL_VIDEO_X11_VISUALID'] = ''
`

Alternatively you can write `export SDL_VIDEO_X11_VISUALID ''` in bash, but you need write it every restart

UliPrantz, 2022-02-20:
After it seemed to work quite well for a time I've got a new problem with the display now. After reboot in the 64-bit and 32-bit version (clean install with the raspberry imager tool today of both versions) the display just goes black after every reboot. The only thing that fixes that issue is to completely power off the pi (disconnecting the power cable) and boot again.

Anonymous-Ol, 2022-03-15:
I am also having this exact same issue. on a raspberry pi 4. I don't know what to do. My only problem is that when I am using an os other than raspberry pi os, like LineageOS, or Ubuntu Desktop for Pi, it won't boot at all.

peter-mount, 2022-03-24:
Just to say I hit this myself today. RPI4B with latest Bullseye (downloaded on Tuesday) & installed a new touchscreen to it today. Only additional hardware is a Zigbee3 adapter connected to the gpio pins.

The screen displayed fine but with no touch until I commented out:
```
# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2
```
from /boot/config.txt then after a reboot touch worked.

One thing I did notice was that before commenting out those two lines the mouse pointer was visible on the desktop, but afterwards it's not - which is fine as this machine is going to be wall mounted as it's the controller for home automation.

[11 more replies]