raspberrypi/linux #4688: dphys-swapfile extremely slow in Bullseye

Qengineering, 2021-11-11:
Sometimes extra swap space is necessary. For instance, the installation from scratch of OpenCV 4.5.4 needs 6 GB of RAM.
Temporarily enlarging the swap space is most of the time the only option.

Altering the amount of space in /etc/dphys-swapfile (CONF_SWAPSIZE=x) and subsequently rebooting is the way to go.
If more than 2048 MB is required, you have to alter the upper limit in /sbin/dphys-swapfile (CONF_MAXSWAP=y) first.

When doing so, it takes an extremely long time to reboot. More than 6 minutes staring at a blank screen, with only a 'live' LED flickering.

Raspberry Pi 4, 2 GB -> Bullseye 32 bit -> 4 minutes
Raspberry Pi 4, 2 GB -> Bullseye 64 bit -> 4 minutes
Raspberry Pi Zero 2, 500 MB -> Bullseye 32 bit -> 6 minutes
Raspberry Pi Zero 2, 500 MB -> Bullseye 64 bit -> 7 minutes

Only expanding the swap space with the /etc/dphys-swap file also takes an extremely long time to accomplish, one or two minutes less than the ones mentioned above. Once set, and reducing the swap space back to its original 100 MB takes no extra time.

In Buster, the setting of /sbin/- and /etc/dphys-swapfile took no significant extra time.

piaseckim, 2022-05-02:
Confirming, 
Pi4/2GB Ram/bullseye/64bit -> 15 minutes to start up dphys-swapfile.service 
My workaround was to create and use swap partition (pi is booting from HDD via USB)
Systemd now starts up all services in ~55s
```
pi@raspberry:~ $ sudo systemd-analyze
Startup finished in 4.513s (kernel) + 50.670s (userspace) = 55.184s 
multi-user.target reached after 50.600s in userspace
```

XECDesign, 2022-05-03:
An updated package is available in the 'untested' component of the repo.

http://archive.raspberrypi.org/debian/pool/untested/d/dphys-swapfile/dphys-swapfile_20100506-7+rpt1_all.deb

Qengineering, 2022-05-04:
My biggest fear is that the 'new' dphys-swapfile performs a time-consuming sort of read/write memory sanity check and thus wear the SD card unnecessarily. That's why I use the good old Linux `$ sudo swapon /var/swapfile`

XECDesign, 2022-05-04:
It does not.

XECDesign, 2022-05-05:
The updated `dphys-swapfile` package is now in the `main` component.

closed 2022-05-05 by XECDesign

darkbasic, 2022-05-09:
Did the fix land in official Debian repos as well? Because I'm using Debian 11 (not Raspberry Pi OS) and it still take ages...

XECDesign, 2022-05-09:
No, for Debian this is the desired behaviour. If you would like to make a case otherwise, you'd need to do so [here](https://www.debian.org/Bugs/).

Qengineering, 2022-05-11:
Works perfect, thanks!