raspberrypi/linux #4690: Default cpu governor differs from mainline aarch64

beadleha, 2021-11-11:
**Describe the bug**
The mainline Linux kernel made shedutil the default config's CPU frequency governor for aarch64 in 5.9. I see no reason why that should be overridden by the raspberry pi kernel, except in cases where it may not be supported. I know of no such cases at the moment. At least I can confirm that schedutil works great on my Pi 400.

**To reproduce**
Check the default CPU frequency governor here for example:
https://github.com/raspberrypi/linux/blob/9878a118cbdb7779fd15caa45915d22bff381ba3/arch/arm64/configs/bcm2711_defconfig#L49

**Expected behaviour**
Since mainline uses schedutil, I would expect schedutil to be the default unless there is a specific reason why it is not suitable for some Raspberry Pi hardware.

**Actual behaviour**
Powersave is the default cpu governor in defconfig for bcmrpi3 (64 bit), bcm2711(32 and 64 bit), bcmrpi (32 bit), bcm2709 (32 bit).

**System**
Pi 400 (bcm2711)
64-bit Void Linux with glibc for Pi 4
5.10.54

**Additional context**
If there are cases where schedutil is unsuitable for some Raspberry Pi models, perhaps we can enable it by default on a case by case basis, for instance on bcm2711 since it appears to be problem-free.

On the other hand if there is no particular reason to use powersave by default on Raspberry Pi, maybe we should follow mainline and use schedutil in general now.

pelwell, 2021-11-11:
How about making a case for why schedutil is a better governor than ondemand, which is what all Pis switch to within seconds of the kernel starting?

beadleha, 2021-11-11:
> How about making a case for why schedutil is a better governor than ondemand, which is what all Pis switch to within seconds of the kernel starting?

1) My Pi on Void Linux was not switching to ondemand. It was using powersave (eg, stuck at 0.60 GHz) until I changed it via /sys/devices/system/cpu/cpufreq/policy0/scaling_governor. Can you describe what would normally cause it to change to ondemand? Maybe I should also look into why that does not happen on Void.

2) A technical comparison between two different options that we do not currently use would be irrelevant. The fact is that mainline uses schedutil for aarch64 since Linux 5.9 and is attempting to make that the default across all platforms. Is there a technical reason why we cannot just do the simplest thing and follow mainline on this? Wouldn't the devices you claim automatically switch to ondemand still do that?

beadleha, 2021-11-11:
I'm just saying that it probably makes the most sense to use the dame default as mainline unless there is a specific reason not to. Users and distributions can easily override it. There appears to be no reason for it to be powersave in particular, and it differs from mainline.

pelwell, 2021-11-12:
If you are asking us to make a change that will affect (at least) all aarch64 users, you should be prepared to back it up with some, you know, evidence that it might actually make things better - either by adding useful new facilities or by just doing its job better. The benchmark results I've seen comparing it to `ondemand` have been a mixed bag, with `schedutil` either being marginally ahead, quite a bit behind, or roughly the same.

beadleha, 2021-11-12:
>  If you are asking us to make a change that will affect (at least) all aarch64 users

Actually, I am asking if we could revert a change. I have heard no technical argument in favor of powersave as default when it is not the default upstream.

> The benchmark results I've seen comparing it to `ondemand` have been a mixed bag, with `schedutil` either being marginally ahead, quite a bit behind, or roughly the same.

That may be true, but I am not proposing to switch away from ondemand. The default is currently powersave, and I believe you'll find that the benchmarks for that are even worse.

pelwell, 2021-11-12:
The current arrangement suits are needs. If you want us to change you'll need to be more persuasive.

closed 2021-11-12 by pelwell

beadleha, 2021-11-12:
Can you tell me why powersave is the default? Is there any reason? It is different from upstream, so there should be a reason. This change would benefit anyone who uses this kernel as-is and expects the defaults to match the ones found in mainline Linux.

Why do you keep bringing up ondemand? The mainline linux kernel uses schedutil, and so should we, unless there is a reason not to. Someone is investing effort in changing it in each new release, and I can see no reason why.

I'm not going to debate performance or the merits of one scheduler versus another. This is a Linux kernel for the Pi, and it should have a reason when it varies from the mainline Linux kernel. Doing otherwise can lead to unforeseen inconveniences. This is simply a best practice.