raspberrypi/linux #4693: No video after updating to 5.10.78-2

mminelli, 2021-11-13:
**Describe the bug**
After updating my RPi 3 model B to the latest kernel `linux-raspberrypi-5.10.78-2`, I have no HDMI output to my TV. This has been independently confirmed by another user.

**To reproduce**
Install `linux-raspberrypi-5.10.78-2` on a Raspberry Pi 3 model B, connected to a monitor via HDMI and reboot.

**Expected behaviour**
Video output works.

**Actual behaviour**
The monitor shows "no signal" and error messages in `dmesg` and `journalctl` mention `"flip_done timed out"`.
Downgrading the kernel to `5.10.76-1` and rebooting fixes the problem.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 3 model B

* Which OS and version (`cat /etc/rpi-issue`)?
Arch Linux arm

* Which firmware version (`vcgencmd version`)?
Nov  8 2021 18:53:02 
Copyright (c) 2012 Broadcom
version 4f73dcaefcfd5b20317e44a81d10e9d74fd3dffe (clean) (release) (start)

* Which kernel version (`uname -a`)?
`Linux alarmpi 5.10.76-1-raspberrypi-ARCH #1 SMP Thu Oct 28 14:06:32 UTC 2021 armv7l GNU/Linux`

**Logs**
I have already downgraded the kernel, so I don't have the logs anymore. If necessary, I can reproduce the problem later.

**Additional context**
Add any other relevant context for the problem.

graysky2, 2021-11-13:
You should include all logs documenting the error.

mminelli, 2021-11-13:
Edited first message above, and also updated the output of `uname -a`.

graysky2, 2021-11-13:
> Downgrading the kernel to 5.10.76-1 and rebooting fixes the problem.

To make debugging easier:
* 5.10.76-1 was built from https://github.com/raspberrypi/linux/commit/581049d718caf95f5feb00607ac748d5841cf27c
* 5.10.78-2 was built from https://github.com/raspberrypi/linux/commit/063a326cbaff498c7cd001648b400f12c0ec4237

6by9, 2021-11-13:
Can you try editing config.txt, and extend the line ```dtoverlay=vc4-kms-v3d``` to ```dtoverlay=vc4-kms-v3d,composite```?
I think #4689 has had some odd repercussions.

Bob-le-pirate, 2021-11-13:
It works with dtoverlay=vc4-kms-v3d,composite.
Thank you very much.

mminelli, 2021-11-13:
@6by9 Thanks a lot for this: confirmed, it works for me!

Btw, since I am using Kodi, I have the `kodi.config.txt` file that has the dtoverlay config.

So now my file looks like

```
$ cat /boot/kodi.config.txt 
dtoverlay=vc4-kms-v3d,cma-512,composite
dtoverlay=rpivid-v4l2
### leave this next entry blank as ensures dt parameters like dtparam=audio=on
### which could appear after "include kodi.config.txt" in /boot/config.txt are
### applied to the base dt instead of to the last dtoverlay loaded
dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1

### RPi 4B/400 ONLY
### For 4k content @ 60 Hz refresh rate, uncomment hdmi_enable_4kp60=1
#
### If it breaks video output as detailed in the following bug, also uncomment
### force_turbo=1 see: https://github.com/raspberrypi/linux/issues/4446
#hdmi_enable_4kp60=1
#force_turbo=1

### uncomment to use the analog audio jack
### the 2nd option prevents crackling audio when HW accel decoding is active
#dtparam=audio=on
#audio_pwm_mode=1
```
Is this a long-term solution, or will there be a different fix for the kernel?

graysky2, 2021-11-13:
@6by9 - I am wondering about scope of affected models.  I do not see this issue on rpi4.  Can you infer from the code which models will need the `foo,composite` line?

I thinking about a generic solution for the linux-raspberry kernel which ships `config.txt` something like:

```
[all]
dtoverlay=vc4-kms-v3d

[rpi3]
dtoverlay=vc4-kms-v3d,composite
```

pelwell, 2021-11-13:
Unfortunately that isn't going to work. The composite parameter causes a device that is present in the base Device Tree to be enabled by setting its status to okay. Applying the same overlay again without the composite flag won't remove or change the okay.

pelwell, 2021-11-13:
> EDIT: on second though, I really think this should be handled upstream.

6by9's suggestion is just a workaround - there will be a better solution.

graysky2, 2021-11-13:
> Applying the same overlay again without the composite flag won't remove or change the okay.

I edited by above to first unset the overlay.  But I agree with you that the fix should be upstream.

pelwell, 2021-11-13:
> I edited my above to first unset the overlay

That's not what `dtoverlay=` does - its two functions are:
1. To mark the end of the current overlay and its list of parameters (which can span multiple lines using `dtparam= ...`.
2. If the first dt directive in config.txt, the firmware will ignore any HAT overlay.

graysky2, 2021-11-13:
OK, bad idea :grin:

popcornmix, 2021-11-16:
rpi-update kernel has reverted this change, so `,composite` is not required until we find out exactly what is going wrong.

mminelli, 2021-11-16:
I updated to 5.10.79-1 and removed `,composite`, but I still see the issue: no video output to my TV.
Adding back `,composite` "fixed" it. Maybe this revert is in a version not yet released?

mminelli, 2021-11-16:
Sure:
```
$ uname -a
Linux alarmpi 5.10.79-1-raspberrypi-ARCH #1 SMP Mon Nov 15 21:41:38 UTC 2021 armv7l GNU/Linux
```

graysky2, 2021-11-16:
@popcornmix

* [5.10.79-1](https://github.com/archlinuxarm/PKGBUILDs/commit/167c2dc0fb54749307719112a5e8e52f79f3a4c7) used https://github.com/raspberrypi/linux/commit/d6bbe477be2215deef6f9a4c6a7e31c58915cca1
* [5.10.79-2](https://github.com/archlinuxarm/PKGBUILDs/commit/d4c37ffd6c0c6fce49de8dc6c28aca819798c8a2) (just pushed the commit to build it) uses https://github.com/raspberrypi/linux/commit/46c99e3d7cf38446491b3c2a826fc05dcebc588d

popcornmix, 2021-11-16:
@mminelli you need 5.10.79-2 to get the composite fix.

mminelli, 2021-11-16:
OK, got it. Will upgrade as soon as it becomes available.

mminelli, 2021-11-16:
Just installed 5.10.79-2 and I can confirm that video output is fine even without `,composite`.

```
$ uname -a
Linux alarmpi 5.10.79-2-raspberrypi-ARCH #1 SMP Tue Nov 16 20:23:10 UTC 2021 armv7l GNU/Linux
```

Thanks for the quick response!

pelwell, 2022-03-03:
Please close the issue if the matter is resolved.

mminelli, 2022-03-03:
Solved.

closed 2022-03-03 by mminelli