raspberrypi/linux #4694: vc4-kms-v3d: Black screen at 2560x1440 @70/120/144Hz

okaestne, 2021-11-13:
Hi,
I tried all day to get a higher refresh rate on my 1440p monitor running without success. When trying to switch to an higher rate (70/120/144 Hz), the screen goes black. 4k60 and 1080p120 do work on my TV (but not 1440p120).
When choosing 70Hz, the screen seems gets a signal for a short time (<0.5 sec) before it looses it.
I wasn't successful with custom xrandr modes either.

With the FKMS driver xrandr doesn't list the 144 Hz mode, but the 70 Hz mode. Here the same issue (no signal) occurs when choosing 70Hz.

The OS keeps running, but sometimes my whole desktop (Cinnamon, compiled from git) is frozen after restoring the old setting (2560x1440 @60Hz). Mouse cursor can still be moved though. In this case dmesg prints:

```
[ 3349.773176] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ 3360.013298] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ 3370.253424] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ 3380.493549] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:205:plane-25] flip_done timed out
[ 3390.733676] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ 3400.973797] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ 3411.213931] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[ 3421.454051] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:70:plane-3] flip_done timed out
[ 3431.694169] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:205:plane-25] flip_done timed out
```

**System**
Raspberry Pi 4 Model B Rev 1.4 4GB
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, c12b1df4ed6416fb0df33ba1731c5b13c1bdbdf8, stage4
Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

```
$ vcgencmd version
Oct 29 2021 10:47:33 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```

```
config.txt
----------

arm_64bit=1
arm_boost=1
arm_freq=1800
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=550
core_freq_min=220
disable_commandline_tags=2
disable_l2cache=1
disable_overscan=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=3
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=250
h264_freq=550
hdmi_enable_4kp60=1
hdmi_samples_limit=0x1f4000
hevc_freq=550
init_uart_clock=0x2dc6c00
isp_freq=550
lcd_framerate=60
mask_gpu_interrupt0=3072
mask_gpu_interrupt1=25635
max_framebuffer_height=2160
max_framebuffer_width=3840
max_framebuffers=2
[... 121 lines cut ...]
  Audio Data Block:
    Linear PCM:
      Max channels: 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 24 20 16
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
  Detailed Timing Descriptors:
    DTD 2:  2560x1440  143.856 Hz  16:9   221.970 kHz 586.000 MHz (527 mm x 296 mm)
                 Hfront    8 Hsync  32 Hback  40 Hpol P
                 Vfront   25 Vsync   8 Vback  70 Vpol N
    DTD 3:  2560x1440  119.881 Hz  16:9   182.818 kHz 482.640 MHz (527 mm x 296 mm)
                 Hfront    8 Hsync  32 Hback  40 Hpol P
                 Vfront    7 Vsync   8 Vback  70 Vpol N
    DTD 4:  2560x1440   69.923 Hz  16:9   104.045 kHz 274.680 MHz (597 mm x 336 mm)
                 Hfront    8 Hsync  32 Hback  40 Hpol P
                 Vfront   34 Vsync   8 Vback   6 Vpol N
    DTD 5:  1920x1080   60.000 Hz  16:9    67.500 kHz 148.500 MHz (527 mm x 296 mm)
                 Hfront   88 Hsync  44 Hback 148 Hpol P
                 Vfront    4 Vsync   5 Vback  36 Vpol P
    DTD 6:  1920x1080  143.855 Hz  16:9   166.440 kHz 332.880 MHz (527 mm x 296 mm)
                 Hfront    8 Hsync  32 Hback  40 Hpol P
                 Vfront   63 Vsync   8 Vback   6 Vpol N
Checksum: 0x38
```

popcornmix, 2023-01-24:
Just tested this on bullseye.
With `video=HDMI-A-1:2560x1440@70D` in cmdline.txt I get an image and:
```
pi@pi4:~ $ kmsprint 
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (96) 2560x1440@69.99 369.276 2560/200/280/480/- 1440/1/3/55/+ 70 (69.99) U 
      Plane 3 (86) fb-id: 339 (crtcs: 3) 0,0 2560x1440 -> 0,0 2560x1440 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 339 2560x1440 RG16
```
With `video=HDMI-A-1:2560x1440@120D` in cmdline.txt I get an image it rejects the mode with:
```
vc4-drm gpu: [drm] User-defined mode not supported: "2560x1440": 120 661598 2560 2784 3064 3568 1440 1443 1448 1545 0x20 0x6
```
because the pixel clock (661MHz) is higher than we support (600MHz).

So it seems to be working as intended here. Are you seeing the same with a recent bullseye RPiOS image and kms?

okaestne, 2023-01-29:
It really has gotten better. I can now use 2560x1440@70Hz and even 120 Hz without issues! I used the exact timings listed in the EDID table I initially posted, which use the CVT 1.2 reduced blanking timings:

```
$ ./cvt12 2560 1440 120 -b
# 2560x1440 @ 120.000 Hz Reduced Blank (CVT) field rate 120.000 Hz; hsync: 183.000 kHz; pclk: 483.12 MHz
Modeline "2560x1440_120.00_rb2"  483.12  2560 2568 2600 2640  1440 1511 1519 1525 +hsync -vsync
```

This way the pixel clock stays far below HDMI 2.0's 600 MHz limit. But I had no luck getting the 144 Hz mode work, which my monitor SHOULD officially support via HDMI (got no other device to verify). It is just below 600 MHz:

```
$ ./cvt12 2560 1440 143.856 -b
# 2560x1440 @ 143.856 Hz Reduced Blank (CVT) field rate 143.856 Hz; hsync: 221.970 kHz; pclk: 586.00 MHz
Modeline "2560x1440_143.86_rb2"  586.00  2560 2568 2600 2640  1440 1529 1537 1543 +hsync -vsync
```

I do still get freezes though, when *trying* to use the 144 Hz mode that is listed in `xrandr` without manual action. Here some info:

```
$ vcgencmd version
Jan  5 2023 10:46:54 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)
$ uname -a
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

<details>
<summary>dmesg snippet of crash</summary>

```
[  391.010184] [drm:vc4_hdmi_encoder_pre_crtc_configure [vc4]] *ERROR* Failed to set HSM clock rate: -22
[  401.138104] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:94:crtc-3] flip_done timed out
[  411.378525] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  411.378826] vc4-drm gpu: [drm] *ERROR* Timed out waiting for commit
[  421.874945] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  421.875203] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:94:crtc-3] commit wait timed out
[  432.115362] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:94:crtc-3] flip_done timed out
[  442.099793] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  442.100049] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:94:crtc-3] commit wait timed out
[  452.340235] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  452.340490] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] commit wait timed out
[  462.580646] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  462.580896] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:84:plane-3] commit wait timed out
[  472.821085] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  472.821340] vc4-drm gpu: [drm] *ERROR* Timed out waiting for commit
[  472.846456] ------------[ cut here ]------------
[  472.846473] fw-clk-pixel-bvb already disabled
[  472.846512] WARNING: CPU: 3 PID: 592 at drivers/clk/clk.c:1026 clk_core_disable+0x238/0x270
[  472.846542] Modules linked in: rfcomm cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes hid_logitech_hidpp joydev vc4 snd_soc_hdmi_codec cec hid_logitech_dj brcmfmac brcmutil drm_kms_helper v3d snd_soc_core rpivid_hevc(C) bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) gpu_sched bcm2835_mmal_vchiq(C) cfg80211 v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig snd_compress snd_bcm2835(C) snd_pcm_dmaengine videobuf2_memops snd_pcm videobuf2_v4l2 videobuf2_common raspberrypi_hwmon snd_timer videodev rfkill i2c_brcmstb snd vc_sm_cma(C) mc syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio drm i2c_dev fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[  472.846787] CPU: 3 PID: 592 Comm: Xorg Tainted: G         C        5.15.84-v8+ #1613
[  472.846798] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  472.846804] pstate: 800000c5 (Nzcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  472.846814] pc : clk_core_disable+0x238/0x270
[  472.846826] lr : clk_core_disable+0x238/0x270
[  472.846836] sp : ffffffc008edb8c0
[  472.846841] x29: ffffffc008edb8c0 x28: ffffffd7b8024068 x27: ffffff805e8c3c00
[  472.846858] x26: 0000000000000014 x25: 0000000000418958 x24: 0000000000000000
[  472.846875] x23: ffffff8040b86e00 x22: ffffff8043bd9b98 x21: ffffff8043bd9080
[  472.846890] x20: ffffff80403a8500 x19: ffffff80403a8500 x18: 0000000000000000
[  472.846905] x17: 0000000000000000 x16: ffffffd81a2e7a38 x15: 000000558047b750
[  472.846920] x14: 0000000000000000 x13: 64656c6261736964 x12: ffffffd81afb6660
[  472.846935] x11: 0000000000000003 x10: ffffffd81af9e620 x9 : ffffffd819cee9bc
[  472.846950] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
[  472.846965] x5 : ffffffa8dcf97000 x4 : 0000000000000000 x3 : 0000000000000002
[  472.846980] x2 : 0000000000000001 x1 : 30c550063bb2e700 x0 : 0000000000000000
[  472.846994] Call trace:
[  472.847000]  clk_core_disable+0x238/0x270
[  472.847012]  clk_core_disable_lock+0x2c/0x48
[  472.847023]  clk_disable+0x28/0x38
[... 45 lines cut ...]
[  472.848891]  vc4_crtc_atomic_disable+0x9c/0xc0 [vc4]
[  472.848940]  drm_atomic_helper_commit_modeset_disables+0x390/0x430 [drm_kms_helper]
[  472.849025]  vc4_atomic_commit_tail+0x168/0x768 [vc4]
[  472.849074]  commit_tail+0xac/0x1a8 [drm_kms_helper]
[  472.849157]  drm_atomic_helper_commit+0x16c/0x378 [drm_kms_helper]
[... cut ...]
[  472.849956]  el0_svc_common.constprop.3+0xfc/0x120
[  472.849965]  do_el0_svc+0x2c/0x90
[  472.849974]  el0_svc+0x24/0x60
[  472.849982]  el0t_64_sync_handler+0x90/0xb8
[  472.849990]  el0t_64_sync+0x1a0/0x1a4
[  472.849997] ---[ end trace bf51a6baffc08e6b ]---
```

</details>

So overall: good job! I can use 2560x1440@120Hz on my Pi4 :tada: I would still ask you leave this issue open as long as the kernel mod crashes, rather than kindly rejecting unsupported modes.

Thanks for your efforts!