raspberrypi/linux #4695: No signal after Xorg startup

wagnerch, 2021-11-13:
**Describe the bug**
Running latest bullseye image, 

1. See the rainbow screen
2. See the splash screen
3. As soon as Xorg starts, No signal, monitor goes into power saving, takes a long time to respond to systemctl stop lightdm or switch VT

I saw initially in dmesg it was complaining about this:
```
[    8.002258] vc4-drm gpu: [drm] The core clock cannot reach frequencies high enough to support 4k @ 60Hz.
[    8.002282] vc4-drm gpu: [drm] Please change your config.txt file to add hdmi_enable_4kp60.
```

So did that, result is the same.  Next decided to try 5.15 kernel, ran BRANCH=next rpi-update, reboot, result is the same.  Figured had nothing to lose at this point, threw in an arm_64bit=1 in config.txt, and well no change.

Oddly the first time it booted on the fresh bullseye image (previously was on buster with fkms), did see the desktop for a brief second but it had a lot of artifacts.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?
Pi 4B 8G

* Which OS and version (`cat /etc/rpi-issue`)?  
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage4

* Which firmware version (`vcgencmd version`)?
Nov  5 2021 19:46:11
Copyright (c) 2012 Broadcom
version b99321a10f46c9bd662fc87998419c5d6974927a (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.15.1-v8+ #1483 SMP PREEMPT Wed Nov 10 13:00:40 GMT 2021 aarch64 GNU/Linux

**Logs**
dmesg after enabling 4kp60:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 SMP Wed Oct 6 16:41:57 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 1864704 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2059584
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:2560x1080M@60 smsc95xx.macaddr=DC:A6:32:B2:DE:0F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=25cc8f78-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[... 2377 lines cut ...]
			 NV12:  (UNKNOWN MODIFIER) (UNKNOWN MODIFIER) (UNKNOWN MODIFIER) LINEAR
			 NV21:  (UNKNOWN MODIFIER) (UNKNOWN MODIFIER) (UNKNOWN MODIFIER) LINEAR
			 NV16:  LINEAR
			 NV61:  LINEAR
			 P030:  (UNKNOWN MODIFIER)
	219 alpha:
		flags: range
		values: 0 65535
		value: 65535
	220 rotation:
		flags: bitmask
		values: rotate-0=0x1 rotate-180=0x4 reflect-x=0x10 reflect-y=0x20
		value: 1
	221 COLOR_ENCODING:
		flags: enum
		enums: ITU-R BT.601 YCbCr=0 ITU-R BT.709 YCbCr=1 ITU-R BT.2020 YCbCr=2
		value: 0
	222 COLOR_RANGE:
		flags: enum
		enums: YCbCr limited range=0 YCbCr full range=1
		value: 0

Frame buffers:
id	size	pitch
```

wagnerch, 2021-11-13:
Buster was working fine.

wagnerch, 2021-11-14:
OK, this is weird.  Wiped the SD card and re-loaded the same image, had the exact same problem on first boot.  Decided to:
1. remove "quiet splash" from cmdline.txt
2. add enable_uart=1, hdmi_enable_4kp60=1, arm_64bit=1 to config.txt
3. reboot, and now it's totally fine and happy.

In both cases I did an apt-get update && apt-get dist-upgrade (the original and this one).   So I am running with 5.10.63-v8+ kernel at the moment.

I was thinking maybe it was an issue with the splash screen from Plymouth (?), so I went through and add quiet splash back in cmdline and it's still fine.  I can't imagine it's enable_uart=1.

popcornmix, 2021-11-14:
>  I can't imagine it's enable_uart=1.

Test without that. It can have the effect of forcing core clock higher which could potentially be beneficial for hdmi.

wagnerch, 2021-11-14:
@popcornmix Yep, that was it.  Last thing I would have thought, but now it makes sense as I had a similar problem testing 5.15 on LE10 (no issue with the current kernel they push).  It started up with no display, and I enabled uart to see what was going on and it worked fine there too -- but there was other issues on 5.15 with video playback.

Is this one of the known issues?  Any other recommended workaround other than enable_uart=1?

popcornmix, 2021-11-16:
`core_freq_min=500` will probably work too.

wagnerch, 2021-11-20:
Yeah, that seems to work as well.  What's interesting is according to bcmstat.sh the core freq is 550 once X is up.  This is also one of the Pi4's with that newer voltage regulator (which I guess is basically all 8G models?), so the ARM freq max is 1800.  All of my 2 gig models have the older circuit design, which I don't overclock.

https://www.raspberrypi.com/news/bullseye-bonus-1-8ghz-raspberry-pi-4/

wagnerch, 2022-01-15:
Seems to be resolved as of commit e13d4e98076f3698cfd03186159e652f5721a241

closed 2022-01-15 by wagnerch