raspberrypi/linux #4697: Camera Interface problems in 2021-10-30 version

Bart63, 2021-11-14:
**Describe the bug**
Camera interface option disappears inside the "Raspberry Pi Configuration" window in a desktop versions after the first update.
`vcgencmd get_camera` returns `supported=0 detected=0`.
Rebooting does not help.
`libcamera` works though.

**To reproduce**
Run 2021-10-30 version of Raspberry Pi OS with desktop (full also works).
Get a start update from the setup window "Welcome to Raspberry Pi".
Check _Interface_ tab inside of the _Raspberry Pi Configuration_ window.

**Expected behaviour**
Camera interface option is still visible.

**Actual behaviour**
Camera interface option disappears (as shown in an image).

**System**
In the attachments.

**Logs**
In the attachments.

[image]
[raspinfo.txt](https://github.com/raspberrypi/linux/files/7533782/raspinfo.txt)
[dmesg.txt](https://github.com/raspberrypi/linux/files/7533781/dmesg.txt)

finakechi, 2021-11-15:
I am having a similar issue.

libcamera is not working for me though.

I have and RPi3 Model B and the HQ Camera module

finakechi, 2021-11-16:
> I am having a similar issue.
> 
> libcamera is not working for me though.
> 
> I have and RPi3 Model B and the HQ Camera module

As an update, I downgraded to Buster and the issue was resolved.

So it does seem to have something to do with Bullseye.

6by9, 2021-11-16:
libcamera is totally independent of the legacy firmware camera stack. There is no longer a need to have an option to enable the camera, therefore it has been removed from the Raspberry Pi Configuration app.

"vcgencmd get_camera" only reports on cameras the firmware supports, so is irrelevant with Bullseye and libcamera.

If /dev/video0 exists, then the appropriate drivers are loaded for libcamera.

djayor, 2021-11-17:
I have a similar problem, when I do: `vcgencmd get_camera` returns `supported=1 detected=1`, but when I run: `libcamera-hello` though the error:
`[0:02:01.541384500] [3922]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
ERROR: *** no cameras available ***`

@6by9 said that you don't need to enable the camera, but I did it anyway with `sudo raspi-config`.

pelwell, 2021-11-17:
Camera support in the firmware (not libcamera) requires config.txt to include `start_x=1`.

libcamera support requires `camera_auto_detect=1` in config.txt (on current/Bullseye firmware), or `dtoverlay=imx219`, `dtoverlay=imx477` etc. (on older/Buster firmware, but still supported).

Pietvc, 2021-11-18:
Where to find config.txt ?

djayor, 2021-11-18:
> Where to find config.txt ?

You can edit with:
$ sudo nano /boot/config.txt

pelwell, 2021-11-18:
https://www.google.com/search?q=Where+to+find+config.txt+%3F&oq=Where+to+find+config.txt+%3F&aqs=chrome..69i57.1092j0j7&client=ms-android-oneplus&sourceid=chrome-mobile&ie=UTF-8

kalunho17, 2021-11-22:
i am facing the same problem. PiCamera is no longer supported in the new version (bullseye).

[Read Here](https://www.raspberrypi.com/news/bullseye-camera-system/)

i can make the camera run with opencv - however the image is somehow cropped/darker than when i run with libcamera.