raspberrypi/linux #4699: Add CONFIG_R8169 for 2.5 GbE support?

geerlingguy, 2021-11-15:
**Is this the right place for my bug report?**

Yes

**Describe the bug**

The Raspberry Pi Compute Module 4 opens up some new possibilities for networking with a Raspberry Pi. The Pi's exposed PCIe Gen 2 lane is perfect for 2.5 Gbps networking, which is a nice more-than-double speed boost over the built-in 1 Gbps NIC (and it can be used in tandem, for compact router builds).

I have already tested two cards that work with `CONFIG_R8169` enabled, [Rosewill's RC-20001](https://pipci.jeffgeerling.com/cards_network/rosewill-rc20001-25gbe.html) and [Syba's Dual-2.5G card](https://pipci.jeffgeerling.com/cards_network/syba-dual-25gbe-pcie-nic.html). And it seems that Radxa's new [Taco](https://wiki.radxa.com/Taco) board also includes the Realtek 8125b chipset.

I can install the driver manually from Realtek's website, and it works at least on the 64-bit Pi OS build, but it would be convenient, since it is a reliable, inexpensive chipset (already being integrated into some CM4 projects), if the support were available in the base release.

**To reproduce**

Install the default kernel onto a Pi with a Realtek 8125b-based 2.5 Gbps NIC. Try to use the interface. `lspci` will show it, but no kernel module is available for it.

**Expected behaviour**

The NIC should be recognized and available to the system in addition to the built-in Broadcom NIC.

**Actual behaviour**

The NIC is not visible until after you manually install Realtek's driver or recompile the kernel with `CONFIG_R8169`.

**System**

* Compute Module 4
* Which OS and version (`cat /etc/rpi-issue`)? N/A
* Which firmware version (`vcgencmd version`)? N/A
* Which kernel version (`uname -a`)? N/A

**Logs**

`lspci` output:

<details>
<summary>Click to expand output</summary>

```
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. Device 8125 (rev 04)
	Subsystem: Realtek Semiconductor Co., Ltd. Device 8125
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: I/O ports at <unassigned> [disabled]
	Region 2: Memory at 600100000 (64-bit, non-prefetchable) [disabled] [size=64K]
	Region 4: Memory at 600110000 (64-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, MSI 01
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <512ns, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s unlimited, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via message/WAKE#
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [b0] MSI-X: Enable- Count=32 Masked-
		Vector table: BAR=4 offset=00000000
		PBA: BAR=4 offset=00000800
	Capabilities: [d0] Vital Product Data
pcilib: sysfs_read_vpd: read failed: Input/output error
		Not readable
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [148 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [168 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [178 v1] Transaction Processing Hints
		No steering table available
	Capabilities: [204 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [20c v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=150us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [21c v1] Vendor Specific Information: ID=0002 Rev=4 Len=100 <?>
```

</details>

**Additional context**

I would be happy to submit a PR if this change is acceptable.

pelwell, 2021-11-15:
`CONFIG_R8169=m` is already set in the 32- and 64-bit Pi 4 defconfigs, and the module is shipping as part of our standard firmware: https://github.com/raspberrypi/rpi-firmware/blob/master/modules/5.10.78-v7l%2B/kernel/drivers/net/ethernet/realtek/r8169.ko https://github.com/raspberrypi/rpi-firmware/blob/master/modules/5.10.78-v8%2B/kernel/drivers/net/ethernet/realtek/r8169.ko

geerlingguy, 2021-11-15:
Oh, interesting... do you know if that was added more recently? I last tested with a build from a few weeks ago, and I had updated it a few days ago, but it was the Buster image and not the latest Bullseye release.

geerlingguy, 2021-11-15:
I performed a `dist-upgrade`, and after had:

```
pi@bullseye:~ $ uname -a
Linux bullseye 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```

Then I ran `sudo rpi-update` and ended up with:

```
pi@bullseye:~ $ uname -a
Linux bullseye 5.10.78-v8+ #1484 SMP PREEMPT Thu Nov 11 16:37:26 GMT 2021 aarch64 GNU/Linux

pi@bullseye:~ $ lspci -v
...
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)
	Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
	Flags: bus master, fast devsel, latency 0, IRQ 66
	I/O ports at <unassigned> [disabled]
	Memory at 600100000 (64-bit, non-prefetchable) [size=64K]
	Memory at 600110000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: <access denied>
	Kernel driver in use: r8169
	Kernel modules: r8169
```

So... nice! Do you know when this change will trickle down to the stable firmware?

pelwell, 2021-11-15:
Sorry - I can't put my finger on my crystal ball.

geerlingguy, 2021-11-15:
@pelwell - It was worth a try ;)

I'll close this as 'fixed' anyways. That was the easiest feature request ever!

closed 2021-11-15 by geerlingguy