raspberrypi/linux #4701: drm: drm_crtc_legacy_gamma_set() underflow

notcarbide, 2021-11-15:
**Describe the bug**
`blob` is used after freed in `drm_crtc_legacy_gamma_set` due to commit 4286cce

**To reproduce**
1. Boot the device.
2. Any time `drm_crtc_legacy_gamma_set` is used without updating the `gamma_lut` blob, there is an underflow.

**System**
 Branch: rpi-5.15.y

**Logs**
```
[   10.242904] refcount_t: underflow; use-after-free.
[   10.242946] WARNING: CPU: 1 PID: 622 at lib/refcount.c:28 refcount_warn_saturate+0x108/0x150
[   10.242967] Modules linked in: 8021q garp stp llc snd_usb_audio hid_logitech_hidpp snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device rpivid_mem joydev snd_soc_hdmi_codec vc4 cec bcm2835_codec(C) drm_kms_helper bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) brcmfmac snd_soc_core videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops brcmutil videobuf2_v4l2 snd_compress videobuf2_common cfg80211 snd_bcm2835(C) snd_pcm_dmaengine videodev snd_pcm rfkill v3d mc sysimgblt snd_timer gpu_sched vc_sm_cma(C) syscopyarea snd raspberrypi_hwmon i2c_brcmstb sysfillrect uio_pdrv_genirq fb_sys_fops uio nvmem_rmem hid_logitech_dj drm i2c_dev drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[   10.243079] CPU: 1 PID: 622 Comm: Xorg Tainted: G         C        5.15.1-zap-v8+ #10
[   10.243084] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[   10.243088] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   10.243093] pc : refcount_warn_saturate+0x108/0x150
[   10.243098] lr : refcount_warn_saturate+0x108/0x150
[   10.243102] sp : ffffffc012cf3b00
[   10.243105] x29: ffffffc012cf3b00 x28: 0000000000000000 x27: ffffff8107f7c058
[   10.243113] x26: ffffff8102b26000 x25: ffffff8102b26200 x24: ffffff8102b26400
[   10.243120] x23: 0000000000000000 x22: ffffff810336e000 x21: 0000000000000000
[   10.243127] x20: ffffffc0093b1ea0 x19: ffffff8107f7c010 x18: 0000000000000000
[   10.243133] x17: ffffffffffffffff x16: 0000000000000000 x15: 0000000000000004
[   10.243140] x14: 0000000000000fff x13: 0000000000000003 x12: 0000000000000003
[   10.243147] x11: 0000000000000000 x10: 0000000000000027 x9 : d7d40bcdcdda0e00
[   10.243153] x8 : d7d40bcdcdda0e00 x7 : 65646e75203a745f x6 : 746e756f63666572
[   10.243160] x5 : ffffffc01191cc77 x4 : ffffffc0119038ee x3 : 0000000000000000
[   10.243166] x2 : ffffff81fefb2858 x1 : ffffffc012cf38e0 x0 : 0000000000000026
[   10.243174] Call trace:
[   10.243176]  refcount_warn_saturate+0x108/0x150
[   10.243181]  drm_mode_object_put+0xdc/0x100 [drm]
[   10.243286]  drm_crtc_legacy_gamma_set+0x394/0x3d0 [drm]
[   10.243366]  drm_mode_gamma_set_ioctl+0x294/0x490 [drm]
[   10.243446]  drm_ioctl_kernel+0x144/0x220 [drm]
[   10.243525]  drm_ioctl+0x274/0x400 [drm]
[   10.243605]  drm_compat_ioctl+0xf4/0x130 [drm]
[   10.243684]  __arm64_compat_sys_ioctl+0x210/0x320
[   10.243690]  invoke_syscall+0x5c/0x170
[   10.243697]  el0_svc_common+0xb4/0xf0
[   10.243702]  do_el0_svc_compat+0x1c/0x30
[   10.243707]  el0_svc_compat+0x20/0x50
[   10.243712]  el0t_32_sync_handler+0x78/0xc0
[   10.243716]  el0t_32_sync+0x1a4/0x1a8
```

**Additional context**
This could be fixed by removing the extra `drm_property_blob_put(blob)` that was added in the mentioned commit. Although if it is important for that to be there, it might be enough to NULL out `blob` after.

6by9, 2021-11-15:
Duplicates #4676 and that splat fixed with #4682.

pelwell, 2021-11-16:
#4682 hasn't been merged because the gamma issues don't appear to be resolved. I can cherry-pick the fixing patch, or you can submit it as a separate PR for expedited merging.

6by9, 2021-11-16:
I'll drop the gamma change and push a new PR as we may as well merge the other fixes that I'd put on that PR as well.

6by9, 2021-11-16:
Changes merged to 5.10.
5.15 should be fixed soon.

notcarbide, 2021-11-19:
Fix with 95bb7ac2e698875a41e62d5cf571b20aee994227

Thanks!

closed 2021-11-19 by notcarbide