raspberrypi/linux #4711: sc16is7xx transmit errors (tty buffer?) and spi (tpm-slb9670) issue on all versions with kernel above 5.4.83-v7+

paulenuta, 2021-11-19:
**Describe the bug**
-Still adding details-
Last working kernel version was `5.4.83-v7+`.
On `5.10.17-v7+` I've also tested with rebuild kernel and sc16is7xx driver from `5.4.83-v7+` with no succes.
In our program (not linux-serial-test) we got randomly 4096 byte in buffer and this was the reason to link the issue to the kernel tty.

**To reproduce**
Raspberry Pi 3 Model B+
SD image: 2021-10-30-raspios-bullseye-armhf-lite.zip
Custom board witn 2 x SC16IS752 and tpm-slb9670 (@pelwell has one)
Set /boot/config.txt:
```
[all]
dtoverlay=gpio-poweroff,gpiopin=27,active_low
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=i2c-rtc,mcp7941x
dtoverlay=sc16is752-i2c-02
dtoverlay=sc16is752-i2c-01
dtoverlay=tpm-slb9670
enable_uart=1
dtoverlay=gpio-fan,gpiopin=12,temp=50000
#gpu_mem=96
dtoverlay=disable-bt
force_turbo=1
#initramfs initrd followkernel
display_auto_detect=1
[EDID=*]
ignore_lcd=1
display_default_lcd=0
hdmi_group=2
hdmi_mode=18
[EDID=ELO-ET1509L-8UWA]
hdmi_group=2
hdmi_mode=16
```
Set /boot/cmdline.txt:
```
console=serial0,115200 console=tty1 root=PARTUUID=e94b1ac0-02 rootfstype=ext4 fsck.repair=yes rootwait
```

Update system, enable I2C, SPI and autologin to console:
```
sudo apt update && sudo apt install -y git &&\
sudo raspi-config nonint do_i2c 0 &&\
sudo raspi-config nonint do_spi 0 &&\
SUDO_USER="${FIRST_USER_NAME}" sudo raspi-config nonint do_boot_behaviour B2 &&\
sudo poweroff
```

List serial ports available to check that sc16is752 is loaded: `ls /dev/ttyS*`
Clone and build linux-serial-test:
```
git clone --depth 1 https://github.com/cbrake/linux-serial-test &&\
cd linux-serial-test &&\
gcc -o linux-serial-test linux-serial-test.c
```

**Expected behaviour**
All data sent (Tx) should be received (Rx) using a loopback cable or just conecting Rx to Tx of same port.

**Actual behaviour**
Some date is not transmitted so not received.
[image]
[image]

Using an ttyUSB0 is OK:
[image]

**System**
[... 699 lines cut ...]
[   10.178170] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

* Which model of Raspberry Pi? >> Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Oct 29 2021 10:49:08
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
`Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`
[image]

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.
[sc16is752-i2c-01-02.zip](https://github.com/raspberrypi/linux/files/7568315/sc16is752-i2c-01-02.zip)

paulenuta, 2022-01-13:
**ping**

danergo, 2022-01-18:
I'm also having this. But I have another detail to add:

My trial is perfectly working on RPi1 (CM): 

`Linux rpi1 5.10.63+ #1457 Tue Sep 28 11:24:51 BST 2021 armv6l GNU/Linux`

And it never works on RPi3 (CM):

`Linux rpi3 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`

As you can see, I do have a 5.10.y kernel on RPi1 which works perfectly. So it might be kernel related, and it might be also related to architecture, or might be a 32/64-bit issue.

danergo, 2022-01-29:
ping

paulenuta, 2022-02-17:
After `sudo rpi_update` I got `uname -a: Linux raspberrypi 5.15.23-v7+ #1525 SMP Wed Feb 16 14:34:22 GMT 2022 armv7l GNU/Linux`.

Running the test `linux-serial-test/linux-serial-test -s -e -p /dev/ttySC0 -b 9600 -A -T -o 120 -i 125` give me no error:

```
[4711.log](https://github.com/raspberrypi/linux/files/8086597/4711.log)

/dev/ttySC0: count for this session: rx=115683, tx=115683, rx err=0
/dev/ttySC0: TIOCGICOUNT: ret=0, rx=20401, tx=219630, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
```
Full log attached.

Thank you @pelwell !

closed 2022-02-17 by paulenuta