raspberrypi/linux #4714: Running libcamera-hello leads to Division by zero in kernel

oniongarlic, 2021-11-19:
**Describe the bug**
Running libcamera-hello on a Model B with Camera V1 leads to Division by zero in kernel oops.

**To reproduce**
Run libcamera-hello 

**Expected behaviour**
No Division by zero in kernel

**Actual behaviour**
[   95.199368] Division by zero in kernel.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Vintage Model B
* Which OS and version (`cat /etc/rpi-issue`)? 11
* Which firmware version (`vcgencmd version`)?
Oct 29 2021 10:49:08 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux pifhdcamera 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux

**Logs**
[   95.199368] Division by zero in kernel.
[   95.199425] CPU: 0 PID: 494 Comm: libcamera-hello Tainted: G         C        5.10.63+ #1459
[   95.199442] Hardware name: BCM2835
[   95.199455] Backtrace: 
[   95.199542] [<c08746c0>] (dump_backtrace) from [<c0874a40>] (show_stack+0x20/0x24)
[   95.199571]  r7:c20b8020 r6:bf3bc280 r5:c20b9178 r4:c0a70ae0
[   95.199606] [<c0874a20>] (show_stack) from [<c0877cb8>] (dump_stack+0x28/0x30)
[   95.199636] [<c0877c90>] (dump_stack) from [<c0874840>] (__div0+0x20/0x28)
[   95.199657]  r5:c20b9178 r4:c28004f8
[   95.199692] [<c0874820>] (__div0) from [<c04f5afc>] (Ldiv0+0x8/0x10)
[   95.199768] [<bf3b7000>] (setup_mmal_port_format [bcm2835_isp]) from [<bf3b82d4>] (bcm2835_isp_node_s_fmt+0x148/0x32c [bcm2835_isp])
[   95.199790]  r5:c2521400 r4:c20b9178
[   95.200347] [<bf3b818c>] (bcm2835_isp_node_s_fmt [bcm2835_isp]) from [<bf1d93e8>] (v4l_s_fmt+0x42c/0x43c [videodev])
[   95.200384]  r8:c27d4b40 r7:bf3ba894 r6:c20b9188 r5:c26b7e40 r4:c2521400
[   95.201016] [<bf1d8fbc>] (v4l_s_fmt [videodev]) from [<bf1d9dd8>] (__video_do_ioctl+0x234/0x464 [videodev])
[   95.201048]  r9:bf3ba894 r8:c20b94d0 r7:00000003 r6:bf1d8fbc r5:c0cc5605 r4:c20b9188
[   95.201635] [<bf1d9ba4>] (__video_do_ioctl [videodev]) from [<bf1da80c>] (video_usercopy+0x118/0x558 [videodev])
[   95.201668]  r10:c2521400 r9:c0bb3028 r8:c2521400 r7:00000000 r6:00000000 r5:c0cc5605
[   95.201684]  r4:c0cc5605
[   95.202266] [<bf1da6f4>] (video_usercopy [videodev]) from [<bf1dac68>] (video_ioctl2+0x1c/0x24 [videodev])
[   95.202415]  r10:00000013 r9:b54c83cc r8:c0bb3028 r7:00000000 r6:c26b7e40 r5:c26b7e41
[   95.202437]  r4:bf1dac4c
[   95.203050] [<bf1dac4c>] (video_ioctl2 [videodev]) from [<bf1d2148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[   95.203636] [<bf1d20fc>] (v4l2_ioctl [videodev]) from [<c020b678>] (sys_ioctl+0x1ac/0x88c)
[   95.203670]  r5:c26b7e41 r4:c0cc5605
[   95.203704] [<c020b4cc>] (sys_ioctl) from [<c0008240>] (ret_fast_syscall+0x0/0x28)
[   95.203722] Exception stack(0xc26dbfa8 to 0xc26dbff0)
[   95.203751] bfa0:                   b54c8690 b4b15068 00000013 c0cc5605 b54c83cc 00000000
[   95.203777] bfc0: b54c8690 b4b15068 b4b15068 00000036 b4b13e10 b54c8690 012425b8 b54c8658
[   95.203795] bfe0: b6f4aff4 b54c83b4 b6eeebf4 b6b9a1ac
[   95.203822]  r10:00000000 r9:c26da000 r8:c00083e4 r7:00000036 r6:b4b15068 r5:b4b15068
[   95.203838]  r4:b54c8690
[   95.349114] Division by zero in kernel.
[   95.349170] CPU: 0 PID: 494 Comm: libcamera-hello Tainted: G         C        5.10.63+ #1459
[   95.349188] Hardware name: BCM2835
[   95.349203] Backtrace: 
[   95.349290] [<c08746c0>] (dump_backtrace) from [<c0874a40>] (show_stack+0x20/0x24)
[   95.349319]  r7:bf3bc280 r6:c0bb3028 r5:c20b9178 r4:c0a70ae0
[   95.349355] [<c0874a20>] (show_stack) from [<c0877cb8>] (dump_stack+0x28/0x30)
[   95.349387] [<c0877c90>] (dump_stack) from [<c0874840>] (__div0+0x20/0x28)
[   95.349408]  r5:c20b9178 r4:c28004f8
[   95.349446] [<c0874820>] (__div0) from [<c04f5afc>] (Ldiv0+0x8/0x10)
[   95.349521] [<bf3b7000>] (setup_mmal_port_format [bcm2835_isp]) from [<bf3b7bb8>] (bcm2835_isp_node_queue_setup+0x90/0x1ec [bcm2835_isp])
[   95.349543]  r5:c20b8020 r4:c20b9178
[... 88 lines cut ...]
[   95.705864] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   95.742315] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   95.782113] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   95.821145] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   95.883581] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   95.930017] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   95.981672] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.033632] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.067788] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.107873] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.143137] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.178712] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.212013] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.247264] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.292490] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.339998] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000
[   96.389237] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8bd10000

**Additional context**
1080p forced in config.txt, otherwise defaults
framebuffer_width=1920
framebuffer_height=1080
hdmi_group=1
hdmi_mode=16
hdmi_drive=2

popcornmix, 2021-11-23:
@naushir looks like `q_data->fmt->depth` must be zero to trigger a division by zero exception.

naushir, 2021-11-23:
@davidplowman have you seen this before?  Not something I've encountered.

I can't really see how q_data->fmt->depth would be zero in this case either.

naushir, 2021-11-23:
@oniongarlic is this easily reproducible?

Could you add the following to the end of ``/boot/cmdline.txt``:
```
bcm2735_isp.debug=5
```
and provide a kernel log after the failure please?

groman, 2021-11-25:
I'm observing something similar on Pi Zero W when trying to use an OV5647 module:

```
pi@pi-zero-webcam:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 Wed Oct 6 16:40:27 BST 2021
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] OF: reserved mem: failed to allocate memory for node 'linux,cma'
[    0.000000] cma: Reserved 8 MiB at 0x0e800000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=B8:27:EB:20:CE:A9 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6858b1ed-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,usb_f_rndis,usb_f_acm,usb_f_uvc bcm2835_isp.debug=5
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 222392K/262144K available (8683K kernel code, 1324K rwdata, 2824K rodata, 420K init, 837K bss, 31560K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 30717 entries in 60 pages
[    0.000000] ftrace: allocated 60 pages with 4 groups
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x324/0x560 with crng_init=1
[    0.000024] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000115] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000201] bcm2835: system timer (irq = 27)
[    0.000888] Console: colour dummy device 80x30
[    0.001521] printk: console [tty1] enabled
[    0.001620] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060369] pid_max: default: 32768 minimum: 301
[    0.060671] LSM: Security Framework initializing
[    0.060995] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.061063] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.063101] cgroup: Disabling memory control group subsystem
[    0.063542] CPU: Testing write buffer coherency: ok
[    0.065398] Setting up static identity map for 0x8200 - 0x8238
[    0.066634] devtmpfs: initialized
[    0.079554] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.079941] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.080022] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.081448] pinctrl core: initialized pinctrl subsystem
[    0.083302] NET: Registered protocol family 16
[    0.088020] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.094030] audit: initializing netlink subsys (disabled)
[    0.095540] thermal_sys: Registered thermal governor 'step_wise'
[    0.096557] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.096647] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.096935] Serial: AMBA PL011 UART driver
[    0.100725] audit: type=2000 audit(0.090:1): state=initialized audit_enabled=0 res=1
[    0.130977] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.150836] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-10-29T10:49:08, variant start
[    0.160849] raspberrypi-firmware soc:firmware: Firmware hash is b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13
[    0.210094] Kprobes globally optimized
[    0.219224] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[    0.222455] SCSI subsystem initialized
[    0.222917] usbcore: registered new interface driver usbfs
[    0.223057] usbcore: registered new interface driver hub
[... 369 lines cut ...]
[  126.882768]  r10:00000015 r9:b10bc38c r8:c0bb3028 r7:00000000 r6:c36c5540 r5:c36c5541
[  126.882778]  r4:bf320c4c
[  126.883170] [<bf320c4c>] (video_ioctl2 [videodev]) from [<bf318148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[  126.883385] [<bf3180fc>] (v4l2_ioctl [videodev]) from [<c020b678>] (sys_ioctl+0x1ac/0x88c)
[  126.883401]  r5:c36c5541 r4:c0cc5605
[... cut ...]
[  126.919406] bcm2835-isp bcm2835-isp: bcm2835_isp_buf_init: vb 133317b6
[  126.919475] cma: cma_alloc: alloc failed, req-size: 352 pages, ret: -12
[  126.919503] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 1441792 failed
[  126.919521] bcm2835-isp bcm2835-isp: setup_mmal_port: setup capture[0]
[  126.927477] bcm2835-isp bcm2835-isp: bcm2835_isp_node_queue_setup: Image size 1440000, nbuffers 2 for node capture[0]
pi@pi-zero-webcam:~ $
```

That's the kernel log with `bcm2835_isp.debug=5`

naushir, 2021-11-25:
I think https://github.com/raspberrypi/linux/pull/4732 might fix this div by 0.  Fairly obvious bug when you look at it, so I am somewhat surprised we have not seen it happen more.

renekliment, 2022-02-04:
I can confirm this has been fixed for me. (Raspberry Pi Zero W)
Thank you!

popcornmix, 2022-02-04:
@oniongarlic can you close issue if this is fixed for you?