raspberrypi/linux #4717: X11 performance 'slow' when using FKMS or KMS

mike632t, 2021-11-19:
**Describe the bug**
I've been working on an application that uses just X11 the primitives (lines and rectangles) to draw the desired display in the application window.

I've tested it successfully on several different platforms including a old netbook (Intel Atom + Intel 945M graphics), a Raspberry Pi 3 Model B using the HDMI output , the official touch screen and the Adafruit 3.5" touchscreen, and a Raspberry Pi 4 using a single HDMI output with FKMS and KMS disabled.

Typical calls use XDrawLine() and XFillRectangle().

XDrawLine(h_display, x_application_window, DefaultGC(h_display, i_screen), i_left , i_upper, i_left, i_height);
XFillRectangle(h_display, x_application_window, DefaultGC(h_display, i_screen), i_left , i_upper, i_width, i_length);

Each time I update the application window I'm drawing approximately 550+ shapes and I'm updating the display about 10 times a second.

However, on the Pi 4 when I enable FKMS (or KMS) then not all the primitives are drawn when the window is updated (even if the window is only updated once every 5 seconds).

**To reproduce**
Enable FKMS or KMS on a Pi 4 and compile and run the code [here](https://github.com/mike632t/x11-calc)

**Expected behaviour**
Application windows contents does not flicker continuously 

**Actual behaviour**
Application windows contents flickers continuously and does not appear to drawn properly between updates

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 4

* Which OS and version (`cat /etc/rpi-issue`)?
Raspbian 10 (minimal install /etc/rpi-issue not present)

* Which firmware version (`vcgencmd version`)?
N/A (minimal install vcgencmd not installed)

* Which kernel version (`uname -a`)?
Linux unknown 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

**Additional context**
Add any other relevant context for the problem.

mike632t, 2023-12-04:
Still slow when using bookworm but can't disable KMS anymore?

6by9, 2023-12-04:
You can disable KMS (remove `dtoverlay=vc4-kms-v3d` from /boot/firmware/config.txt, and add `framebuffer_depth=32` as your application needs it), but it's not the default configuration. Wayfire requires a DRM backend to run.

Running x11-calc-10 as the first binary built.
On a Pi5 reverted to X11 (not Wayland), with or without KMS, you're spinning 1 core at 100%. No obvious flickering, and nothing apparently wrong with the display.

Running on a Pi4 is the same, with or without KMS.

Reverted commits 
```
6ecc806 (HEAD -> stable) Revert "Don't draw display segment background on ARM"
ab3a39f Revert "Don't draw display segment background on ARM"
f80b4ef Revert "Changed the way the code tests to see if it is being built on a Raspberry Pi."
68b35b1 Revert "Changed the way the code tests to see if it is being built on a Raspberry Pi (again)."
```
to go back to the full app, and it appears to still be the same. No tearing or apparently missing primitives.

Drawing a complete UI using the basic X primitives is going to be CPU intensive as each call will be going out to the X server to complete, and none of it will be double-buffered so it can tear. It feels like there must be a better way to achieve the desired effect.
I suspect the framebuffer is also in uncached memory too as it needs to be read out by the hardware. All read-modify-write accesses will therefore be quite slow, and worse if they are done on a pixel-by-pixel basis rather than decent sized bursts.

mike632t, 2023-12-25:
I've had to modify the code to try to work around the problem (basically if compiling on ARM the display is drawn using a single line for each segment).  If you want to test the full code I can create a separate branch...