raspberrypi/linux #4719: 7" HDMI Display does NOT work with KMS driver

rnayabed, 2021-11-20:
**Is this the right place for my bug report?**
Should be, this is related to KMS/FKMS driver.

**Describe the bug**
Raspberry Pi OS Bullseye ships with KMS driver by default now. It has been a mixed bag for me, since none of the HDMI touch screens that I have from Waveshare (7", 5") work with it. I do get a rainbow screen and initial kernel log messages, but after some time it blanks and I get nothing on the screen.

I have not yet tried desktop mode, since i only use the lite package.

**To reproduce**
Just fresh install Raspberry Pi OS 32-bit Bullseye on an SD Card (preferably the lite version), try using an external 3rd party display like the ones from Waveshare and it's clones.

**Expected behaviour**
It should work fine. Using `vc4-fkms-v3d` works just fine. This problem can be blamed to only the `vc4-kms-v3d` overlay.

**Actual behaviour**
The System starts up. Rainbow Screen is showed, some basic kernel logs are shown. Then screen blanks and stays that way forever.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 
Pi 4B. But should be reproducible with other Pi as well

* Which OS and version (`cat /etc/rpi-issue`)? 
```
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Oct 29 2021 10:47:33 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux pi4b 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```
**Logs**
If applicable, add the relevant output from `dmesg` or similar.
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 SMP Wed Oct 6 16:41:57 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037efffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 491264
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 294656 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 489536
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1024x600M@60 smsc95xx.macaddr=DC:A6:32:CF:48:9B vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyS0,115200 console=tty1 root=PARTUUID=5e63546e-02 rootfstype=ext4 fsck.repair=yes rootwait
[... 406 lines cut ...]
[    8.679532] 8021q: 802.1Q VLAN Support v1.8
[    8.681418] uart-pl011 fe201000.serial: no DMA platform data
[    8.954910] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k FS
[    9.168253] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.450809] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    9.451472] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.515245] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   11.515298] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.315897] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   15.818400] Bluetooth: Core ver 2.22
[   15.818456] NET: Registered protocol family 31
[   15.818462] Bluetooth: HCI device and connection manager initialized
[   15.818476] Bluetooth: HCI socket layer initialized
[   15.818486] Bluetooth: L2CAP socket layer initialized
[   15.818501] Bluetooth: SCO socket layer initialized
[   15.832345] Bluetooth: HCI UART driver ver 2.3
[   15.832356] Bluetooth: HCI UART protocol H4 registered
[   15.832413] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.832578] Bluetooth: HCI UART protocol Broadcom registered
[   16.023526] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.023536] Bluetooth: BNEP filters: protocol multicast
[   16.023550] Bluetooth: BNEP socket layer initialized
[   16.036667] NET: Registered protocol family 38
[   16.064975] cryptd: max_cpu_qlen set to 1000
```

rnayabed, 2021-11-20:
```
[    7.427523] vc4-drm gpu: [drm] Cannot find any crtc or sizes
```

6by9, 2021-11-20:
Not finding any displays implies the HDMI monitor does incorrect things via hotplug and/or EDID reading.

Bullseye should have ```modetest``` available via apt. I don't recall which package it is in, or if it is installed by default, but if not installed then ```apt-cache search modetest``` should tell you the relevant package.
Run ```modetest -M vc4``` and post the output (you can skip the planes section as it is effectively static information that is repeated multiple times for each plane).

The key thing is likely to be knowing the connected status of the HDMI interfaces. You should also be able to read it via ```cat /sys/class/drm/card1-HDMI-1/status``` (may be card0, and you'll also have HDMI-2). If it reads ```disconnected``` then the panel isn't setting hotplug correctly.

rnayabed, 2021-11-20:
@6by9 Sorry if i sound dumb, but why is it that the KMS driver doesnt work without correct EDID? FKMS driver and legacy driver seem to just work fine ? Just before the KMS driver is even loaded, I can see the rainbow screen and kernel logs

rnayabed, 2021-11-20:
> Bullseye should have modetest available via apt. I don't recall which package it is in, or if it is installed by default, but if not installed then apt-cache search modetest should tell you the relevant package.
Run modetest -M vc4 and post the output (you can skip the planes section as it is effectively static information that is repeated multiple times for each plane).

`apt-cache search modetest` returns nothing.

> The key thing is likely to be knowing the connected status of the HDMI interfaces. You should also be able to read it via cat /sys/class/drm/card1-HDMI-1/status (may be card0, and you'll also have HDMI-2). If it reads disconnected then the panel isn't setting hotplug correctly.

Sadly, that seems to be the issue. I am getting `disconnected` all the time.

rnayabed, 2021-11-20:
Hmm, with FKMS driver i get `connected` 🤔🤔

6by9, 2021-11-20:
FKMS doesn't support hotplug - https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c#L1303 will always return connected. What is connected at boot is what you get until you reboot.

KMS supports hotplug properly, so changing monitors will result in it reconfiguring correctly. It does require the device to follow the HDMI standard for hotplug though, which it sounds like your display doesn't.

You can try forcing the output to be enabled at a particular resolution by adding an entry ```video=HDMI-1:1280x720M@60D``` to /boot/cmdline.txt (do not add any carriage returns to that file). As documented in https://github.com/torvalds/linux/blob/master/Documentation/fb/modedb.rst, the 'D' on the end forces the output to be enabled and use a digital output. Replace the resolution and refresh rate with the correct values for your display.

6by9, 2021-11-20:
Under FKMS you could run ```tvservice -M``` to monitor the events as you connect and disconnect the display. I suspect it won't change.

rnayabed, 2021-11-20:
OH MY GOD IT WORKED. I LOVE YOU! I SERIOUSLY LOVE YOU. YOU ARE THE BEST. IF I WAS RICH I WOULD DONATE EVERYTHING TO YU. THANK YOU !!!!

rnayabed, 2021-11-20:
ADDING `video=HDMI-A-1:1024x600M@60D` fixed itt

6by9, 2021-11-20:
We are aware that shifting to the standard Linux approaches is heavily reliant on devices following the relevant specifications. There are many workaround hacks that have been added to the firmware over the years for devices that misbehave, and those are effectively being discarded. It looks like your displays rely on one of those hacks, which probably means it will misbehave on almost any other HDMI device that uses Linux.

(IIRC the firmware tries reading the EDID from the display even if no hotplug signal, and views the display as connected if it gets a valid EDID back).

rnayabed, 2021-11-20:
@6by9 is there a way we can ask the Pi to not look for any EDID/hotplugging and just still connect it, so that we do not have to put any cmdline addition like that ? 

FKMS already works that way, it doesnt rely on EDID as you said. Would it be possible to do the same for KMS?

6by9, 2021-11-20:
You have that way of getting the Pi to not look at hotplug by modifying /boot/cmdline.txt.

The KMS driver did also do an EDID probe as well as checking hotplug, however there are significant issues with that approach when HDMI2.0 and scrambling (required for >4k30) are involved, as hotplug being deasserted (regardless of whether an EDID can still be read) requires a renegotiation/reenabling of scrambling. Hence that check was removed.

AnttiNannimus, 2021-11-26:
This was also a problem for me using the RPi-branded 7" Touch Screen and Bullseye on an RPi 3B+. This solution worked for me:

Issue can be resolved as a work around with the following line placed in /boot/config.txt

dtparam=i2c_vc_baudrate=50000

From: https://raspberrypi.stackexchange.com/questions/133490/blank-touch-screen-after-upgrade-to-bullseye

Good luck!

6by9, 2021-11-26:
> This was also a problem for me using the RPi-branded 7" Touch Screen and Bullseye on an RPi 3B+. This solution worked for me:

Totally unrelated to this issue about HDMI displays, and covered by https://github.com/raspberrypi/linux/issues/4686.

rnayabed originally posted on that thread, but because it was unrelated to the DSI display was requested to start a new issue, which is this one!

rnayabed, 2023-06-10:
> ADDING video=HDMI-A-1:1024x600M@60D fixed it

almost 2 years later, this method no longer works.
I get this additional error now:
```
[    7.126949] vc4-drm gpu: [drm] User-defined mode not supported: "1024x600": 60 49000 1024 1029 1042 1312 600 602 605 622 0x68 0xa
```

@6by9 

```
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage4
pi@raspberrypi:~ $ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@raspberrypi:~ $ 
```

rnayabed, 2023-06-10:
dmesg:

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=HDMI-A-1:1024x600M@60 smsc95xx.macaddr=DC:A6:32:CF:48:9B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=75402b78-02 rootfstype=ext4 fsck.repair=yes video=HDMI-A-1:1024x600M@60D rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 1365148K/2019328K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 129892K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39744 entries in 156 pages
[    0.000000] ftrace: allocated 156 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000317] Console: colour dummy device 80x25
[    0.000913] printk: console [tty1] enabled
[    0.000982] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.001022] pid_max: default: 32768 minimum: 301
[    0.001165] LSM: Security Framework initializing
[    0.001373] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001428] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.002731] cgroup: Disabling memory control group subsystem
[    0.004993] cblist_init_generic: Setting adjustable number of callback queues.
[... 380 lines cut ...]
[    9.659299] input: Logitech Wireless Keyboard PID:4023 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:046D:C534.0002/0003:046D:4023.0004/input/input19
[    9.660218] logitech-hidpp-device 0003:046D:4023.0004: input,hidraw3: USB HID v1.11 Keyboard [Logitech Wireless Keyboard PID:4023] on usb-0000:01:00.0-1.1/input1:1
[... cut ...]
[   15.405609] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.405630] Bluetooth: BNEP filters: protocol multicast
[   15.405646] Bluetooth: BNEP socket layer initialized
[   15.412462] Bluetooth: MGMT ver 1.22
[   15.422055] NET: Registered PF_ALG protocol family
[   15.498387] Bluetooth: RFCOMM TTY layer initialized
[   15.498417] Bluetooth: RFCOMM socket layer initialized
[   15.498441] Bluetooth: RFCOMM ver 1.11
[   20.370690] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

rnayabed, 2023-06-10:
`cat /sys/class/drm/card1-HDMI-A-1/status` shows `connected` now but I get blank screen. I am using KMS driver.

6by9, 2023-06-12:
From your timing
```"1024x600": 60 49000 1024 1029 1042 1312```
ie 1024 active pixels,
1029-1024 = 5 pixels as the horizontal front porch
1032-1029 = 3 pixels as the horizontal sync period
1312-1032 = 280 pixels as the horizontal back porch.

https://www.raspberrypi.com/documentation/computers/config_txt.html#raspberry-pi-4-hdmi-pipeline
> Running everything at 2 pixels per clock means that the Raspberry Pi 4 can not support a timing where any of the horizontal timings are not divisible by 2. The firmware and Linux kernel will filter out any mode that does not fulfill this criteria.

Since commit https://github.com/raspberrypi/linux/commit/57fb32e632be4d406b4594829e3befdae1100c12 in Nov 2020 that requirement has been enforced by the driver.

Sorry this is largely down to the display being non-compliant with the HDMI specs.
Does it have an EDID? Does ```sudo modprobe i2c-dev``` ```sudo i2c-detect -y -r 20``` return address 0x50 responding?
If so, then ```i2cdump -y 20 0x50 i``` should dump it out and allow some analysis to be done. You can also tell the kernel to use a file based on that instead of trying to read it from the display.

pelwell, 2023-06-16:
The output of `i2cdetect -l` would be useful.

6by9, 2023-06-16:
```sudo modprobe i2c-dev``` may be required to expose the I2C buses to userspace.

> Note that I was using FKMS driver on a Pi 0 W while writing this.

What's the relevance of this?

rnayabed, 2023-06-16:
> `sudo modprobe i2c-dev` may be required to expose the I2C buses to userspace.
> 
> > Note that I was using FKMS driver on a Pi 0 W while writing this.
> 
> What's the relevance of this?

I just thought it would be better if I specify the model just to be sure :)

rnayabed, 2023-06-16:
I did run `sudo modprobe i2c-dev`.

Running `i2cdetect -l` returns nothing.

6by9, 2023-06-16:
> > `sudo modprobe i2c-dev` may be required to expose the I2C buses to userspace.
> > > Note that I was using FKMS driver on a Pi 0 W while writing this.
> > 
> > What's the relevance of this?
> 
> I just thought it would be better if I specify the model just to be sure :)

??!?!

This is your issue with the subject "7" HDMI Display does NOT work with KMS driver" and first post saying
> Using vc4-fkms-v3d works just fine. This problem can be blamed to only the vc4-kms-v3d overlay

You then post information saying you're running FKMS. So how much of your recent posts is related to FKMS vs that related to KMS?

FKMS is considered deprecated, and no further work will be done on it.

rnayabed, 2023-06-16:
There has been a misunderstanding. I meant to say that the results are the same even in fkms mode. (no output by i2c programs). running the commands that you have given show no output regardless of fkms or kms mode. 

Sorry, wont use fkms in this discussion anymore until asked to! Only kms!