raspberrypi/linux #4720: RPi3 /sys/class/thermal/thermal_zone0/temp not working for armv7

Bob-le-pirate, 2021-11-20:
Dear all,

I find this bug on kernel linux-raspberrypi4 for armv7 that seems to be solved for arm64 (look at #4077).

It seems to be related only to linux-raspberrypi4 (works well with linux-raspberrypi)

`# uname -a `
`Linux raspberrypi3 5.10.79-2-raspberrypi4-ARCH #1 SMP Tue Nov 16 20:28:12 UTC 2021 armv7l GNU/Linux`

`# cat /sys/class/thermal/thermal_zone0/temp`
`cat: /sys/class/thermal/thermal_zone0/temp: Argument invalide`

Best regards

pelwell, 2021-11-20:
You talk of linux-raspberrypi4 in your text but the command log shows linux-raspberrypi3 - what are these devices and what software are they running?

Bob-le-pirate, 2021-11-20:
I use archlinux arm and the kernel is linux-raspberrypi4 for the PI 3 as it was indicated here : [https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4](url).

`cat /proc/cpuinfo | grep Model`
`Model		: Raspberry Pi 3 Model B Rev 1.2`

pelwell, 2021-11-22:
This is working for me on a rev 1.2 3B using our kernel build (from `bcm2709_defconfig`):
```
pi@raspberrypi:/sys/class/thermal/thermal_zone0$ cd
pi@raspberrypi:~$ cat /sys/class/thermal/thermal_zone0/temp
46698
pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.10.79-v7+ #1487 SMP Tue Nov 16 12:24:24 GMT 2021 armv7l GNU/Linux
```
@Dark-Sky May have some insight into the difference with the Arch kernel.

Dark-Sky, 2021-11-22:
> 

Although most of our programs are from arch-arm I build my kernels always starting with RPi's _defconfig's as a base and add what people want for extra modules. Looking at arch-arms's config/config8 it looks like they need to enable CONFIG_BCM2835_THERMAL=y in their config.

[image]

Bob-le-pirate, 2021-11-23:
After updating to last kernel 5.10.81-1-raspberrypi4-ARCH no evolution : /sys/class/thermal/thermal_zone0/temp still not working.
Looking in [linux-raspberrypi4/config8](https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4/files/config8), CONFIG_BCM2835_THERMAL=y seems to be enabled but it doesn't work.

Dark-Sky, 2021-11-23:
Your OP says you are using the linux-raspberrypi4 for armv7 if that is still true you need to be looking in this config: https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4/files/config

You need to inform them to enable this if you are still using linux-raspberrypi4 for armv7.

Bob-le-pirate, 2021-11-23:
Thank you Dark-Sky, you're right I'm using 32 bits kernel on my RPI3. 
@graysky2, can you update linux-raspberrypi4/config to modify 
```
- # CONFIG_BCM2835_THERMAL is not set 
+ CONFIG_BCM2835_THERMAL=y
```

If necessary I can write a pull request.

pelwell, 2021-11-24:
It sounds like this problem is understood so I'm ducking out. Please close the issue when you are ready.

graysky2, 2021-11-24:
@Bob-le-pirate - Please try 5.10.81-2 and report back.  Might be faster if you build it yourself, the builders are working on a queue.

Bob-le-pirate, 2021-11-25:
Thank you @graysky2,
I try to build (first time for me), but I have an error with file 0001-Make-proc-cpuinfo-consistent-on-arm64-and-arm.patch
I write the content of the [file](https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4/files/0001-Make-proc-cpuinfo-consistent-on-arm64-and-arm.patch).
May be I made a mistake ?

graysky2, 2021-11-25:
Just clone the repo or download the zip snapshot.  The build of the kernel was successful.  Is your mirror up to date?

Bob-le-pirate, 2021-11-25:
Mirror is not up to date, please can you give me the address where I can download the snapshot ?

Bob-le-pirate, 2021-11-27:
Thank you @graysky2, last kernel core/linux-rpi 5.10.81-3 solve the problem

closed 2021-11-27 by Bob-le-pirate