raspberrypi/linux #4722: /dev/sda: Can't open blockdev  while booting from SD card

kraj, 2021-11-20:
I am building 
```
Linux raspberrypi4-64 5.10.78-v8 #1 SMP PREEMPT Mon Nov 8 18:19:44 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

and seeing the following in `dmesg` output 

```
[    3.057573] Run /init as init process
[    3.061304]   with arguments:
[    3.061314]     /init
[    3.061323]   with environment:
[    3.061332]     HOME=/
[    3.061341]     TERM=linux
[    3.105654] mmc0: new ultra high speed DDR50 SDHC card at address aaaa
[    3.105802] mmc1: new high speed SDIO card at address 0001
[    3.113252] mmcblk0: mmc0:aaaa SL32G 29.7 GiB
[    3.124602]  mmcblk0: p1 p2 p3
[    3.134619] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops)
[    7.134212] /dev/sda: Can't open blockdev
[    7.142259] /dev/sda1: Can't open blockdev
[    7.235588] EXT4-fs (mmcblk0p3): mounted filesystem without journal. Opts: (null)
[    7.319051] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    7.393575] EXT4-fs (mmcblk0p3): mounted filesystem without journal. Opts: (null)
[    7.707346] systemd[1]: System time before build time, advancing clock.

```

I do not have any USB disk attached to my board and I also tried after disconnecting my serial connection but this message keeps coming all the time

```
[    7.134212] /dev/sda: Can't open blockdev
[    7.142259] /dev/sda1: Can't open blockdev
```

I am using yocto project with gcc 11.2

Let me know if you require more details

pelwell, 2021-11-22:
Look in your cmdline.txt and /etc/fstab.

pelwell, 2021-11-22:
Post/pastebin your entire dmesg.

kraj, 2021-11-22:
> Post/pastebin your entire dmesg.

here you go [1]
[1] http://sprunge.us/RwPBHK

pelwell, 2021-11-22:
There's nothing obvious in there. I think you need Yocto-specific help, and this isn't the place for that.

kraj, 2021-11-22:
> There's nothing obvious in there. I think you need Yocto-specific help, and this isn't the place for that.

so you think its being initiated by some usersapce process like systemd ?

Here is my `/etc/fstab`
```
# stock fstab - you probably want to override this with a machine specific one

/dev/root            /                    auto       defaults              1  1
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,ptmxmode=0666,gid=5      0  0
tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0

# uncomment this if your device has a SD/MMC/Transflash slot
#/dev/mmcblk0p1       /media/card          auto       defaults,sync,noauto  0  0

/dev/mmcblkp3	/data	ext4	defaults	0	0
```

and /proc/cmdline is 
```
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:01:12:B5 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
```

pelwell, 2021-11-22:
> so you think its being initiated by some usersapce process like systemd ?

It must be. The kernel doesn't keep a record of the file systems (where could it, in the general case?), and it can't be auto-probing a device that doesn't exist.

kraj, 2021-11-22:
> > so you think its being initiated by some usersapce process like systemd ?
> 
> It must be. The kernel doesn't keep a record of the file systems (where could it, in the general case?), and it can't be auto-probing a device that doesn't exist.

alright. I will try strac'ing a bit here and see what I find.

kraj, 2021-11-22:
closing it here.

closed 2021-11-22 by kraj