raspberrypi/linux #4725: Kernels 5.15.2 & 5.10.79: Screen flickering back when using KMS

Dark-Sky, 2021-11-21:
I had posted this issue a few days ago on this same issue https://github.com/raspberrypi/linux/issues/4660 and never received a response so I have no clue if it is being worked on or if it was even read so starting a new issue as this involves newer kernels. Until this is resolved KMS with V3D rendering is almost to the point of being unusable.

Others that have the same issue:

https://archlinuxarm.org/forum/viewtopic.php?f=65&t=15629

pelwell, 2022-03-02:
Is this issue still needed, or has it been eclipsed by updates to #4660?

Dark-Sky, 2022-03-02:
It is fine right now but I am worried about if changes are made it could come back again as this started the issue time after time trying to get this working:

https://github.com/raspberrypi/firmware/issues/1274#issuecomment-1051020968

closed 2024-12-12 by Dark-Sky