raspberrypi/linux #4729: drm/vc4: vc4_dsi change breaks a panel driver

penk, 2021-11-23:
We found the following vc4_dsi commit (partially) breaks our panel driver: 
https://github.com/raspberrypi/linux/commit/a2b8b34d78d9721196b0010b00d0db5f4392aa5b 

We have parent_rate: `3000000091`, and pll_clock: `425148000`; 
before the commit, calculated divider is `8`, and after that it's calculated `7`, resulting a wrong adjusted_mode. 

This is how the panel looks like when encountered this issue: 
https://i.imgur.com/IYz5qa5.jpeg

The panel is a 8" ILI9881C 1280x800 4-Lane DSI display, here's the drm_display_mode from driver, in case: 

    static const struct drm_display_mode default_mode = {
        .clock = 70858,
        .hdisplay = 800,
        .hsync_start = 800 + 45,
        .hsync_end = 800 + 45 + 45,
        .htotal = 800 + 45 + 45 + 4,
        .vdisplay = 1280,
        .vsync_start = 1280 + 10,
        .vsync_end = 1280 + 10 + 25,
        .vtotal = 1280 + 10 + 25 + 6,
        .flags = 0,
        .width_mm = 107,
        .height_mm = 172,
    };

Any suggestion would be appreciated, thanks.

6by9, 2021-11-23:
I have a load of patches in progress for DSI, and the divider calcs is in one of them.

The original code was incorrect, however the maximum value for the divider is also wrong - it can go up to 255 IIRC. There's little point in allowing it up to that value as an 11.7MHz link frequency is very unlikely, and going round that loop so many time is a little time consuming. Increasing it to 32 should solve your problem.

ILI9881C does claim to support DSI burst mode (which is what is being used here), but implementations do differ.

penk, 2021-11-23:
@6by9 thank you for the prompt reply, we will modify the loop value for the time being, 
and keep a close eye on the future changes, hope it'll progress well and land soon! :-)

6by9, 2021-11-23:
Just looking at the pending patch I have, I did increase the limit up to 255.
```
commit f1a6e05fab5347643cc2cab16246d20e6cd42e91
Author: Dave Stevenson <dave.stevenson@raspberrypi.com>
Date:   Fri Oct 22 16:48:50 2021 +0100

    drm/vc4: dsi: Correct max divider to 255 (not 7)
    
    The integer divider from parent PLL to DSI clock is capable
    of going up to /255, not just /7 that the driver was trying.
    This allows for slower link frequencies on the DSI bus where
    the resolution permits.
    
    Signed-off-by: Dave Stevenson <dave.stevenson@raspberrypi.com>

diff --git a/drivers/gpu/drm/vc4/vc4_dsi.c b/drivers/gpu/drm/vc4/vc4_dsi.c
index dafb0aa7eaa3..8e2c8788d202 100644
--- a/drivers/gpu/drm/vc4/vc4_dsi.c
+++ b/drivers/gpu/drm/vc4/vc4_dsi.c
@@ -1091,7 +1091,7 @@ vc4_dsi_fixup_clock(struct vc4_dsi *dsi, unsigned long pixel_clock_hz)
        /* Find what divider gets us a faster clock than the requested
         * pixel clock.
         */
-       for (divider = 1; divider < 7; divider++) {
+       for (divider = 1; divider < 255; divider++) {
                if (parent_rate / (divider + 1) < pll_clock)
                        break;
        }
```

6by9, 2021-11-23:
If you have a working panel driver and overlay, and the panel is vaguely commercially available, then I'm happy for it to be merged into our tree. It gives more examples for others as to how DSI works.

penk, 2021-11-23:
@6by9 That'll be awesome! 

Overlay: https://github.com/cutiepi-io/cutiepi-drivers/blob/master/Display/panel-nwe080.dts 
Driver: https://github.com/cutiepi-io/cutiepi-drivers/blob/master/Display/drivers/gpu/drm/panel/panel-nwe080.c 

The panel model is called `TXW800018S0-LSS` and can be found [here](http://www.txwlcd.cn/en/h-pd-437.html#fai_14_top&pfc=&_pp=0_35)

6by9, 2021-11-23:
If it's based on ILI9881, then extending the existing https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c driver would be nicest.

Your overlay has a few oddities
```compatible = "brcm,bcm2835``` - it doesn't need a reference to 2711.
Don't set ```power-domains``` from overlays - they're in the base DT.
You define
``` 
               vdio-gpios = <&gpio 21 0>;
                vci-gpios = <&gpio 16 0>;
                bst-gpios = <&gpio 6 1>;
```
but the driver doesn't appear to look for them.
There is no node ```vc4_dsi```.

Typically you'd combine the display, backlight, and touch into one overlay.
 
The backlight can be controlled via the [gpio-backlight module](https://elixir.bootlin.com/linux/latest/source/drivers/video/backlight/gpio_backlight.c) rather than forcing it high in config.txt. Add a line
```backlight = <&gpio_backlight>;``` to your panel definition, and the framework will control it automatically as the display is enabled/disabled.
GPIO 12 does support PWM, so you could use [pwm_bl](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/video/backlight/pwm_bl.c) with it.  

The driver says STMicroelectronics is the copyright holder, and several ST employees are the authors. Is this really the case, or is that copy/paste from the driver you've modified? If it is based on an existing driver, state that in the headers.
We are being more careful with attribution, so could you add a Signed-of-by line to your patches too?

Fix those few items and create a Pull Request, and I don't see much reason not to merge it.

penk, 2021-11-23:
@6by9 Thanks for the comment, I'll check with the developer and prepare a PR accordingly.

penk, 2021-11-24:
@6by9 Here's the patch, [Add-panel-driver-NWE080-and-overlay-for-CutiePi.patch.txt](https://github.com/raspberrypi/linux/files/7597784/Add-panel-driver-NWE080-and-overlay-for-CutiePi.patch.txt)
Please let me know if it's an acceptable format or needs any revision / correction. 

Changelog 
- Combine display, touch, and backlight into one overlay 
- Use pwm_bl for backlight as suggested 
- Clean up copyright / author line in the driver 

I've got two related questions: 
- Where in overlay should I add the "backlight = " line (if needed)? 
- After checking the divider calculation loop (original issue), I found that since `(3000000091 / 8) < 425148000`, the calculated divider unfortunately is still `7`. Is this something we should change in the driver itself? 

Thanks in advance.

6by9, 2021-11-24:
Creating a pull request by following https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request is the easiest way for us to review patches and merge them.

If it was previously using /8, then I believe you will have ended up with the mode
```
    .clock = 62500,
    .hdisplay = 800,
    .hsync_start = 856,
    .hsync_end = 901,
    .htotal = 788,
    .vdisplay = 1280,
    .vsync_start = 1280 + 10,
    .vsync_end = 1280 + 10 + 25,
    .vtotal = 1280 + 10 + 25 + 6,
```
Which is invalid as htotal is less than hdisplay. If you could log it to confirm then that would be useful. Otherwise I'll try to hook it up to our DSI analyser tomorrow.

> Where in overlay should I add the "backlight = " line (if needed)?

It goes in the panel node, eg https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/vc4-kms-dsi-7inch-overlay.dts#L54, so 
```
+            display1: panel@0 {
+                compatible = "nwe,nwe080";
+                reg=<0>;
+                backlight = <&rpi_backlight>;
+                reset-gpios = <&gpio 20 0>;
+                port {
+                    panel_dsi_in1: endpoint {
+                        remote-endpoint = <&dsi1_out_port>;
+                    };
+                };

As you're using i2c6, the compatible string at the start of the overlay needs to be ```"brcm,bcm2711";``` so it'll only load on a Pi4/CM4. I'll review the rest later.

penk, 2021-11-27:
@6by9 Thank you for the wait, I've created the PR here: https://github.com/raspberrypi/linux/pull/4742 

As for the divider issue, it looks like it was the `hsync_start` and `hsync_end` causing the issue; clock and htotal made no difference. 

- when divider is `7` 
        adjusted_mode->hsync_end: `897`
        adjusted_mode->hsync_start: `852`
- when divider is `8`
        adjusted_mode->hsync_end: `791`
        adjusted_mode->hsync_start: `746`

Any suggestion would be appreciated.

6by9, 2021-11-27:
Having hsync_end and hsync_start less than hdisplay has to be wrong. 
The pixel valve is programmed at https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_crtc.c#L355 with 
```
mode->htotal - mode->hsync_end
mode->hsync_end - mode->hsync_start
mode->hsync_start - mode->hdisplay
mode->hdisplay
```
mode->hsync_start - mode->hdisplay is going to be negative, except that the variables are unsigned values so will wrap to something weird. The other values are likely to end up OK as they've all been modified by the same amount.
Your output frame rate is going to be different from the one you're expecting.

I haven't got a Pi running to be able to check the exact paths, but /sys/kernel/debug/dri/0 (or possibly dri/1) should contain a load of files crtcN_regs. I believe it is crtc1_regs which feeds DSI, so could you dump that out. The values of PV_HORZA and PV_HORZB in each case would be of interest. 
PV_HORZA top 16bits are back porch, and bottom 16 bits are sync width
PV_HORZB top 16bits are front porch, and bottom 16 bits are active width (should be 800 or 0x0320)

penk, 2021-11-28:
Here's the result: (log attached: [debug-dri0.zip](https://github.com/raspberrypi/linux/files/7612833/debug-dri0.zip) )

working

    PV_HORZA = 0x006e002d
    PV_HORZB = 0xffca0320

failed 

    PV_HORZA = 0x0004002d
    PV_HORZB = 0x00340320

What do you make of it?

6by9, 2021-11-28:
That decodes as
HACTIVE 800
HBP 4
HSYNC 45
HFP 65482

Total H duration of 66331 clocks. With a vertical sizing of 1321 lines, and a resulting 62500kHz pixel clock, I make that 0.71fps. I'd need to get the analyser on it to confirm that.

6by9, 2021-12-09:
I've had the analyser on this.

Prior to the change it was running at 375Msps/lane, or 187.5MHz.
My logging from vc4_dsi_encoder_mode_fixup gives
```
[    7.427136] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: desired pll_clock 425148000, parent_rate 3000000091
[    7.427226] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: New pll_clock 375000011, pixel_clock 62500001
[    7.427314] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: htimings were 70858 845/890/894
[    7.427558] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: htimings now 62500 739/784/788
```
Which should mean that HFP is negative, and crtc1_regs reports
```
      PV_HORZA = 0x0004002d
      PV_HORZB = 0xffc30320
```
which is nearly as you had.

The analyser reads the horizontal timing as
```
375Msps/lane
45.2Hz
16.75us line time
Hsync 44 pixels 0.79us
HBP 0 pixels 0.06us
Hactive 800 pixels 13us
HFP 0 pixels 2.93 us
HTotal 844
```
I'd guess that the HFP field in PV_HORZB may only be 8 active bits (not 16), so the 0xFFC3 gets truncated to 0xC3, or 195. 13us / 800  * 195 = 3.16us, so nearly as measured at 2.93us.

Anyway, the panel isn't running at the frame rate you thought it was - 45.2Hz instead of 60.
With the patch it is running at the correct frame rate, but it seems that your panel doesn't like the increased DSI rate.

Looking at https://www.phoenixdisplay.com/wp-content/uploads/2019/05/ILI9881C-ILITEK.pdf, Page 207 5.4.14 registers B6h-B7h select the number of lanes, and is configured as per table 2 on page 37. As you're running in 4 lane mode, the LANSEL pin has to be low. Clearing MIPI_LANE_SEL appears to therefore drop you from 4 to 3 lanes by removing D3N/P and not affecting where the other lanes map. Switching to 3 lanes may make the display happier. My logging says
```
[    7.598987] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: desired pll_clock 566864000, parent_rate 3000000091
[    7.599081] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: New pll_clock 600000018, pixel_clock 75000002
[    7.599170] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: htimings were 70858 845/890/894
[    7.599254] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* vc4_dsi_encoder_mode_fixup: htimings now 75000 897/942/946
```
so a smaller bump up from 70.858MHz to 75MHz pixel clock (600Msps/lane, or 300MHz).
With that the analyser reports
60.1Hz
Line time 12.60us
HSync 41 pixels 0.61us
HBP 0 pixels 0.05us
HActive 800 pixels 11.97us
HFP 0 pixels 0us
Htotal 841 pixels 12.60us.

penk, 2021-12-20:
@6by9 Thank you very much for the help and pinpointed to the problem. 

As a good side effect of the upstreaming task, this issue has been fixed with extending ili9881c driver and newly calculated clock value in https://github.com/raspberrypi/linux/pull/4742

I'll close this one.

closed 2021-12-20 by penk