raspberrypi/linux #4730: Unable to handle kernel paging request at virtual address 64662f72

septatrix, 2021-11-23:
**Describe the bug**
A kernel oops happens. Afterwards the rpi is unresponsive, hangs on operations like `lsusb` and soon afterwards even SSH access is no longer possible.

**To reproduce**
Sadly no easy way to reproduce however it occurs somewhat regularly (ca. once a week)

**Expected behaviour**
The kernel oops should not happen

**Actual behaviour**
The kernel oops happened

**System**

* Which model of Raspberry Pi? Pi4B+ 8GB
* Which OS and version? `Raspberry Pi reference 2021-05-07 dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2`, Raspbian GNU/Linux 10 (buster)
* Which firmware version? `778b6a4f3c7d8d48bb63c02c47bcfbac79417bea (clean) (release) (start_x)`
* Which kernel version? `Linux <hostname> 5.10.63-v7l+ #1457 SMP Tue Sep 28 11:26:14 BST 2021 armv7l GNU/Linux`

**Logs**
<details>
<summary>dmesg output with kernel oops</summary>

```
[279800.385401] usbip-host 1-1.2: stub up
[279800.449989] usbip-host 1-1.2: usb_set_interface done: inf 0 alt 0
[279801.523799] usbip-host 1-1.1: stub up
[279821.313386] usbip-host 1-1.1: urb completion with non-zero status -71
[279821.314679] usbip-host 1-1.1: unlinked by a call to usb_unlink_urb()
[279821.392012] usbip-host 1-1.1: USB disconnect, device number 69
[279821.392060] ------------[ cut here ]------------
[279821.392093] WARNING: CPU: 0 PID: 14781 at fs/sysfs/group.c:281 sysfs_remove_group+0xa8/0xac
[279821.392108] sysfs group 'power' not found for kobject 'ep_03'
[279821.392122] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc ftdi_sio usbserial vc4 cec v3d drm_kms_helper gpu_sched drm bcm2835_codec(C) v4l2_mem2mem drm_panel_orientation_quirks snd_soc_core snd_compress bcm2835_v4l2(C) bcm2835_isp(C) snd_pcm_dmaengine bcm2835_mmal_vchiq(C) raspberrypi_hwmon videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) videodev snd_pcm mc snd_timer vc_sm_cma(C) snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem nvmem_rmem uio_pdrv_genirq uio usbip_host usbip_core ip_tables x_tables ipv6
[279821.392762] CPU: 0 PID: 14781 Comm: kworker/0:1 Tainted: G         C        5.10.63-v7l+ #1457
[279821.392771] Hardware name: BCM2711
[279821.392793] Workqueue: usb_hub_wq hub_event
[279821.392808] Backtrace: 
[279821.392833] [<c0b835a4>] (dump_backtrace) from [<c0b83938>] (show_stack+0x20/0x24)
[279821.392847]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6b3c
[279821.392863] [<c0b83918>] (show_stack) from [<c0b87d24>] (dump_stack+0xcc/0xf8)
[279821.392881] [<c0b87c58>] (dump_stack) from [<c0220b10>] (__warn+0xfc/0x114)
[279821.392895]  r10:00000001 r9:00000009 r8:c04f811c r7:00000119 r6:00000009 r5:c04f811c
[279821.392905]  r4:c0e48434 r3:c1205094
[279821.392919] [<c0220a14>] (__warn) from [<c0b840dc>] (warn_slowpath_fmt+0xa4/0xd8)
[279821.392930]  r7:00000119 r6:c0e48434 r5:c1205048 r4:c0e484dc
[279821.392946] [<c0b8403c>] (warn_slowpath_fmt) from [<c04f811c>] (sysfs_remove_group+0xa8/0xac)
[279821.392959]  r9:c2399000 r8:c3321820 r7:c1205048 r6:c424e608 r5:c0c4d9e4 r4:00000000
[279821.392978] [<c04f8074>] (sysfs_remove_group) from [<c0859698>] (dpm_sysfs_remove+0x58/0x6c)
[279821.392989]  r6:c3ffe988 r5:c424e64c r4:c424e608
[279821.393004] [<c0859640>] (dpm_sysfs_remove) from [<c0849964>] (device_del+0xc0/0x420)
[279821.393014]  r5:c424e64c r4:c424e608
[279821.393026] [<c08498a4>] (device_del) from [<c0849ce0>] (device_unregister+0x1c/0x30)
[279821.393038]  r10:00000001 r9:c2399000 r8:c3319600 r7:c3321800 r6:c3ffe988 r5:00000002
[279821.393047]  r4:c424e608
[279821.393061] [<c0849cc4>] (device_unregister) from [<c0904944>] (usb_remove_ep_devs+0x2c/0x38)
[279821.393070]  r5:00000002 r4:c4cc2fb8
[279821.393085] [<c0904918>] (usb_remove_ep_devs) from [<c08fc430>] (remove_intf_ep_devs+0x50/0x6c)
[279821.393094]  r5:00000002 r4:00000070
[279821.393109] [<c08fc3e0>] (remove_intf_ep_devs) from [<c08fd0cc>] (usb_disable_device+0x90/0x134)
[279821.393120]  r7:c3321800 r6:00000000 r5:00000002 r4:c3f66000
[279821.393136] [<c08fd03c>] (usb_disable_device) from [<c08f36e0>] (usb_disconnect+0xcc/0x248)
[279821.393148]  r9:c2399000 r8:c3319600 r7:c3f66078 r6:c3f660bc r5:c2399000 r4:c3f66000
[279821.393165] [<c08f3614>] (usb_disconnect) from [<c08f5864>] (hub_event+0xbc4/0x1308)
[279821.393177]  r10:00000001 r9:c2398f00 r8:00000100 r7:00000001 r6:c22e2000 r5:c2399000
[279821.393186]  r4:00000000
[279821.393204] [<c08f4ca0>] (hub_event) from [<c023db68>] (process_one_work+0x250/0x5a0)
[279821.393216]  r10:00000000 r9:00000000 r8:00000000 r7:efef7800 r6:efef26c0 r5:c4d70080
[279821.393224]  r4:c2398f00
[279821.393239] [<c023d918>] (process_one_work) from [<c023df18>] (worker_thread+0x60/0x5c4)
[... 79 lines cut ...]
[279821.891162]  r10:efef26c0 r9:c1203d00 r8:efef26d8 r7:00000008 r6:efef26c0 r5:c4d70094
[279821.899089]  r4:c4d70080
[279821.901720] [<c023deb8>] (worker_thread) from [<c02459f8>] (kthread+0x170/0x174)
[279821.909216]  r10:c43e5e74 r9:c4d70080 r8:c023deb8 r7:c4232000 r6:00000000 r5:c3ec3680
[279821.917143]  r4:c358bd40
[279821.919774] [<c0245888>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[279821.927093] Exception stack(0xc4233fb0 to 0xc4233ff8)
[279821.932240] 3fa0:                                     00000000 00000000 00000000 00000000
[279821.940517] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[279821.948794] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[279821.955506]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0245888
[279821.963439]  r4:c3ec3680
[279821.966072] Code: e1a07000 e1a06001 e3c44001 e1a00004 (e5945010) 
[279821.972820] ---[ end trace 4cc37ef73f243bea ]---
[279821.980930] usbip-host 1-1.1: submit_urb error, -19
[279821.986367] usbip-host 1-1.1: lock for reset
[281110.483949] usbip-host 1-1.2: recv a header, -110
[281110.584378] usbip-host 1-1.2: reset full-speed USB device number 68 using xhci_hcd
[281110.723035] usbip-host 1-1.2: device reset
```

</details>

**Additional context**
I included a few kernel messages before and after the oops as it seems to be related to USB devices.

pelwell, 2021-11-23:
The `stub up` message is coming from `usbip_sockfd_store`:
```
/*
 * usbip_sockfd gets a socket descriptor of an established TCP connection that
 * is used to transfer usbip requests by kernel threads. -1 is a magic number
 * by which usbip connection is finished.
 */
static ssize_t usbip_sockfd_store(struct device *dev, struct device_attribute *attr,
			    const char *buf, size_t count)
{
	struct stub_device *sdev = dev_get_drvdata(dev);
...
		dev_info(dev, "stub up\n");
```
A few questions:
1. What hardware do you have attached?
2. What is typically running at the time of the failures?
3. How is the system powered (including USB device, hubs etc.)?

septatrix, 2021-11-23:
> A few questions:
> 
> 1. What hardware do you have attached?

There is a JTAGICE3 and a USB-to-UART converter attached. Additionally also the camera module. Full output of `lsusb`:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 03eb:2140 Atmel Corp. AVR JTAGICE3 (v3.x) Debugger and Programmer
Bus 001 Device 125: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

> 2. What is typically running at the time of the failures?

The pi is headless, the services which are running are only the `usbip` daemon and a mjpeg server via which the camera stream is accessible. Full output of `pstree`;

```
systemd─┬─2*[agetty]
        ├─alsactl
        ├─avahi-daemon───avahi-daemon
        ├─cron
        ├─dbus-daemon
        ├─dhcpcd
        ├─mjpeg-server.sh─┬─raspivid───7*[{raspivid}]
        │                 └─raspivid_mjpeg_───4*[{raspivid_mjpeg_}]
        ├─nginx───4*[nginx]
        ├─rngd───3*[{rngd}]
        ├─rsyslogd───3*[{rsyslogd}]
        ├─sshd───sshd───sshd───bash───pstree
        ├─systemd───(sd-pam)
        ├─systemd-journal
        ├─systemd-logind
        ├─systemd-timesyn───{systemd-timesyn}
        ├─systemd-udevd
        ├─thd
        ├─usbip-rebind.sh───sleep
        ├─usbipd
        └─wpa_supplicant
```

> 3. How is the system powered (including USB device, hubs etc.)?

The Pi itself is powered with the official PSU and the USB devices are directly attached without any intermediate hubs.

oltdaniel, 2021-11-25:
### Null derefence issue
I can report an additional kernel error, that does not throw an `Unable to handle kernel paging request at virtual address X` but throws an even more worrying ops, with `Unable to handle kernel NULL pointer dereference at virtual address X`.

The setup is identical to the one mentioned in the initial issue. These kinds of errors occurred more often with non official PSU units. After we've switched, everything worked fine for 2-3 weeks, until they started to occur again.  

<details>
<summary>dmesg output</summary>
<p>

```
[1821791.396604] usbip-host 1-1.1: urb completion with non-zero status -71
[1821791.397634] usbip-host 1-1.1: unlinked by a call to usb_unlink_urb()
[1821791.473943] usbip-host 1-1.1: USB disconnect, device number 47
[1821791.474666] ------------[ cut here ]------------
[1821791.474701] WARNING: CPU: 3 PID: 28016 at fs/sysfs/group.c:281 sysfs_remove_group+0xa8/0xac
[1821791.474717] sysfs group 'power' not found for kobject '1-1.1:1.1'
[1821791.474730] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc vc4 cec drm_kms_helper v3d ftdi_sio gpu_sched usbserial raspberrypi_hwmon drm bcm2835_v4l2(C) snd_bcm2835(C) videobuf2_vmalloc bcm2835_codec(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 drm_panel_orientation_quirks videobuf2_common snd_soc_core videodev snd_compress snd_pcm_dmaengine mc snd_pcm vc_sm_cma(C) snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem nvmem_rmem uio_pdrv_genirq uio usbip_host usbip_core ip_tables x_tables ipv6
[1821791.475369] CPU: 3 PID: 28016 Comm: stub_rx Tainted: G         C        5.10.63-v7l+ #1457
[1821791.475378] Hardware name: BCM2711
[1821791.475387] Backtrace: 
[1821791.475415] [<c0b835a4>] (dump_backtrace) from [<c0b83938>] (show_stack+0x20/0x24)
[1821791.475429]  r7:ffffffff r6:00000000 r5:60000113 r4:c12e6b3c
[1821791.475445] [<c0b83918>] (show_stack) from [<c0b87d24>] (dump_stack+0xcc/0xf8)
[1821791.475462] [<c0b87c58>] (dump_stack) from [<c0220b10>] (__warn+0xfc/0x114)
[1821791.475476]  r10:c4d09e00 r9:00000009 r8:c04f811c r7:00000119 r6:00000009 r5:c04f811c
[1821791.475486]  r4:c0e48434 r3:c1205094
[1821791.475500] [<c0220a14>] (__warn) from [<c0b840dc>] (warn_slowpath_fmt+0xa4/0xd8)
[1821791.475511]  r7:00000119 r6:c0e48434 r5:c1205048 r4:c0e484dc
[1821791.475527] [<c0b8403c>] (warn_slowpath_fmt) from [<c04f811c>] (sysfs_remove_group+0xa8/0xac)
[1821791.475540]  r9:c1912380 r8:c3f40078 r7:c1205048 r6:c4d09220 r5:c0c4d9e4 r4:00000000
[1821791.475559] [<c04f8074>] (sysfs_remove_group) from [<c0859698>] (dpm_sysfs_remove+0x58/0x6c)
[1821791.475570]  r6:00000001 r5:c4d09264 r4:c4d09220
[1821791.475585] [<c0859640>] (dpm_sysfs_remove) from [<c0849964>] (device_del+0xc0/0x420)
[1821791.475595]  r5:c4d09264 r4:c4d09220
[1821791.475611] [<c08498a4>] (device_del) from [<c08fd0d4>] (usb_disable_device+0x98/0x134)
[1821791.475623]  r10:c4d09e00 r9:c1912380 r8:c34b9900 r7:c4d09200 r6:00000001 r5:00000002
[1821791.475631]  r4:c3f40000
[1821791.475645] [<c08fd03c>] (usb_disable_device) from [<c08fd9f8>] (usb_set_configuration+0x2c8/0xa78)
[1821791.475658]  r9:c1912380 r8:c34b9900 r7:00000002 r6:00000002 r5:c3f40000 r4:00000002
[1821791.475690] [<c08fd730>] (usb_set_configuration) from [<bf085a98>] (stub_rx_loop+0x910/0xb90 [usbip_host])
[1821791.475702]  r10:00000001 r9:80002f00 r8:c34b9900 r7:00000000 r6:00000000 r5:00000001
[1821791.475711]  r4:c2c74980
[1821791.475735] [<bf085188>] (stub_rx_loop [usbip_host]) from [<c02459f8>] (kthread+0x170/0x174)
[1821791.475748]  r10:c3d91d9c r9:c34b9904 r8:bf085188 r7:c4ebc000 r6:00000000 r5:c4d29b40
[1821791.475756]  r4:c4cc4640
[1821791.475770] [<c0245888>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[1821791.475780] Exception stack(0xc4ebdfb0 to 0xc4ebdff8)
[1821791.475791] dfa0:                                     00000000 00000000 00000000 00000000
[1821791.475803] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[1821791.475814] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[1821791.475827]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0245888
[1821791.475836]  r4:c4d29b40
[1821791.475847] ---[ end trace 4cc37ef73f243be9 ]---
[1821791.475865] 8<--- cut here ---
[1821791.479194] Unable to handle kernel NULL pointer dereference at virtual address 00000010
[1821791.487513] pgd = 5739a3cc
[1821791.490440] [00000010] *pgd=80000000004003, *pmd=00000000
[1821791.496058] Internal error: Oops: 207 [#1] SMP ARM
[1821791.501030] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc vc4 cec drm_kms_helper v3d ftdi_sio gpu_sched usbserial raspberrypi_hwmon drm bcm2835_v4l2(C) snd_bcm2835(C) videobuf2_vmalloc bcm2835_codec(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 drm_panel_orientation_quirks videobuf2_common snd_soc_core videodev snd_compress snd_pcm_dmaengine mc snd_pcm vc_sm_cma(C) snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem nvmem_rmem uio_pdrv_genirq uio usbip_host usbip_core ip_tables x_tables ipv6
[1821791.554864] CPU: 3 PID: 28016 Comm: stub_rx Tainted: G        WC        5.10.63-v7l+ #1457
[1821791.563306] Hardware name: BCM2711
[1821791.566888] PC is at klist_put+0x20/0xa4
[1821791.570986] LR is at klist_del+0x14/0x18
[1821791.575082] pc : [<c07721d0>]    lr : [<c0772268>]    psr: a0000113
[... cut ...] rd, which is externally powered. Both are connected to the USB 3.0 root hub of the Raspberry PI.

This means, there should not be any relevant power draw on these ports, which could cause any of these mentioned problems. At least in theory, which we cannot prove due to the issue, that we aren't able to reproduce this reliably.

However, I think we can all agree, that these kind of things should not cause any kernel errors resulting the Raspberry PI becoming unresponsive or killing the USB chain.

oltdaniel, 2021-11-25:
Something else I would like to add is, the use of the `usbip` driver. Every new device is exposed via a udev rules to the as an usbip device. Which means, these devices are managed by the usbip driver not the regular usb driver directly. If this could be the origin of these issues as well, we should consider including/contacting the [maintainers of the usbip kernel stuff](https://www.kernel.org/doc/html/latest/process/maintainers.html#usb-over-ip-driver) directly.

pelwell, 2021-11-25:
Without pointing a finger at anyone, `usbip` is a factor that distinguishes your use case from most others.

oltdaniel, 2021-11-25:
After collecting the syslogs of all 26 machines, searching for `Internal error: Oops:` I was able to identify, that both errors are related to the usbip disconnect event for the specific device Atmel ICE. Which could mean, the usbip implementation or handling of an disconnect Event is faulty.

septatrix, 2021-11-25:
However it could also be that the `usbip` messages are just a symptom of the problem. Without further investigation I would not directly jump to the conclusion that usbip is necessarily at fault here. It could also simply be that the issue would occur with any kind of unlinking even local ones but as the device is bound to `usbip` as the firmware (or however the bind concept works in the kernel) this is simply the place where this gets noted at first.

septatrix, 2021-11-25:
Especially considering that there are also dozens of usbip disconnects/unbinds which work without a problem

oltdaniel, 2021-11-25:
@septatrix There are still some "weird" race conditions here, as seen in https://github.com/raspberrypi/linux/commit/513765b186c99c09b245067b2ee28cec27a2982d .

oltdaniel, 2022-04-08:
**Update**: We've upgraded all our PIs to the new 64bit bullseye image and the new kernel Oops messages give us a better insight into the issue.

```bash
Unable to handle kernel NULL pointer dereference at virtual address 000000000000004c
# ....
Call trace:
  stub_rx_loop+0x42c/0xb20 [usbip_host]
  kthread+0x140/0x158
  ret_from_fork+0x10/0x20
```

Additionally we can confirm now, the issue is with the Atmel ICE JTAG Debugger. Its USB drivers reset the device from time to time if they aren't used causing this event to happen.

<details>
  <summary>Full Error</summary>
<pre><code>Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.715021] usbip-host 1-1.1: urb completion with non-zero status -71
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.716526] usbip-host 1-1.1: urb completion with non-zero status -71
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.716549] Unable to handle kernel NULL pointer dereference at virtual address 000000000000004c
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.725500] Mem abort info:
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.728438]   ESR = 0x96000006
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.731638]   EC = 0x25: DABT (current EL), IL = 32 bits
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.737077]   SET = 0, FnV = 0
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.740231]   EA = 0, S1PTW = 0
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.743488]   FSC = 0x06: level 2 translation fault
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.748470] Data abort info:
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.751466]   ISV = 0, ISS = 0x00000006
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.755442]   CM = 0, WnR = 0
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.758847] user pgtable: 4k pages, 39-bit VAs, pgdp=000000010586e000
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.765458] [000000000000004c] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.774799] Internal error: Oops: 96000006 [#1] PREEMPT SMP
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.780477] Modules linked in: binfmt_misc usbip_host usbip_core cfg80211 rfkill 8021q garp stp llc imx219 snd_soc_hdmi_codec ftdi_sio usbserial raspberrypi_hwmon vc4 i2c_mux_pinctrl v3d cec gpu_sched i2c_mux bcm2835_codec(C) drm_kms_helper bcm2835_v4l2(C) bcm2835_unicam v4l2_dv_timings bcm2835_isp(C) v4l2_mem2mem v4l2_fwnode snd_soc_core bcm2835_mmal_vchiq(C) snd_compress i2c_brcmstb v4l2_async videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_pcm_dmaengine videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm vc_sm_cma(C) videodev mc snd_timer snd i2c_bcm2835 syscopyarea rpivid_mem sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.845219] CPU: 1 PID: 410314 Comm: stub_rx Tainted: G         C        5.15.30-v8+ #1536
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.853569] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.859484] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.866528] pc : stub_rx_loop+0x42c/0xb20 [usbip_host]
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.871760] lr : stub_rx_loop+0x408/0xb20 [usbip_host]
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.876979] sp : ffffffc00aba3d90
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.880372] x29: ffffffc00aba3d90 x28: ffffff810653f400 x27: ffffff8132a4f8c0
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.887594] x26: 0000000000000008 x25: ffffff810653f400 x24: ffffffc001072638
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.894815] x23: 0000000000000000 x22: ffffff8132a4f840 x21: ffffffc001073418
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.902036] x20: 0000000000000001 x19: ffffff8114922f08 x18: 0000000000000000
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.909257] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.916479] x14: 0000000000000000 x13: 0002000000000080 x12: 0000000000000000
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.923700] x11: 0000000000000002 x10: 0d00000000000000 x9 : ffffffc008802990
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.930922] x8 : 0000000000000100 x7 : 0000000002000000 x6 : ffffffc00aba3bf0
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.938143] x5 : 00000000ffffffd5 x4 : 000000000000000f x3 : 0000000000000001
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.939066] usbip-host 1-1.1: USB disconnect, device number 93
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.945364] x2 : 0000000000000001 x1 : ffffffc001073418 x0 : 000000000000000c
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.945370] Call trace:
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.945373]  stub_rx_loop+0x42c/0xb20 [usbip_host]
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.945379]  kthread+0x140/0x158
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.945388]  ret_from_fork+0x10/0x20
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.972872] Code: 2a1403e2 f94002c4 f8776820 aa1503e1 (f9402000) 
Apr  6 10:06:33 i11-psp-pi-23 kernel: [162820.979048] ---[ end trace 55bc025beadb0834 ]---</code></pre>
</details>