raspberrypi/linux #4735: iPNPstring in f_printer USB gadget is reduced by two bytes

vlisivka, 2021-11-26:
## Description:

Printer USB gadget uses `iPNPstring` to communicate device name and command language with host. Linux driver for printer gadget sends `GET_DEVICE_ID` response packet without 2 last bytes, which may cause trouble for the host driver.

## Steps to reproduce:

  * Use Raspberry Pi, or an other device with USB OTG plug. Raspberry Pi 4 was used by issue author.
  * Configure plug to be in `peripheral` mode, e.g. by adding `dtoverlay=dwc2,dr_mode=peripheral` to `/boot/config.txt`.
  * Connect OTG port to host and reboot Raspberry Pi.
  * Load `g_printer` module using command `sudo modprobe g_printer`.
  * Use command `./get-iPNPstring.pl /dev/usb/lp1` to get iPNPstring from the device. (See [get-iPNPstring.pl.gz](https://github.com/raspberrypi/linux/files/7609975/get-iPNPstring.pl.gz) ). As alternative, kernel `usbmon` or WireShark can be used to capture raw USB packet for `GET_DEVICE_ID` response.
  
## Expected result:

It's expected to receive same string as defined in module: `iPNPstring='MFG:linux;MDL:g_printer;CLS:PRINTER;SN:1;'`
  
##  Actual result:
  
`iPNPstring='MFG:linux;MDL:g_printer;CLS:PRINTER;SN:'`

(Notice that last 2 chars are missing).

## Workarround:

Just add two space to the end of printer gadget `iPNPstring`.

## Root cause:

In `drivers/usb/gadget/function/f_printer.c`, length of iPNPstring is used as length of the whole packet, without length of 2 byte size field.

P33M, 2021-11-29:
This driver is unmodified upstream code. I would recommend posting the issue on the linux-usb mailing list, as you've identified a clear bug. If a patch is accepted then you can update us here when it appears in upstream 5.10.
```
./scripts/get_maintainer.pl drivers/usb/gadget/function/f_printer.c
Felipe Balbi <balbi@kernel.org> (maintainer:USB GADGET/PERIPHERAL SUBSYSTEM)
Greg Kroah-Hartman <gregkh@linuxfoundation.org> (supporter:USB SUBSYSTEM,commit_signer:2/2=100%)
"Maciej Żenczykowski" <maze@google.com> (commit_signer:1/2=50%,authored:1/2=50%,added_lines:2/3=67%,removed_lines:1/1=100%)
Zqiang <qiang.zhang@windriver.com> (commit_signer:1/2=50%,authored:1/2=50%,added_lines:1/3=33%)
Peter Chen <peter.chen@nxp.com> (commit_signer:1/2=50%)
linux-usb@vger.kernel.org (open list:USB GADGET/PERIPHERAL SUBSYSTEM)
linux-kernel@vger.kernel.org (open list)
```

vlisivka, 2021-12-03:
Path is sent to upstream: https://lore.kernel.org/linux-usb/YaeUkBRS9KuUQZcK@donbot/T/#t