raspberrypi/linux #4737: IMX477 ISP output images contain black fringing and aliasing

dustinkerstein, 2021-11-26:
When capturing still images I am seeing black fringing and aliasing around highlights / high contrast borders.

[image]

The above image was originally created with the following command:  `libcamera-still -q 100 --denoise off --shutter 1000000 --gain 1 --awb auto --immediate -o ~/off_denoise.jpg`

When outputting the RAW DNG, there is no fringing / aliasing:

[image]

The above was created with `libcamera-still --denoise off --raw --shutter 1000000 --gain 1 --awb auto --immediate -o ~/raw.jpg`

Note both images have been cropped after capture to highlight the issue. Please let me know if this should be posted on the libcamera bug tracker instead, but I'm fairly certain that I have seen this issue with the old camera stack as well.

I am testing on a Compute Module 4 + Official IO Board + Bullseye Lite 64bit + 5.10.63-v8+ Kernel. Full system info here - https://pastebin.com/654vBepL

Let me know if you need any further info or help replicating.

naushir, 2021-11-29:
Packing/unpacking modes would not cause the fringing that you are seeing here.  The pixel data is exactly the same, just that we either represent data as 12-bits directly or 12-bits in a 16-bit word.

This is possibly due to overshapening in the hardware pipeline. Try reducing the sharpening strength a bit (details can be found [here](https://github.com/raspberrypi/documentation/tree/master/linux/software/libcamera/rpi_SOFT_libcamera_1p0.pdf) and [here](https://www.raspberrypi.com/documentation/accessories/camera.html)) and see if that helps.

dustinkerstein, 2021-11-29:
> This is possibly due to overshapening in the hardware pipeline. Try reducing the sharpening strength a bit (details can be found [...]

That's exactly what it was. Thanks! Dropping sharpening down to 0.25 feels like a reasonable balance but there are still artifacts there. It seems like with using a tuning file I can access the three sharpening parameters (threshold, strength, limit) rather than just the single parameter exposed by the libcamera-apps. I imagine the settings look like:
```
"rpi.sharpen":
{
        "threshold": 1.0,
        "strength": 0.25,
        "limit": 0.25
},
```
I'll play around with that later this week. Thanks for the help!

closed 2021-11-29 by dustinkerstein