raspberrypi/linux #4738: v4l2h264enc encoder CBR mode throws error and fails to encode

dustinkerstein, 2021-11-26:
When encoding using Gstreamer's v4l2h264enc encoder in CBR mode (video_bitrate_mode=1), Gstreamer throws the following error:

```
$ gst-launch-1.0 videotestsrc pattern=snow num-buffers=100 ! video/x-raw,width=1920,height=1080,framerate=10/1 ! v4l2h264enc extra-controls="controls,video_bitrate_mode=1,video_bitrate=25000000,repeat_sequence_header=1" \
! 'video/x-h264,profile=high,level=(string)4.2' ! h264parse ! mp4mux ! filesink location=~/out.mp4
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
ERROR: from element /GstPipeline:pipeline0/v4l2h264enc:v4l2h264enc0: Failed to process frame.
Additional debug info:
../sys/v4l2/gstv4l2videoenc.c(828): gst_v4l2_video_enc_handle_frame (): /GstPipeline:pipeline0/v4l2h264enc:v4l2h264enc0:
Maybe be due to not enough memory or failing driver
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason error (-5)
ERROR: pipeline doesn't want to preroll.
Freeing pipeline ...
```

VBR mode (video_bitrate_mode=0) works without issue. I'm unable to test FFmpeg's h264_v4l2m2m encoder, as it only supports the default VBR mode since the necessary v4l2 controls aren't exposed - https://trac.ffmpeg.org/ticket/2305 (it would be awesome to see this fixed)

I am testing on a Compute Module 4 + Official IO Board + Bullseye Lite 64bit + 5.10.63-v8+ Kernel. Full system info here - https://pastebin.com/654vBepL

Let me know if you need any further info or help replicating.

6by9, 2021-12-10:
CBR is only support in main or baseline profiles, not high.

CABAC entropy encoding is done as a parallel process to the main encode, but until it has the final frame it can't judge the bitrate to feed back into rate control.

dustinkerstein, 2021-12-16:
Got it. I'll go ahead and close this issue. Thanks for the explanation.

closed 2021-12-16 by dustinkerstein