raspberrypi/linux #4739: v4l2h264enc doesn't seem to support framerates < 9

dustinkerstein, 2021-11-26:
The following command just hangs during pipeline initialization:
```
gst-launch-1.0 -vvv videotestsrc pattern=snow num-buffers=100 ! video/x-raw,width=1920,height=1080,framerate=8/1 ! v4l2h264enc extra-controls="controls,video_bitrate_mode=0,video_bitrate=25000000,repeat_sequence_header=1" \
! 'video/x-h264,profile=high,level=(string)4.1' ! h264parse ! mp4mux ! filesink location=~/out.mp4
```
But this one (with FPS=9) works fine:
```
gst-launch-1.0 -vvv videotestsrc pattern=snow num-buffers=100 ! video/x-raw,width=1920,height=1080,framerate=9/1 ! v4l2h264enc extra-controls="controls,video_bitrate_mode=0,video_bitrate=25000000,repeat_sequence_header=1" \
! 'video/x-h264,profile=high,level=(string)4.1' ! h264parse ! mp4mux ! filesink location=~/out.mp4
```
FFmpeg seems to work fine with all framerates:

```
ffmpeg -y -f lavfi -r 1 -i nullsrc=s=1920x1080 -filter_complex "geq=random(1)*255:128:128" -t 100 -c:v h264_v4l2m2m -b:v 25M -minrate 25M -maxrate 25M ~/out.mp4
```
I am testing on a Compute Module 4 + Official IO Board + Bullseye Lite 64bit + 5.10.63-v8+ Kernel. Full system info here - https://pastebin.com/654vBepL

Let me know if you need any further info or help replicating.

dustinkerstein, 2021-11-27:
And a quick sanity check with the x264enc encoder works correctly:

```
gst-launch-1.0 videotestsrc pattern=snow num-buffers=100 ! video/x-raw,width=1920,height=1080,framerate=8/1 ! x264enc ! 'video/x-h264,profile=high,level=(string)4.1' ! h264parse ! mp4mux ! filesink location=~/out.mp4
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:12.167761837
Setting pipeline to NULL ...
Freeing pipeline ...
```

6by9, 2021-12-10:
Weird. The motion estimation phase appears to be stalling, but it shouldn't be affected by frame rate at all as the codec just encodes the 

It needs some investigation, although it seems dropping the bitrate to 20Mbit/s means it doesn't stall. Keeping the bitrate that high when the frame rate is coming down does seem a little unnecessary.

dustinkerstein, 2021-12-16:
Yah, it is definitely a bit strange. And with higher framerates you can see some visual indication that #4741 might not be specific to FFmpeg after all.

My use-case requires high visual quality, so being able to max out the effective quality via artificial framerates is important. 

Side note / rant - I really do wish encoders just used bits/frame vs. bits/sec.

6by9, 2022-03-29:
I have a clue on this one now. 

Running your command
```
gst-launch-1.0 -vvv -e videotestsrc pattern=snow num-buffers=100 ! video/x-raw,width=1920,height=1080,framerate=9/1 ! v4l2h264enc extra-controls="controls,video_bitrate_mode=0,video_bitrate=25000000,repeat_sequence_header=1" ! 'video/x-h264,profile=high,level=(string)4.1' ! h264parse ! mp4mux ! filesink location=~/out.mp4
```
and looking at the sizes of the initial buffers I get
```
size=1266025
size=1252473
size=1252150
size=1145993
size=1037389
size=947380
size=836314
size=726569
size=632108
size=520659
size=404977
size=316834
size=214526
size=115417
```
Because the source material is noise (which isn't a typical image for H264 to encode) the bitrate control seriously overshoots on that first frame (1.25Mbytes at 9fps would be 91Mbit/s).
The output FIFO is only 2MB in size. Decreasing the framerate means that it decreases the QP for that initial frame from 25 to 20, and I suspect the encoder then exceeds the output FIFO with that first frame and stalls.

Removing ```pattern=snow``` and it'll encode fine at 1fps. Similarly reducing the target bitrate and it's happy.
So it looks like it's down to the initial QP setting should you feed in noise.

As noted in #4741, with FFmpeg the framerate isn't being passed down to the encoder, so it won't be setting the initial QP in the same way, and it works.

It would be possible (but ugly) to increase the size of that output buffer, but it's not something that I really want to do as this isn't a sensible use case.