raspberrypi/linux #4743: USB hub enumeration failure: error -71

HendrikR, 2021-11-27:
I have a Raspberry Pi 3 Model B Plus Rev 1.3, with kernel 5.10.63-v7+ Raspbian running on it.
Connected to one of the Pi's USB ports is a hub, Chipset idVendor=0424, idProduct=2514 [see here](https://devicehunt.com/view/type/usb/vendor/0424/device/2514).
When I try to connect a USB (input) device, it cannot enumerate and outputs errors:

    Nov 27 20:16:17 retropie kernel: usb 1-1.1.3.1: new low-speed USB device number 15 using dwc_otg
    Nov 27 20:16:17 retropie kernel: usb 1-1.1.3.1: device descriptor read/64, error -71
    Nov 27 20:16:17 retropie kernel: usb 1-1.1.3.1: device descriptor read/64, error -71
    Nov 27 20:16:17 retropie kernel: usb 1-1.1.3.1: new low-speed USB device number 16 using dwc_otg
    Nov 27 20:16:17 retropie kernel: usb 1-1.1.3.1: device descriptor read/64, error -71
    Nov 27 20:16:18 retropie kernel: usb 1-1.1.3.1: device descriptor read/64, error -71
    Nov 27 20:16:18 retropie kernel: usb 1-1.1.3-port1: attempt power cycle
    Nov 27 20:16:18 retropie kernel: usb 1-1.1.3.1: new low-speed USB device number 17 using dwc_otg
    Nov 27 20:16:19 retropie kernel: usb 1-1.1.3.1: device not accepting address 17, error -71
    Nov 27 20:16:19 retropie kernel: usb 1-1.1.3.1: new low-speed USB device number 18 using dwc_otg
    Nov 27 20:16:19 retropie kernel: usb 1-1.1.3.1: device not accepting address 18, error -71
    Nov 27 20:16:19 retropie kernel: usb 1-1.1.3-port1: unable to enumerate USB device

Strangely enough, I *can* connect non-input USB devices. I tested several USB sticks and another USB hub, all of which connected without problems. I could even connect the (working) USB hub to the faulty one, and then connect an input device to the second USB hub, without receiving errors.
The issue may be related to [this one](https://github.com/raspberrypi/linux/issues/2692). As outlined there, I tried setting `dwc_otg.fiq_fsm_mask=0x0`, but still got the errors.

HendrikR, 2021-11-29:
I tried several devices on the Hub, with the HID (human input device) ones producing an error, while others worked:
- (error) A cheap USB gamepad: idVendor=081f, idProduct=e401, bcdDevice= 1.06
- (error) A Lenovo ThinkPad USB keyboard: idVendor=17ef, idProduct=6047, bcdDevice= 3.30
- (error) A USB mouse: idVendor=09da, idProduct=000a, bcdDevice= 0.14
- (works) A USB cardreader: idVendor=214b, idProduct=7250, bcdDevice= 1.00
- (works) Another USB Hub from a different manufacturer: idVendor=05e3, idProduct=0608, bcdDevice=77.64

I also tried the (faulty) USB hub on my desktop computer, where it produced no errors whatsoever with any device.

P33M, 2021-11-30:
Please post a full `sudo lsusb -v` for the bus with the faulty hub attached.

HendrikR, 2021-11-30:
[Output of `sudo lsusb -v`](https://gist.github.com/HendrikR/a74b5a2be92d0a4457e76892c90119c0)

P33M, 2021-12-01:
Your faulty hub is actually a `058f:6254 Alcor Micro Corp. USB Hub` which is a single-TT hub. These are not recommended for use on Pi 0-3 for a variety of reasons. I would suggest sourcing a multi-TT hub.