raspberrypi/linux #4749: official 7inch DSI panel spurious touch events

ardera, 2021-11-30:
**Describe the bug**
- When using the official 7inch panel Spurious touch events happen at a rate of about 2.2 events per second with the latest touch fix in #4736. Most of the time at coordinates 799, 479, sometimes at 499, -3616.

**To reproduce**
- attach an official 7inch DSI panel to the Pi
- upgrade kernel to latest version using `rpi-update` (because touch needs https://github.com/raspberrypi/linux/pull/4736 to be working at all)
- run `evtest` (`sudo apt install evtest`) and select the `generic ft5x06` device
- don't touch the screen
- there's still touch events reported

**Expected behaviour**
- no touch events reported when the screen is not being touched, as was the case in older raspbian as well (so it must be somehow possible to get it working)

**Actual behaviour**
- touch down and touch up events are reported even though the screen is not being touched, with about 2.2 touch down events per second, at pos 799, 479 and rarerly 499, -3616
- (see `evtest` output)

**System**

[`raspinfo.txt`](https://github.com/raspberrypi/linux/files/7625369/raspinfo.txt)

* Model: Raspberry Pi 4B+
* 
  ```
  $ cat /etc/rpi-issue
  Raspberry Pi reference 2021-10-30
  Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 
  288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2
  ```
*
  ```
  $ vcgencmd version
  Nov 16 2021 17:38:45
  Copyright (c) 2012 Broadcom
  version 99c5aad7da0d1e3b2c28669a3941db77734b9f2d (clean) (release) (start)
  ```
*
  ```
  $ uname -a
  Linux hpi4 5.10.81-v7l+ #1492 SMP Fri Nov 26 21:42:52 GMT 2021 armv7l GNU/Linux
  ```

**Logs**
[output of `evtest`](https://github.com/raspberrypi/linux/files/7625375/evtest.txt)

6by9, 2021-11-30:
On first power up I do see spurious events being generated by the controller, but not after the first touch on the screen. Does that match with your observations?

The ts controller is reporting these touches, so it seems to be in a funny state.

ardera, 2021-11-30:
It seems like they still ocurr after the first touch, even while I'm touching the screen.

What's more weird is that it seems like the `ABS_MT_SLOT` events are missing, it just generates two pairs of `ABS_MT_POSITION_...` after another

<details>

<summary> evtest output while touching the screen </summary>

```
pi@hpi4:~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      vc4
/dev/input/event1:      vc4
/dev/input/event2:      generic ft5x06 (00)
Select the device event number [0-2]: 2
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "generic ft5x06 (00)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    799
      Min        0
      Max      799
    Event code 1 (ABS_Y)
      Value    479
      Min        0
      Max      479
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      799
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      479
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1638281585.764099, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638281585.764099, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281585.764099, -------------- SYN_REPORT ------------
Event: time 1638281585.794164, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24895
Event: time 1638281585.794164, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281585.794164, -------------- SYN_REPORT ------------
Event: time 1638281586.004096, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638281586.004096, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281586.004096, -------------- SYN_REPORT ------------
Event: time 1638281586.034167, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24896
Event: time 1638281586.034167, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281586.034167, -------------- SYN_REPORT ------------
Event: time 1638281586.184085, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638281586.184085, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281586.184085, -------------- SYN_REPORT ------------
Event: time 1638281586.215333, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24897
Event: time 1638281586.215333, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281586.215333, -------------- SYN_REPORT ------------
[... 124 lines cut ...]
Event: time 1638281589.844084, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281589.844084, -------------- SYN_REPORT ------------
Event: time 1638281589.874164, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24905
Event: time 1638281589.874164, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281589.874164, -------------- SYN_REPORT ------------
Event: time 1638281589.934100, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638281589.934100, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281589.934100, -------------- SYN_REPORT ------------
Event: time 1638281589.964165, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24906
Event: time 1638281589.964165, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281589.964165, -------------- SYN_REPORT ------------
Event: time 1638281590.024098, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638281590.024098, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281590.024098, -------------- SYN_REPORT ------------
Event: time 1638281590.054162, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24907
Event: time 1638281590.054162, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281590.054162, -------------- SYN_REPORT ------------
Event: time 1638281590.234123, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638281590.234123, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638281590.234123, -------------- SYN_REPORT ------------
Event: time 1638281590.264172, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 24908
Event: time 1638281590.264172, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638281590.264172, -------------- SYN_REPORT ------------
```
</details>

6by9, 2021-11-30:
AIUI ```ABS_MT_SLOT``` is only needed if the slot is changed. Touch with 2 fingers and you'll see slot 0 and slot 1 being selected.

Something very odd is going on with those tracking IDs. The value the driver generates is between 0 and 15 due to the masking at https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/input/touchscreen/edt-ft5x06.c#L271. How evtest is seeing values of 24000 is really strange.

ardera, 2021-11-30:
Ah I think I got it. In the old touch driver they didn't iterate up to the max supported points in this loop: https://github.com/raspberrypi/linux/blob/30bb91977d65c22cec3eaeadb06a6ff7f36ecbc5/drivers/input/touchscreen/edt-ft5x06.c#L253

but to the number of current touch contacts, reported by the FT. Which is a single byte that's reported right before the first touch coordinate (so `buf[offset-1]`)

6by9, 2021-11-30:
> Ah I think I got it. In the old touch driver they didn't iterate up to the max supported points in this loop: 

Sadly it's not it.
I have local changes that do that:
```
diff --git a/drivers/input/touchscreen/edt-ft5x06.c b/drivers/input/touchscreen/edt-ft5x06.c
index f9a55dd5755f..7a8b2d29b023 100644
--- a/drivers/input/touchscreen/edt-ft5x06.c
+++ b/drivers/input/touchscreen/edt-ft5x06.c
@@ -197,6 +197,7 @@ static irqreturn_t edt_ft5x06_ts_isr(int irq, void *dev_id)
        int i, type, x, y, id;
        int offset, tplen, datalen, crclen;
        int error;
+       unsigned int num_points;
 
        switch (tsdata->version) {
        case EDT_M06:
@@ -244,9 +245,16 @@ static irqreturn_t edt_ft5x06_ts_isr(int irq, void *dev_id)
 
                if (!edt_ft5x06_ts_check_crc(tsdata, rdbuf, datalen))
                        goto out;
-       }
+               num_points = tsdata->max_support_points;
+       } else {
+               /* Register 2 is TD_STATUS, containing the number of touch
+                * points.
+                */
+               num_points = min(rdbuf[2] & 0xf, tsdata->max_support_points);
+               dev_err(dev, "TS reported %u points\n", num_points);
+       } 
 
-       for (i = 0; i < tsdata->max_support_points; i++) {
+       for (i = 0; i < num_points; i++) {
                u8 *buf = &rdbuf[i * tplen + offset];
```
but on first read the touchscreen is reporting 10 touch events in that register.

6by9, 2021-11-30:
> > Ooh, magic. That means I can revert 16c7561 and make it a one-liner to set that flag.
> Exactly 😄

Except it seems not to work - I'm not seeing any release events at all with that.
Ignoring it for now as the existing patch does work.

ardera, 2021-11-30:
> Sadly it's not it. I have local changes that do that:

oh. then I have no idea

> Except it seems not to work - I'm not seeing any release events at all with that.
Ignoring it for now as the existing patch does work.

true, that makes sense. though just to be sure did you call `input_mt_sync_frame` at the end of the ISR? i think that call's not there in repo version

6by9, 2021-11-30:
> true, that makes sense. though just to be sure did you call input_mt_sync_frame at the end of the ISR? i think that call's not there in repo version

No I hadn't changed the ```input_sync``` to ```input_mt_sync_frame```.
Need to drop the ```input_mt_report_pointer_emulation``` as well, as that is done by ```input_mt_sync_frame```.
One to look at another day though.

#4750 has my latest patches, but it doesn't solve the garbage at the start.

I'm trying extending the reset pulse (currently 6ms) to see if that helps, but I'm running out of ideas. I'm still not seeing the garbage results coming back after the first touch though. I am on a Pi3B+, but that shouldn't make any difference.

thijstriemstra, 2021-11-30:
> upgrade kernel to latest version using rpi-update (because touch needs edt-ft5x06 touchscreen fixes #4736 to be working at all)

I upgraded the kernel:

```
$ sudo rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 5.10.63-v7+
#############################################################
WARNING: This update bumps to rpi-5.10.y linux tree
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234
'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.

DO NOT use 'rpi-update' as part of a regular update process.

##############################################################
Would you like to proceed? (y/N)

 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   173  100   173    0     0    748      0 --:--:-- --:--:-- --:--:--   748
100  121M  100  121M    0     0  3159k      0  0:00:39  0:00:39 --:--:-- 4761k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 5.10.81+
 *** depmod 5.10.81-v7l+
 *** depmod 5.10.81-v7+
 *** depmod 5.10.81-v8+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 8a74f2a62979db8793ae7d5a1b6d8f29467e1c27
 *** A reboot is needed to activate the new firmware
```

But this didn't fix touch unfortunately. It does a right-click, garbled something.

> type 0 is event down, and type 2 is event on. The x and y co-ords are outside the display area, it's repeating ID/slot values.
(type = TOUCH_EVENT_RESERVED events are ignored, so not logged).

Seeing similar crap with Raspberry pi 3b+ `evtest` output:

```
Event: time 1638311693.056529, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638311693.056529, -------------- SYN_REPORT ------------
Event: time 1638311693.145183, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638311693.145183, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638311693.145183, -------------- SYN_REPORT ------------
Event: time 1638311693.176296, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 188
Event: time 1638311693.176296, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638311693.176296, -------------- SYN_REPORT ------------
Event: time 1638311693.775214, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638311693.775214, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638311693.775214, -------------- SYN_REPORT ------------
Event: time 1638311693.806120, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 189
Event: time 1638311693.806120, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638311693.806120, -------------- SYN_REPORT ------------
Event: time 1638311694.135243, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value -3616
Event: time 1638311694.135243, type 3 (EV_ABS), code 1 (ABS_Y), value -3616
Event: time 1638311694.135243, -------------- SYN_REPORT ------------
Event: time 1638311694.165307, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 479
Event: time 1638311694.165307, type 3 (EV_ABS), code 1 (ABS_Y), value 479
Event: time 1638311694.165307, -------------- SYN_REPORT ------------
Event: time 1638311694.345218, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638311694.345218, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638311694.345218, -------------- SYN_REPORT ------------
Event: time 1638311694.375309, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 190
Event: time 1638311694.375309, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638311694.375309, -------------- SYN_REPORT ------------
Event: time 1638311694.946114, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1638311694.946114, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1638311694.946114, -------------- SYN_REPORT ------------
Event: time 1638311694.975306, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 191
Event: time 1638311694.975306, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1638311694.975306, -------------- SYN_REPORT ------------
```

ardera, 2021-12-07:
@6by9 do you have the firmware sources so you can look at what the closed-source driver did differently? or is there no obvious deviation there?

I have a i2c sniffer that i used for debugging the last two times the DSI panel was buggy (😉), I'll use that and see if I find something

ardera, 2021-12-16:
I don't know what changed but I can't reproduce it anymore.