raspberrypi/linux #4756: RPi 3 B+ oops on 14TB USB drive that works fine in RPi 4

lparsons42, 2021-12-05:
I have two RPis on my home network currently; one is a 3B+ that has been my home file server for some time, the other is a 4 that I have been building to eventually become my new home web server / gateway.  

The 3B+ has had a couple external USB drives on it for some time, all mounted ext4.  Two have been there for over a year now without incident; of those one is a bus powered 1TB SSD the other an externally powered 8TB HDD.  Both are WD.  

More recently I picked up a 14TB externally powered HDD, also WD.  I found that if I mount it on the RPi 4 (not the ultimate goal) it runs fine.  I can partition it, format it, write to it, etc without issue.  However the 3B+ does not work well with it at all.  I can usually mount it, but very few operations go through without kernel oops.  For example, here is what happened when I tried to rsync from the 8TB to the 14TB (both drives connected via USB to the 3B+):

Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453410] rsync           D    0  2888   2887 0x00000000
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453437] [<80830934>] (__schedule) from [<80830fa4>] (schedule+0x50/0xa8)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453454] [<80830fa4>] (schedule) from [<803aab68>] (wait_transaction_locked+0xc0/0xf8)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453472] [<803aab68>] (wait_transaction_locked) from [<803ab0e4>] (start_this_handle+0x49c/0x768)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453488] [<803ab0e4>] (start_this_handle) from [<803aba38>] (jbd2__journal_start+0x104/0x210)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453506] [<803aba38>] (jbd2__journal_start) from [<80349ffc>] (__ext4_journal_start_sb+0x6c/0xf8)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453523] [<80349ffc>] (__ext4_journal_start_sb) from [<8036c810>] (ext4_da_write_begin+0x1b8/0x4bc)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453541] [<8036c810>] (ext4_da_write_begin) from [<80236e44>] (generic_perform_write+0xcc/0x1dc)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453560] [<80236e44>] (generic_perform_write) from [<80239830>] (__generic_file_write_iter+0x19c/0x1f8)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453578] [<80239830>] (__generic_file_write_iter) from [<803567c8>] (ext4_file_write_iter+0x2ac/0x4ac)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453596] [<803567c8>] (ext4_file_write_iter) from [<802ab2dc>] (__vfs_write+0x10c/0x170)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453611] [<802ab2dc>] (__vfs_write) from [<802ab528>] (vfs_write+0xb4/0x1bc)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453625] [<802ab528>] (vfs_write) from [<802ab7fc>] (ksys_write+0x74/0xe8)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453639] [<802ab7fc>] (ksys_write) from [<802ab888>] (sys_write+0x18/0x1c)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453653] [<802ab888>] (sys_write) from [<80101000>] (ret_fast_syscall+0x0/0x28)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453660] Exception stack(0xa546ffa8 to 0xa546fff0)
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453672] ffa0:                   00000000 75a71008 00000001 75a71008 00040000 00040000
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453686] ffc0: 00000000 75a71008 00091d08 00000004 00004031 00091d08 000009cc 00000000
Dec  5 15:29:22 PiWeb2020 kernel: [ 3561.453695] ffe0: 00000000 7efd323c 0004857c 76e865cc

I'll at times see similar errors in /var/log/messages even from simpler tasks like trying to 'ls' on that drive.  Again, the 8TB drive does just fine with this and more complicated requests and has for over a year.

P33M, 2021-12-06:
What happens if you move all your connected devices to a downstream, powered USB hub?

lparsons42, 2021-12-08:
Thank you for the suggestion.  I purchased a powered USB3 hub and tried that.  The results are different, but not yet correct.

With the drive connected to the USB3 hub I can mount the drive and copy some files to to.  However eventually the drive becomes unresponsive through the hub as well; I can see it in 'df' but if I 'cd' to the drive, 'ls' will hang indefinitely.  I also found that if I try to boot the 3B+ with it connected through the hub (or reboot the pi with the 14TB drive attached through the hub), the 3B+ will hang until I disconnect it (which it also did without the hub).  I haven't found anything in dmesg or /var/log/messages that helps with that part.

P33M, 2021-12-14:
If your ls command hangs, then ultimately after a timeout there will be errors in dmesg. Do any such errors appear?

lparsons42, 2021-12-14:
The errors posted at the top of this thread are all I could ever get it to produce.  I couldn't find any associated with the stalled boot attempts either.  Commands could hang for 20 minutes or more and nothing would show up in dmesg, /var/log/messages, or anywhere else I though to check.
For the time being I put the 14TB and 8TB drives on the pi 4, so I could start copying data from the 8TB to the 14TB.  Total data volume is ~5.2TB; currently I expect it will take about another 2-3 days to finish copying (via USB 3.0 using rsync on a single system).  When this finishes I'll try just the 14TB drive on the 3B+ and see how that behaves without the 8TB on the same; otherwise I'll try to get it to throw some errors at me.

P33M, 2021-12-14:
Please post a full `sudo lsusb -vv` with both drives connected to either a Pi 4 or the 3B+.

lparsons42, 2021-12-15:
Here it is from the Pi 4.  Probably worth noting though is this is with just the 8TB and 14TB drives attached; the 3B+ had issues when the 14TB, 8TB, and 1TB SSD were all simultaneously attached; I didn't try these two on there without the SSD (and the SSD was bus powered while the other two had external power supplies).  Once my file copying operation is complete I'll try a few other configurations to see if I can get more useful returns on the 3B+.  

In the interest of brevity I took out the devices that weren't the two external hard drives.  If the RPi hardware hub, or my keyboard and mouse are of interest I can paste them back in:

Bus 002 Device 003: ID 1058:25ee Western Digital Technologies, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x1058 Western Digital Technologies, Inc.
  idProduct          0x25ee 
  bcdDevice           40.07
  iManufacturer           2 Western Digital
  iProduct                3 My Book 25EE
  iSerial                 1 37484A5958475A46
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                8mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
[... 84 lines cut ...]
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us 
    Deep BESL value    61440 us 
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

lparsons42, 2021-12-18:
Now I just tried to mount the 14TB drive on the 3B+ after the rsync finished on the Pi4.  The 3B+ is quite unhappy with it, as seen in dmesg:

[100545.902912] usb 1-1.1.3: USB disconnect, device number 5
[100565.932922] usb 1-1.3: USB disconnect, device number 6
[834195.930568] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
[834196.063007] usb 1-1.3: New USB device found, idVendor=1058, idProduct=264d, bcdDevice=30.12
[834196.063032] usb 1-1.3: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[834196.063048] usb 1-1.3: Product: easystore 264D
[834196.063064] usb 1-1.3: Manufacturer: Western Digital
[834196.063079] usb 1-1.3: SerialNumber: 324347484547384A
[834196.064070] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[834196.064789] scsi host1: usb-storage 1-1.3:1.0
[834197.101835] scsi 1:0:0:0: Direct-Access     WD       easystore 264D   3012 PQ: 0 ANSI: 6
[834197.102737] sd 1:0:0:0: Attached scsi generic sg2 type 0
[834197.104722] sd 1:0:0:0: [sdb] Spinning up disk...
[834197.105387] scsi 1:0:0:1: Enclosure         WD       SES Device       3012 PQ: 0 ANSI: 6
[834197.109456] scsi 1:0:0:1: Attached scsi generic sg3 type 13
[834198.140574] ................ready
[834213.741287] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[834213.741709] sd 1:0:0:0: [sdb] 27344764928 512-byte logical blocks: (14.0 TB/12.7 TiB)
[834213.741729] sd 1:0:0:0: [sdb] 4096-byte physical blocks
[834213.742363] sd 1:0:0:0: [sdb] Write Protect is off
[834213.742383] sd 1:0:0:0: [sdb] Mode Sense: 47 00 10 08
[834213.743000] sd 1:0:0:0: [sdb] No Caching mode page found
[834213.743019] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[834213.876377]  sdb: sdb1
[834213.880096] sd 1:0:0:0: [sdb] Attached SCSI disk
[834214.007233] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=2520, sector_sz=512)
[834214.007282] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 08 00 00 00 00 08 00 00
[834214.007328] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[834214.007353] sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x3 [current] 
[834214.007378] sd 1:0:0:0: [sdb] tag#0 ASC=0x11 ASCQ=0x0 
[834214.007401] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 08 00 00 00 00 08 00 00
[834214.007431] blk_update_request: critical medium error, dev sdb, sector 2048 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[834214.149803] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=8916, sector_sz=512)
[834214.149836] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 06 5d df f4 00 00 00 00 f0 00 00
[834214.149874] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[834214.149892] sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x3 [current] 
[834214.149909] sd 1:0:0:0: [sdb] tag#0 ASC=0x11 ASCQ=0x0 
[834214.149926] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 06 5d df f4 00 00 00 00 f0 00 00
[834214.149948] blk_update_request: critical medium error, dev sdb, sector 27344761856 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0
[834244.500726] usb 1-1.3: reset high-speed USB device number 8 using dwc_otg
[834263.056863] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=63964, sector_sz=512)
[834263.056883] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 06 5d df f4 00 00 00 00 f0 00 00
[834263.056903] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[834263.056913] sd 1:0:0:0: [sdb] tag#0 Sense Key : 0xb [current] 
[834263.056921] sd 1:0:0:0: [sdb] tag#0 ASC=0x47 ASCQ=0x3 
[834263.056930] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 06 5d df f4 00 00 00 00 f0 00 00
[834263.056941] blk_update_request: I/O error, dev sdb, sector 27344761856 op 0x0:(READ) flags 0x84700 phys_seg 16 prio class 0
[834263.057963] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=5636, sector_sz=512)
[834263.057973] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 06 5d df f4 f0 00 00 00 10 00 00
[834263.057986] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[834263.057994] sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x3 [current] 
[834263.058003] sd 1:0:0:0: [sdb] tag#0 ASC=0x11 ASCQ=0x0 
[834263.058011] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 06 5d df f4 f0 00 00 00 10 00 00
[834263.058021] blk_update_request: critical medium error, dev sdb, sector 27344762096 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
[834293.780767] usb 1-1.3: reset high-speed USB device number 8 using dwc_otg
[834293.912328] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=63964, sector_sz=512)
[834293.912359] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 0a e0 00 00 00 f0 00 00
[834293.912412] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=0x00 cmd_age=30s
[834293.912434] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 0a e0 00 00 00 f0 00 00
[834293.912458] blk_update_request: I/O error, dev sdb, sector 2784 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0
[834294.187135] sd 1:0:0:0: [sdb] Unaligned partial completion (resid=48008, sector_sz=512)
[834294.187171] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 0c c0 00 00 00 f0 00 00
[834294.187214] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[834294.187236] sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x3 [current] 
[834294.187255] sd 1:0:0:0: [sdb] tag#0 ASC=0x11 ASCQ=0x0 
[834294.187275] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 00 00 0c c0 00 00 00 f0 00 00
[834294.187300] blk_update_request: critical medium error, dev sdb, sector 3264 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0

This worked fine on the Pi4.  I think I might just change my plans and have the 3B+ be my webserver / gateway and make the 4Pi into my network file server.