raspberrypi/linux #4760: RPi4 USB stops working with video playback

smallint, 2021-12-09:
I have a dvb-s usb device (TBS 5990) connected to a PI4. It is running ArchLinux:

```
uname -a
Linux kodipi 5.10.83-1-rpi-ARCH #1 SMP Tue Dec 7 15:22:25 UTC 2021 armv7l GNU/Linux
```
The USB device is connected to a externally powered hub otherwise the Pi does not boot. I can stream dvb data via network to another computer. As soon as the Pi4 starts video playback in Kodi (from harddisc, from internet, does not matter) then the USB device stops working. After stopping video playback, the device is usable again after some seconds.

This setup has been working with older kernels (< 5.10) and the old kms output. I have tried to debug the issue but dmesg does not show anything related to that. Do you have any ideas what could causing this and how to debug it?

6by9, 2021-12-09:
The TBS 5990 isn't supported by mainline Linux, therefore I'm assuming you're building TBS's media_tree drivers from https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/usb/cx231xx/cx231xx-cards.c#L984.
This will be replacing the Pi V4L2 drivers, so at a guess your system is dropping back to software decoding. Potentially that will be starving the process handling your DVB device of CPU time. What does ```htop``` report when you start video playback?

smallint, 2021-12-09:
No, I have my own module build: https://github.com/smallint/tbs-usb

smallint, 2021-12-09:
It just adds the cx231xx dvb module and its dependencies. This has been working since kernel 4 for all of my Pis and iMX6 boxes and it is still working ;)

smallint, 2021-12-09:
Playback a 720p stream uses 16% CPU.