raspberrypi/linux #4767: Flip done time out using KMS driver and calling snd_pcm_writei

pjx3, 2021-12-14:
**Describe the bug**

When using the KMS driver and calling snd_pcm_writei, flip done time outs occur and the system becomes unstable. Removing the call to snd_pcm_writei or running the FKMS driver does not cause the flip done time outs.

**To reproduce**

```
git clone https://github.com/pjx3/kms_fdto.git
cd kms_fdto
cmake .
make
kms_fdto/kms_fdto
```

**Expected behaviour**

No flip done time outs (rendering occurs normally, a single triangle).

**Actual behaviour**

Flip done time outs and system becomes unresponsive (screen is black).

**System**

System Information
------------------

Raspberry Pi Compute Module 4 Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage4

Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux
Revision        : a03140
Serial          : 10000000e3299001
Model           : Raspberry Pi Compute Module 4 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Dec  1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       29737152 26695572   1781024  94% /
devtmpfs          280444        0    280444   0% /dev
tmpfs             412540        0    412540   0% /dev/shm
tmpfs             412540     5716    406824   2% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs             412540        0    412540   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095    56096    202000  22% /boot
/dev/mmcblk1p1  31154688   102432  31052256   1% /media/sdcard
tmpfs              82508        0     82508   0% /run/user/1000

Filename                                Type            Size            Used            Priority
/var/swap                               file            1780732         0               -2

[... 880 lines cut ...]
[    7.514975] usb 1-1.2: new full-speed USB device number 3 using dwc2
[    7.646858] usb 1-1.2: New USB device found, idVendor=04d9, idProduct=0346, bcdDevice= 0.11
[    7.646895] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    7.646925] usb 1-1.2: Product: USB-HID Keyboard
[    7.653036] input: USB-HID Keyboard as /devices/platform/soc/fe980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:04D9:0346.0001/input/input2
[    7.715895] hid-generic 0003:04D9:0346.0001: input,hidraw0: USB HID v1.11 Keyboard [USB-HID Keyboard] on usb-fe980000.usb-1.2/input0
[    7.728566] hid-generic 0003:04D9:0346.0002: hiddev96,hidraw1: USB HID v1.11 Device [USB-HID Keyboard] on usb-fe980000.usb-1.2/input1
[    7.740934] input: USB-HID Keyboard System Control as /devices/platform/soc/fe980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:04D9:0346.0003/input/input3
[    7.805359] input: USB-HID Keyboard Consumer Control as /devices/platform/soc/fe980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:04D9:0346.0003/input/input4
[    7.805832] input: USB-HID Keyboard as /devices/platform/soc/fe980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:04D9:0346.0003/input/input5
[    7.806336] hid-generic 0003:04D9:0346.0003: input,hidraw2: USB HID v1.11 Keyboard [USB-HID Keyboard] on usb-fe980000.usb-1.2/input2
[    8.399378] 8021q: 802.1Q VLAN Support v1.8
[    8.465845] random: crng init done
[    8.465858] random: 7 urandom warning(s) missed due to ratelimiting
[    8.544213] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.714658] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.805505] Adding 1780732k swap on /var/swap.  Priority:-2 extents:1706 across:28888212k SSFS
[    8.829006] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.829536] bcmgenet fd580000.ethernet eth0: Link is Down
[   12.955122] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.955175] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.736782] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead

**Additional context**
Also tested on kernel 5.15 (using rpi-update next).

popcornmix, 2021-12-14:
Thanks for example. I apt installed libvulkan-dev and glslang-tools, and it built okay.
But it crashes with a segfault.
```
(gdb) bt
#0  0x00000000 in ?? ()
#1  0x00012790 in createDirect2DisplaySurface (physicalDevice=0x5b7b8, width=3840, height=2160) at /home/pi/projects/kms_fdto/kms_fdto/kms_fdto.cpp:149
#2  0x000133cc in findPhysicalDevice () at /home/pi/projects/kms_fdto/kms_fdto/kms_fdto.cpp:377
#3  0x000167dc in initVulkan () at /home/pi/projects/kms_fdto/kms_fdto/kms_fdto.cpp:1156
#4  0x0001794c in main () at /home/pi/projects/kms_fdto/kms_fdto/kms_fdto.cpp:1478
```
Are you using apt version of vulkan? I'm on bullseye. I'm on 5.15 kernel.
I'm booting to desktop, but I've tried running from ssh and directly attached console, both with desktop active and after a switch of VT (ctrl-alt-F1).
My monitor defaults to 4k, but changing g_renderWidth and g_renderHeight didn't help.

pjx3, 2021-12-14:
hmm, i boot to console, so i think that could be the issue.
these are my packages, same as you have.

***@***.***:~ $ apt list --installed *vulkan*
Listing... Done
libvulkan-dev/oldstable,now 1.1.97-2 armhf [installed]
libvulkan1/oldstable,now 1.1.97-2 armhf [installed,automatic]
vulkan-tools/oldstable,now 1.1.97+dfsg1-2 armhf [installed]
vulkan-validationlayers-dev/oldstable,now 1.1.97-2 armhf [installed]
vulkan-validationlayers/oldstable,now 1.1.97-2 armhf [installed,automatic]
***@***.***:~ $ apt list --installed *glsl*
Listing... Done
glslang-tools/oldstable,now 7.10.2984-1 armhf [installed]
***@***.***:~ $

i'll have a go at start here, see if it does the same thing.

popcornmix, 2021-12-14:
Oh - you are on oldstable (buster I assume)? I'm on bullseye (with newer packages):
```
pi@pi4:~/projects/kms_fdto $ apt list --installed *vulkan*
Listing... Done
libvulkan-dev/stable,now 1.2.162.0-1 armhf [installed]
libvulkan1/stable,now 1.2.162.0-1 armhf [installed,automatic]
mesa-vulkan-drivers-dbgsym/stable,now 20.3.5-1+rpt3+rpi1 armhf [installed]
mesa-vulkan-drivers/stable,now 20.3.5-1+rpt3+rpi1 armhf [installed]
```

pjx3, 2021-12-14:
yes sorry, haven't switched to bullseye yet, i'm still on 5.10 / buster since that's when i hit this issue first. 
i have also tried this on 5.15 via rpi-update next, with same result.
i don't use the desktop at all, hence boot to console.

popcornmix, 2021-12-14:
With boot to console I get segfault still on bullseye.
On buster I got:
```
kms_fdto: /home/pi/projects/kms_fdto/kms_fdto/kms_fdto.cpp:143: void createInstance(): Assertion `result == VK_SUCCESS' failed.
Aborted
```

pjx3, 2021-12-14:
heh, weird, it's not even creating an instance, must be library differences.
if you run vulkaninfo, what do you get? (it's in the vulkan-tools package)
here's the first part of mine... 

pi@raspberrypi:~/Downloads/kms_fdto $ vulkaninfo

==========
VULKANINFO
==========

Vulkan Instance Version: 1.1.97

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000148001:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_BLEND_OPERATION_ADVANCED_PROPERTIES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000080000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_PUSH_DESCRIPTOR_PROPERTIES_KHR

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000099000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_DISCARD_RECTANGLE_PROPERTIES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000196000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_DRIVER_PROPERTIES

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000197000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FLOAT_CONTROLS_PROPERTIES

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000028001:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_TRANSFORM_FEEDBACK_PROPERTIES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000218001:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FRAGMENT_DENSITY_MAP_PROPERTIES_EXT

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000177000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_8BIT_STORAGE_FEATURES

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000148000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_BLEND_OPERATION_ADVANCED_FEATURES_EXT

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000082000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FLOAT16_INT8_FEATURES_KHR

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000180000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_SHADER_ATOMIC_INT64_FEATURES

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000028000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_TRANSFORM_FEEDBACK_FEATURES_EXT

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000221000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_SCALAR_BLOCK_LAYOUT_FEATURES

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000218000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FRAGMENT_DENSITY_MAP_FEATURES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000148001:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_BLEND_OPERATION_ADVANCED_PROPERTIES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000080000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_PUSH_DESCRIPTOR_PROPERTIES_KHR

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000099000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_DISCARD_RECTANGLE_PROPERTIES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000196000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_DRIVER_PROPERTIES

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000197000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FLOAT_CONTROLS_PROPERTIES

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000028001:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_TRANSFORM_FEEDBACK_PROPERTIES_EXT

v3dv_GetPhysicalDeviceProperties2: ignored VkStructureType 1000218001:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FRAGMENT_DENSITY_MAP_PROPERTIES_EXT

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000177000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_8BIT_STORAGE_FEATURES

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000148000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_BLEND_OPERATION_ADVANCED_FEATURES_EXT

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000082000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FLOAT16_INT8_FEATURES_KHR

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000180000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_SHADER_ATOMIC_INT64_FEATURES

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000028000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_TRANSFORM_FEEDBACK_FEATURES_EXT

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000221000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_SCALAR_BLOCK_LAYOUT_FEATURES

v3dv_GetPhysicalDeviceFeatures2: ignored VkStructureType 1000218000:VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FRAGMENT_DENSITY_MAP_FEATURES_EXT

Instance Extensions:
====================
Instance Extensions     count = 13
        VK_EXT_debug_report                 : extension revision 10
        VK_EXT_debug_utils                  : extension revision  1
        VK_KHR_device_group_creation        : extension revision  1
        VK_KHR_display                      : extension revision 23
        VK_KHR_external_fence_capabilities  : extension revision  1
        VK_KHR_external_memory_capabilities : extension revision  1
        VK_KHR_external_semaphore_capabilities: extension revision  1
        VK_KHR_get_display_properties2      : extension revision  1
        VK_KHR_get_physical_device_properties2: extension revision  2
        VK_KHR_get_surface_capabilities2    : extension revision  1
        VK_KHR_surface                      : extension revision 25
        VK_KHR_xcb_surface                  : extension revision  6
        VK_KHR_xlib_surface                 : extension revision  6

pjx3, 2021-12-14:
I just updated the code to add a debug call back to print error messages coming from vulkan.

pjx3, 2021-12-14:
thanks, was that running on buster console?
my vulkan (buster) is 1.1.97 vs your 1.2.162 (but it should be backward compat)
the real weirdness is that you have llvmpipe listed as your gpu (which is a software rasteriser afaik) whilst mine is the v3d driver...

	deviceType     = PHYSICAL_DEVICE_TYPE_CPU
	deviceName     = llvmpipe (LLVM 11.0.1, 128 bits)

        deviceType     = INTEGRATED_GPU
        deviceName     = V3D 4.2

i've no idea how that might happen, have you compiled mesa locally??

popcornmix, 2021-12-14:
Not built mesa locally. From desktop glxinfo reports:
```
    Vendor: Broadcom (0x14e4)
    Device: V3D 4.2 (0xffffffff)
```
so the 3d hardware driver is available (from desktop at least).
Kodi (without desktop started) also finds the v3d hardware:
```
2021-12-14 14:37:01.707 T:756      INFO <general>: GL_VENDOR = Broadcom
2021-12-14 14:37:01.707 T:756      INFO <general>: GL_RENDERER = V3D 4.2
2021-12-14 14:37:01.707 T:756      INFO <general>: GL_VERSION = OpenGL ES 3.1 Mesa 20.3.5
2021-12-14 14:37:01.707 T:756      INFO <general>: GL_SHADING_LANGUAGE_VERSION = OpenGL ES GLSL ES 3.10
```

pjx3, 2021-12-14:
ah... i think i know what's up. i built the v3dv driver locally a while back & have been updating it from source ever since.
i think you need to install the mesa-vulkan-drivers package. there's also a mesa-vulkan-drivers-dbgsym which might be useful.

popcornmix, 2021-12-14:
```
pi@pi4:~ $ sudo apt install mesa-vulkan-drivers mesa-vulkan-drivers-dbgsym
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
mesa-vulkan-drivers is already the newest version (20.3.5-1+rpt3+rpi1).
mesa-vulkan-drivers-dbgsym is already the newest version (20.3.5-1+rpt3+rpi1).
```
Seem to already have those.

pjx3, 2021-12-14:
gah! guess it must be the loader then... how do i update my vulkan packages so they're on 1.2 like yours? the package manager only seems to want to serve me up the older ones?

pjx3, 2021-12-14:
hmm... actually think it is the drivers. i just installed that package here and had a look at the files. there's nothing related to broadcom in there. i have this locally..

pi@raspberrypi:~/Downloads/kms_fdto $ ls /usr/share/vulkan/icd.d
broadcom_icd.armv7l.json  radeon_icd.arm.json

popcornmix, 2021-12-14:
> how do i update my vulkan packages so they're on 1.2

If you have a spare sdcard, then install bullseye and just install vulkan from apt. It's a newer version than the buster version.

pjx3, 2021-12-14:
okay, i'll have to get a card but will give it a go & get back to you. sorry for the waste of time! :(

pjx3, 2021-12-14:
sorry to bother again but could you try one last thing for me please? i'm guessing the latest version requires this env var to find the icd...

export VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/broadcom_icd.armv7l.json
kms_fdto/kms_fdto

popcornmix, 2021-12-14:
Ah -
```
./kms_fdto/kms_fdto
```
results in seg fault. But it's different with that defined:
```
pi@pi4:~/projects/kms_fdto $ VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/broadcom_icd.armv7l.json ./kms_fdto/kms_fdto
DEBUG[0]: ReadDataFilesInSearchPaths: Searching the following paths for manifest files: /etc/xdg/vulkan/implicit_layer.d:/etc/vulkan/implicit_layer.d:/usr/local/share/vulkan/implicit_layer.d:/usr/share/vulkan/implicit_layer.d:/home/pi/.local/share/vulkan/implicit_layer.d

DEBUG[0]: ReadDataFilesInSearchPaths: Searching the following paths for manifest files: /etc/xdg/vulkan/explicit_layer.d:/etc/vulkan/explicit_layer.d:/usr/local/share/vulkan/explicit_layer.d:/usr/share/vulkan/explicit_layer.d:/home/pi/.local/share/vulkan/explicit_layer.d

DEBUG[0]: ReadDataFilesInSearchPaths: Searching the following paths for manifest files: /usr/share/vulkan/icd.d/broadcom_icd.armv7l.json

DEBUG[0]: Searching for ICD drivers named /usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so
DEBUG[0]: Build ICD instance extension list
DEBUG[0]: Instance Extension: VK_KHR_external_memory_capabilities (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.1
DEBUG[0]: Instance Extension: VK_KHR_get_physical_device_properties2 (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.1
DEBUG[0]: Instance Extension: VK_KHR_get_surface_capabilities2 (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.1
DEBUG[0]: Instance Extension: VK_KHR_surface (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.25
DEBUG[0]: Instance Extension: VK_KHR_xcb_surface (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.6
DEBUG[0]: Instance Extension: VK_KHR_xlib_surface (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.6
DEBUG[0]: Instance Extension: VK_EXT_debug_report (/usr/lib/arm-linux-gnueabihf/libvulkan_broadcom.so) version 0.0.9
DEBUG[0]: loader_validate_instance_extensions: Instance extension VK_KHR_display not supported by available ICDs or enabled layers.
kms_fdto: /home/pi/projects/kms_fdto/kms_fdto/kms_fdto.cpp:169: void createInstance(): Assertion `result == VK_SUCCESS' failed.
Aborted
```

pjx3, 2021-12-14:
ah thanks for that. it looks like the packaged driver is missing the VK_KHR_display extensions. don't know why that's missing, have to look into it. perhaps the driver devs removed it or the package builder changed the default config??

don't really want to burn a load of time setting up bullseye for no reason (i'm on a cm4 with emmc so can't just smack it on another card, need a lite for that).

bluestang2006, 2021-12-15:
> ah thanks for that. it looks like the packaged driver is missing the VK_KHR_display extensions. don't know why that's missing, have to look into it. perhaps the driver devs removed it or the package [...]

@popcornmix @pjx3 

MESA `20.3.5-1+rpt3+rpi1` from the Bullseye repo is severely outdated and didn't make the cut with the VK_KHR_display extensions so it does not work in KMS, only X11. 

You need to build MESA from upstream to get Vulkan to work in KMS for now.

bluestang2006, 2021-12-15:
@pjx3 something is wrong with your vulkaninfo output...

This is what mine reports, but keep in mind I build my drivers from upstream:
```
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.162

Instance Extensions: count = 14
===============================
	VK_EXT_debug_report                    : extension revision 10
	VK_EXT_debug_utils                     : extension revision 2
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 23
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 2
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 1
=================
VK_LAYER_KHRONOS_validation (Khronos Validation Layer) Vulkan version 1.2.162, layer version 1:
	Layer Extensions: count = 3
		VK_EXT_debug_report        : extension revision 9
		VK_EXT_debug_utils         : extension revision 1
		VK_EXT_validation_features : extension revision 2
	Devices: count = 1
		GPU id = 0 (V3D 4.2)
		Layer-Device Extensions: count = 3
			VK_EXT_debug_marker     : extension revision 4
			VK_EXT_tooling_info     : extension revision 1
			VK_EXT_validation_cache : extension revision 1

Presentable Surfaces:
=====================

Device Groups:
==============
Group 0:
	Properties:
		physicalDevices: count = 1
			V3D 4.2 (ID: 0)
		subsetAllocation = 0

	Present Capabilities:
		V3D 4.2 (ID: 0):
			Can present images from the following devices: count = 1
				V3D 4.2 (ID: 0)
		Present modes: count = 1
			DEVICE_GROUP_PRESENT_MODE_LOCAL_BIT_KHR

Device Properties and Extensions:
=================================
GPU0:
VkPhysicalDeviceProperties:
---------------------------
	apiVersion     = 4198599 (1.1.199)
	driverVersion  = 88485987 (0x5463063)
	vendorID       = 0x14e4
	deviceID       = 0xbe485fd3
	deviceType     = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
	deviceName     = V3D 4.2
[... 364 lines cut ...]
----------------------------------------
	protectedMemory = false

VkPhysicalDeviceSamplerYcbcrConversionFeatures:
-----------------------------------------------
	samplerYcbcrConversion = false

VkPhysicalDeviceShaderDrawParametersFeatures:
---------------------------------------------
	shaderDrawParameters = false

VkPhysicalDeviceUniformBufferStandardLayoutFeaturesKHR:
-------------------------------------------------------
	uniformBufferStandardLayout = true

VkPhysicalDeviceVariablePointersFeatures:
-----------------------------------------
	variablePointersStorageBuffer = true
	variablePointers              = false

VkPhysicalDeviceVertexAttributeDivisorFeaturesEXT:
--------------------------------------------------
	vertexAttributeInstanceRateDivisor     = true
	vertexAttributeInstanceRateZeroDivisor = false
```

pjx3, 2021-12-15:
@bluestang2006 thanks for confirming. those ignored structs warnings only occur if i build mesa in debug config not in release so perhaps that's the same for you? would be interested to know.

anyway, i'm on a fresh bullseye image now. got latest packages & rebuilt everything. i had to change the audio device to "default" in my code but now it runs fine as long as you boot from cli and have built the mesa driver yourself.

so it looks like it's down to the new PulseAudio stuff. ALSA seems to be going through that now and with it KMS & snd_pcm_writei seem happy (no time outs). i need to test it more thoroughly with actual audio but will close this issue for now.

thanks for the help. that was a lot of faffing around to find it but glad it's sorted now, phew...

closed 2021-12-15 by pjx3