raspberrypi/linux #4779: hdmi_pixel_encoding=2 full RGB is short lived

Ulysses1337, 2021-12-17:
**Is this the right place for my bug report?**
Presumably, because it is driver related.

**Describe the bug**
hdmi_pixel_encoding is not handled properly in KMS.

**To reproduce**
Take a fresh install of Bullseye Lite and a display which defaults to limited RGB. Configure it with hdmi_pixel_encoding=2.

**Expected behaviour**
The console is rendered in deep black and stays that way.

**Actual behaviour**
The first stage looks good. When the 4 raspberries disappear, the display mode seems to reset. The content reappears in 16-235 as it finishes the boot sequence. The limitation stays as I proceed with my work.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
`Raspberry Pi 4 Model B Rev 1.2`
* Which OS and version (`cat /etc/rpi-issue`)?
`Raspberry Pi reference 2021-10-30`
`Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage4`
* Which firmware version (`vcgencmd version`)?
`Nov 18 2021 16:16:49 `
`Copyright (c) 2012 Broadcom`
`version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)`
* Which kernel version (`uname -a`)?
`Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux`

**Additional context**
Buster worked fine with this config option. If I take a Buster install and make it use KMS, it has the same issue.

JamesH65, 2021-12-18:
Most of the hdmi_* options in config.txt do not have any effect in KMS. KMS takes control of the entire display system, whilst those config.txt entries are implemented in the firmware stack, which is not used in KMS.

There may be a KMS equivalent? But I am not sure what it might be, This is probably worth a read. https://www.kernel.org/doc/html/v5.1/gpu/drm-kms.html

Ulysses1337, 2021-12-18:
Thanks James. There is a KMS property called Broadcast RGB. I take it I'd need to build the driver to try anything myself. :scream:

6by9, 2021-12-20:
Broadcast RGB is currently an Intel custom property. There is ongoing discussion in mainline as to how this should be supported generically.

KMS properties can be listed with ```xrandr --prop```, and then set with ```xrandr --output HDMI-1 --set <prop_name> <value>```, or ```proptest```.

Ulysses1337, 2021-12-22:
That's handy to know. I confirm the lack of that property or any others of interest on my VideoCore. Time to relax and wait for upstream.

popcornmix, 2023-01-24:
I think this issue may be resolved by https://github.com/raspberrypi/linux/pull/5249
@Ulysses1337 do you agree?

Ulysses1337, 2023-01-24:
Not sure. vc4_hdmi_output_format doesn't quite line up with the values for hdmi_pixel_encoding.

6by9, 2023-01-25:
hdmi_pixel_encoding merged selecting RGB vs YCbCr 444, and full range vs limited range.

KMS has "Broadcast RGB" to select between full and limited range, and "output format" to select between RGB, YCbCr 4:4:4, and YCbCr 4:2:2.
What options that hdmi_pixel_encoding exposed do you believe can not be achieved with those 2 properties?

Ulysses1337, 2023-01-25:
I didn't know Intel's Broadcast was brought into the fold. If BRGB works on VideoCore and applies to YUV as well, that ought to cover all the combos.

6by9, 2023-01-25:
Mainline still haven't decided on how to select limited/full range, so support for "Broadcast RGB" was added in #4850 (with minor fixups afterwards).

Likewise they haven't a standardised way of selecting RGB/YCbCr444/YCbCr422, so "Output format" from #5249 added that in as a proprietary property..

Ulysses1337, 2023-01-25:
Sounds promising. Will I need to call xrandr at every startup, or can I permanently configure the encoding? My use case lacks a desktop environment.

6by9, 2023-01-25:
Currently it would need to be via xrandr or modetest.

In theory you can set properties via the "video=" entry on the kernel command line, but only certain (generic) strings are currently handled - https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/gpu/drm/drm_modes.c#L1637
It'd be nice to extend that so that the driver could match other properties on the connector, but that really needs a structure name change as it becomes less and less ```struct drm_cmdline_mode``` and mode ```struct drm_cmdline_connector_props```.

Throwing
```modetest -M vc4 -w 32:"Output format":3```
into .profile seems to work if not booting X.
(Note that the 32 to identify connector HDMI-1 may change, however there isn't a nice way through mode test to specify the connectors. ```kmstest -c 0 --property="Output format"=3``` is far nicer, but throws up a test pattern which requires a key press to quit from. A hybrid app would be nice.)

Ulysses1337, 2023-01-25:
All righty, many thanks. I'll give it a whirl when the next Pi OS lands.