raspberrypi/linux #4781: Can I use USB in RPI-4.19.y-rt?

yangfan852219770, 2021-12-20:
I used the 4.19-rt kernel, when I plugged USB, I could not find the device. In addtion, the **usb-storage** module isn't included in **lsmod | grep -i usb** command.

burakgerz, 2021-12-25:
Did you follow exactly this guide?
https://www.raspberrypi.com/documentation/computers/linux_kernel.html

Most probably you did not deploy the compiled kernel modules to your raspberry?

yangfan852219770, 2021-12-28:
> Did you follow exactly this guide? https://www.raspberrypi.com/documentation/computers/linux_kernel.html
> 
> Most probably you did not deploy the compiled kernel modules to your raspberry?

Thank you! I have solved this problem by re-installing the system.

closed 2021-12-28 by yangfan852219770