raspberrypi/linux #4784: Full Update - Internet not accessible!!

MrSiO, 2021-12-21:
**Describe the bug**
Trying to use the FULL UPDATE button to upgrade my piCorePlayer with no success lately.

Tried from:
- RPi Zero W  (running 6.0.0)
- RPi 3 (running 6.0.0)
- RPi 4 (running 8.1.0)

I alway get the following error:
`Internet not accessible!!`

Tried both with ehternet and wifi.
No firewall involved.  
From SSH I can issues WGET all over the internet so internet acces is fine from RPi's.

**To reproduce**
I can reproduce anytime by clicking the FULL UPDATE button on all 3 of my RPi's.

**Expected behaviour**
It should update as it used to.  

**Actual behaviour**
After about 30 seconds I get a timeout with error message: ` Internet not accessible!!`

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?  `RPi Zero W & RPi3 & RPi4`
* Which OS and version (`cat /etc/rpi-issue`)?  `6.0.0  &  8.1.0`
* Which kernel version (`uname -a`)?
`Linux RPi-LMS-Player1-Sonette 4.19.105-pcpCore_v7 #1 SMP Sat Feb 22 00:32:02 EST 2020 armv7l GNU/Linux
`

**Logs**
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.105-pcpCore_v7 (pCP@SoundStation) (gcc version 7.4.1 20181213 (crosstool-NG 1.24.0.6-afaf7b9)) #1 SMP Sat Feb 22 00:32:02 EST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] On node 0 totalpages: 255488
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xb0/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36748 r8192 d20596 u65536
[    0.000000] pcpu-alloc: s36748 r8192 d20596 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:F8:60:28 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  host=RPi-LMS-Player1-Sonette tz=EST5EDT,M3.2.0,M11.1.0 dwc_otg.fiq_fsm_mask=0xF dwc_otg.lpm_enable=0 console=tty1 root=/dev/ram0 elevator=deadline rootwait quiet nortc loglevel=3 noembed smsc95xx.turbo_mode=N noswap consoleblank=0 waitusb=2
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 985292K/1021952K available (8192K kernel code, 626K rwdata, 2184K rodata, 1024K init, 821K bss, 28468K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 627 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26373 entries in 78 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[... 213 lines cut ...]
[   13.561432] Bluetooth: HCI socket layer initialized
[   13.561452] Bluetooth: L2CAP socket layer initialized
[   13.561507] Bluetooth: SCO socket layer initialized
[   13.572866] Bluetooth: HCI UART driver ver 2.3
[   13.572884] Bluetooth: HCI UART protocol H4 registered
[   13.573066] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.574465] Bluetooth: HCI UART protocol Broadcom registered
[   13.689250] uart-pl011 3f201000.serial: no DMA platform data
[   14.194998] random: crng init done
[   14.882858] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   15.128502] Under-voltage detected! (0x00050005)
[   18.510310] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   18.510319] Bluetooth: BNEP filters: protocol multicast
[   18.510335] Bluetooth: BNEP socket layer initialized
[   18.533637] Bluetooth: RFCOMM TTY layer initialized
[   18.533662] Bluetooth: RFCOMM socket layer initialized
[   18.533696] Bluetooth: RFCOMM ver 1.11
[   18.782220] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
[   25.528479] Voltage normalised (0x00000000)
[   35.738721] Finished piCorePlayer v6.0.0 startup.
`

**Additional context**
Used to work a while ago...
Any other logs I could get to to see why internet access is timing out?  Thx !

JamesH65, 2021-12-21:
Pi Core player is not a Raspberry Pi supplied OS, so unless you see the same issue on our Raspberry Pi OS, I'd suggest approaching the PiCorePlayer devs for help, as the problem is almost certainly at their end.

https://docs.picoreplayer.org/

MrSiO, 2021-12-21:
> Pi Core player is not a Raspberry Pi supplied OS, so unless you see the same issue on our Raspberry Pi OS, I'd suggest approaching the PiCorePlayer devs for help, as the problem is almost certainly at their end.
> 
> https://docs.picoreplayer.org/

Thx.
Will do.

closed 2021-12-21 by MrSiO