raspberrypi/linux #4785: ffmpeg rtsp playback failure using h264_v4l2m2m

acostach, 2021-12-21:
Hi @6by9 , @pelwell ,

One of our users encountered this issue where playback of rtsp streams fails with 
`Error while decoding stream #0:0: Resource temporarily unavailable`
on Pi4 64bit with newer kernel versions like 5.10.78, while it was not failing with 5.10.31.

The debug log of the playback failure from our user is attached here: https://gist.github.com/acostach/bfd41e2976f40533a290310d25416a08

We can also reproduce the playback failure with a demo stream:
ffmpeg -c:v h264_v4l2m2m -i "rtsp://rtsp.stream/pattern" -filter:v fps=1 -f rawvideo -pix_fmt nv12 test.out

```
root@4d0a24f:~# balena run --rm   --device /dev/video10   --device /dev/video11   --device /dev/video12   --device /dev/bus/usb   --entrypoint ffmpeg linuxserver/ffmpeg -loglevel debug  -c:v h264_v4l2m2m -i "rtsp://rtsp.stream/pattern" -filter:v fps=1 -f rawvideo -pix_fmt nv12 test.out
ffmpeg version 4.4 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --disable-debug --disable-doc --disable-ffplay --enable-ffprobe --enable-avresample --enable-gpl --enable-libaom --enable-libass --enable-libfdk_aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libxml2 --enable-libx264 --enable-libx265 --enable-libxvid --enable-nonfree --enable-openssl --enable-small --enable-stripping --enable-version3
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'h264_v4l2m2m'.
Reading option '-i' ... matched as input url with argument 'rtsp://rtsp.stream/pattern'.
Reading option '-filter:v' ... matched as option 'filter' (set stream filtergraph) with argument 'fps=1'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'nv12'.
Reading option 'test.out' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url rtsp://rtsp.stream/pattern.
Applying option c:v (codec name) with argument h264_v4l2m2m.
Successfully parsed a group of options.
Opening an input file: rtsp://rtsp.stream/pattern.
[tcp @ 0x5595308d60] No default whitelist set
[tcp @ 0x5595308d60] Original list of addresses:
[tcp @ 0x5595308d60] Address 23.88.67.97 port 554
[tcp @ 0x5595308d60] Address 2a01:4f8:272:581d::2 port 554
[tcp @ 0x5595308d60] Interleaved list of addresses:
[tcp @ 0x5595308d60] Address 23.88.67.97 port 554
[tcp @ 0x5595308d60] Address 2a01:4f8:272:581d::2 port 554
[tcp @ 0x5595308d60] Starting connection attempt to 23.88.67.97 port 554
[tcp @ 0x5595308d60] Successfully connected to 23.88.67.97 port 554
[rtsp @ 0x5595305fc0] SDP:
v=0
o=- 0 0 IN IP4 127.0.0.1
s=Stream
c=IN IP4 0.0.0.0
t=0 0
m=video 0 RTP/AVP 96
b=AS:2097
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1;profile-level-id=42c015;sprop-parameter-sets=Z0LAFdkBQfsBagwCC0oAAAMAAgAAAwB5HixckA==,aMuMsg==
a=control:trackID=0
m=audio 0 RTP/AVP 97
a=rtpmap:97 MP4A-LATM/44100
a=fmtp:97 cpresent=0;config=40002410adca00
a=control:trackID=1

[rtsp @ 0x5595305fc0] video codec set to: h264
[rtsp @ 0x5595305fc0] RTP Packetization Mode: 1
[rtsp @ 0x5595305fc0] RTP Profile IDC: 42 Profile IOP: c0 Level: 15
[rtsp @ 0x5595305fc0] Extradata set to 0x559530cd10 (size: 40)
[... 632 lines cut ...]
  Metadata:
    title           : Stream
  Duration: N/A, start: -0.363968, bitrate: N/A
  Stream #0:0: Video: h264, yuv420p(tv, bt470bg/smpte170m/bt709, progressive), 320x240 [SAR 1:1 DAR 4:3], 30 fps, 30 tbr, 90k tbn, 60 tbc
  Stream #0:1: Audio: aac, 44100 Hz, mono, fltp
[h264_v4l2m2m @ 0x55947389a0] Using device /dev/video10
[h264_v4l2m2m @ 0x55947389a0] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x55947389a0] requesting formats: output=H264 capture=YU12
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> rawvideo (native))
Press [q] to stop, [?] for help
Output #0, rawvideo, to 'test.out':
  Metadata:
    title           : Stream
    encoder         : Lavf58.76.100
  Stream #0:0: Video: rawvideo (NV12 / 0x3231564E), nv12(tv, progressive), 320x240, q=2-31, 921 kb/s, 1 fps, 1 tbn
    Metadata:
      encoder         : Lavc58.134.100 rawvideo
frame=   24 fps=1.2 q=-0.0 size=    2560kB time=00:00:24.00 bitrate= 873.8kbits/s dup=8 drop=0 speed=1.25x 

root@4d0a24f:~# uname -r
5.10.87-v8 <- with b7e6b495eff31298ba4665f71b2414cc9a8f99c2 reverted
```

What would be the recommended approach here? Thank you

acostach, 2021-12-22:
Do you think we should revert this change on our side for now @6by9 or is it something that can perhaps be mitigated in a different way? Thank you

pelwell, 2021-12-22:
Unless @6by9 thinks otherwise I would say this is unlikely to be investigated until January. If you have a version of the kernel that known to not have the problem then you may want to advise users to install it and not update until told to do so.

acostach, 2022-01-20:
Hi @6by9 , leaving a ping here to remind you about this. Thanks!

6by9, 2022-02-02:
Tested on 5.10.95 and 5.15.18 32bit kernels and both are fine.

I did have an extra patch on top of 5.10.95 that did then have issues, even though it shouldn't have made any real difference.
I'm checking for an uninitialised variable that might cause these sort of random changes.

acostach, 2022-02-02:
Thanks for looking into this @6by9 , I tried again now on the Pi4 with kernel 5.10.94 and still see this:

```
Error while decoding stream #0:0: Resource temporarily unavailable
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
    Last message repeated 2 times
Error while decoding stream #0:0: Resource temporarily unavailable
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
    Last message repeated 1 times
Error while decoding stream #0:0: Resource temporarily unavailable
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
^C^C^C^C^CReceived > 3 system signals, hard exiting
root@dd1ecc1:/tmp# uname -r
5.10.94-v8
```

6by9, 2022-02-02:
My extra patch did have a bug, but it looks like the driver is doing the right thing if FFmpeg have changed to follow the spec.

Raspberry Pi OS is only on FFmpeg 4.3.3, so may not have been affected in the same way as your 4.4.

acostach, 2022-02-03:
Thank you very much, we'll try the newer revision and get back. Do we need to include any extra patch for the bug you mentioned?

6by9, 2022-02-03:
> Do we need to include any extra patch for the bug you mentioned?

No, the patch with a bug hasn't been pushed anywhere yet. It's only handling cleanup on an error path.

acostach, 2022-03-01:
Thanks again for your help, using https://github.com/FFmpeg/FFmpeg/commit/3fc72c9fc16df3b7edc329ed9eeaaa1e65976c33 solved the problem.

closed 2022-03-01 by acostach