raspberrypi/linux #4786: On Pi 4 sshd does not start when no display device is created

cnighswonger, 2021-12-21:
**Describe the bug**
sshd is not started when no display device is created

**To reproduce**
- Start with a fresh image of Raspian OS with Desktop
- Enable sshd
- Reboot with no monitors plugged into either hdmi port
- Note failed attempts to ssh into the Pi
- Reboot with monitor/keyboard plugged in
- Edit /boot/config.txt and uncomment hdmi_force_hotplug=1 to force creation of a display device
- Reboot with monitor unplugged
- Note successful attempts to ssh into the Pi

**Expected behaviour**
Once sshd is enabled it should start regardless of the state of existence of any display device.

**Actual behaviour**
Once sshd is enabled it does not start if there is no display device created.

**System**
* `Raspberry Pi 4 Model B Rev 1.2`
* `Raspberry Pi reference 2020-05-27 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2`
* `Dec  1 2021 15:01:54 Copyright (c) 2012 Broadcom version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)`
* `Linux pi-ctlr 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux`

**Logs**
Specify what log data is needed, and I will post it.

**Additional context**
None at this time.

HughSparks, 2022-02-10:
I can confirm this is happening to me too.

6by9, 2022-02-10:
There has been a significant kernel bump since this report, and a new Raspberry Pi OS image released.
Please confirm that your system is up to date - ```uname -a``` should report 5.10.94 IIRC.

```hdmi_force_hotplug=1``` only forces enabling of HDMI with the firmware drivers. Once vc4-kms-v3d takes over it is irrelevant.

HughSparks, 2022-02-12:
My system is up to date as it was a fresh install with the rpi flashing software. The firmware is also up to date. My kernel is 5.10.92-v8+.  I attempted a full upgrade but no change.