raspberrypi/linux #4790: Power/Activity LED adjustment possible on CM4 w/ dfrobot IoT Router Board Mini running OpenWRT 21.02.1

satmandu, 2021-12-23:
**Is this the right place for my bug report?**

Apologies if this is the wrong place for this bug report, but I just wanted to know if what I was doing with the stock OpenWRT `5.4.154` kernel was expected to fail.

I have a `CM4108032` in a [dfrobot IoT Router Board Mini](https://wiki.dfrobot.com/Compute_Module_4_IoT_Router_Board_Mini_SKU_DFR0767) running stock OpenWRT 21.02.1.

I'm trying to get the Power and Activity LEDs to turn off after the system boots up. Many OpenWRT systems have LED toggles which can be adjusted (AFAIK) through sysfs toggles. On my RPI4B systems I just use the config.txt options to handle this. Does this work for CM4 based boards, or is each board's implementation so different I should just approach DFRobot support about this? Is this dependent upon having specific support wired into the kernel which isn't available on this older and/or upstream kernel?

Obviously trying a kernel from https://github.com/raspberrypi/rpi-firmware would be my next step, but I thought it was worth asking about...

**Describe the "bug"**
Added this to `/boot/config.txt`
```
# Disable the PWR LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
# Disable the Activity LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
```
Rebooted and Power and Activity LEDs do not turn off after boot.

**Expected behaviour**
Power and Activity LEDs turn off after boot.

**Actual behaviour**
Power and Activity LEDs stay on after boot.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `CM4108032`
* Which OS and version (`cat /etc/rpi-issue`)? `OpenWRT 21.02.1` running OpenWRT's `5.4.154`
```
cat /etc/os-release
NAME="OpenWrt"
VERSION="21.02.1"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.1"
VERSION_ID="21.02.1"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16325-88151b8303"
OPENWRT_BOARD="bcm27xx/bcm2711"
OPENWRT_ARCH="aarch64_cortex-a72"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.1 r16325-88151b8303"
```
* Which firmware version (`vcgencmd version`)?
```
vcgencmd version
Feb 16 2021 14:17:31
Copyright (c) 2012 Broadcom
version d6d82cf99bcb3e9a166a34cfde53130957a36bd3 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
uname -a
Linux office 5.4.154 #0 SMP Sun Oct 24 09:01:35 2021 aarch64 GNU/Linux
```

pelwell, 2021-12-23:
That's s difficult question to answer without the hardware or a schematic, but you can start by taking a look in /sys/class/leds to see what is defined (and try changing the trigger, etc.). Next you can use gpioset on each of the accessible GPIOs (0-13, 16-27 avoiding the UART) to hunt for other LED controls.

satmandu, 2021-12-24:
Thanks @pelwell, that sounds like a good place to start.

closed 2021-12-24 by satmandu

satmandu, 2021-12-24:
For anyone who comes here for future reference `/sys/class/leds/led0/brightness` appears to handle the green LED.
`echo 0 > /sys/class/leds/led0/brightness` appears to turn it off.

Also probably relevant to https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/114 .