raspberrypi/linux #4791: TC358743 driver doesn't get loaded with new DT overlays

JakubVanek, 2021-12-24:
**Describe the bug**

TC358743 doesn't show up in dmesg and the V4L2 hierarchy when the latest `tc358743.dtbo` is used.

* This version works: https://github.com/raspberrypi/rpi-firmware/blob/9a25ccebfd9e2ff9116af665e782b3dba1d8a201/overlays/tc358743.dtbo
* This version doesn't work: https://github.com/raspberrypi/rpi-firmware/blob/0ac1538ab9776167d461ce7d86153ddbf975167d/overlays/tc358743.dtbo

This seems to be related to the recent CAM0 changes - https://github.com/raspberrypi/linux/commit/131f1322039284932ccb601a5cffdd9ca5d36d96.

**To reproduce**

 1. Set up a RPi4 with a TC358743 on CAM1 and `dtoverlay=tc358743` or `dtoverlay=tc358743,4lane=1` in config.txt
 2. Replace /boot/overlays/tc358743.dtbo with https://github.com/raspberrypi/rpi-firmware/blob/0ac1538ab9776167d461ce7d86153ddbf975167d/overlays/tc358743.dtbo

**Expected behaviour**

The bridge shows up in the kernel device hierarchy and an informational message appears in dmesg:
```
[    7.304488] tc358743 10-000f: tc358743 found @ 0x1e (i2c-22-mux (chan_id 1))
```

**Actual behaviour**

TC358743 doesn't get registered in the V4L2 API (i.e. the media controller node of Unicam is empty). Also no message about the driver is printed in dmesg. However, the device tree overlay does get loaded - the tc358743 node appears in `/sys/firmware/devicetree/`.

**System**

* Which model of Raspberry Pi? `Raspberry Pi Compute Module 4 Rev 1.0` and `Raspberry Pi 4 Model B Rev 1.4`
* Which OS and version? Currently a custom Buildroot 2021.11-based OS, I can retest with Raspbian if needed
* Which firmware version?
  ```
  [    0.337336] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-12-22T14:24:35, variant start_x
  [    0.357436] raspberrypi-firmware soc:firmware: Firmware hash is 720889ee7c970afe516868d20515a73892f9c127
  ```
* Which kernel version? Local build of 5.10.88 (commit [606d1fc3bc97a8be1d21e143ca88f3ba085ddef1](https://github.com/raspberrypi/linux/commits/606d1fc3bc97a8be1d21e143ca88f3ba085ddef1))

**Logs**

<details>
<summary>dmesg with old tc358743.dtbo</summary>
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.88-v7l (builder@xxx) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot kz-2021-12-t12) 10.3.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Fri Dec 24 01:57:34 CET 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] random: fast init done
[    0.000000] earlycon: pl11 at MMIO32 0x00000000fe201000 (options '')
[    0.000000] printk: bootconsole [pl11] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002c000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 308224 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50892 r8192 d22836 u81920
[    0.000000] pcpu-alloc: s50892 r8192 d22836 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[... 578 lines cut ...]
[    6.776067] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[    6.776113] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[    6.918831] rtc-pcf85063 10-0051: registered as rtc0
[    6.921355] rtc-pcf85063 10-0051: setting system clock to 2021-12-24T14:46:46 UTC (1640357206)
[    7.162532] EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode. Opts: commit=300
[    7.162584] ext4 filesystem being mounted at /mnt/persist supports timestamps until 2038 (0x7fffffff)
[    7.361886] systemd-journald[140]: Received client request to flush runtime journal.
[   10.021674] random: crng init done
[   10.021682] random: 7 urandom warning(s) missed due to ratelimiting
[   13.361809] audit: type=1334 audit(1640357212.929:4): prog-id=7 op=LOAD
[   13.361880] audit: type=1334 audit(1640357212.929:5): prog-id=8 op=LOAD
[   13.409365] NET: Registered protocol family 38
[   13.521359] cryptd: max_cpu_qlen set to 1000
[   13.722217] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   13.735639] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   13.736639] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   13.737514] bcmgenet fd580000.ethernet eth0: Link is Down
[   16.832794] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   16.832829] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.232679] cam-dummy-reg: disabling
[   33.232700] cam1-reg: disabling
[   43.982883] audit: type=1334 audit(1640357242.628:6): prog-id=8 op=UNLOAD
[   43.982907] audit: type=1334 audit(1640357242.628:7): prog-id=7 op=UNLOAD
```
</details>

JakubVanek, 2021-12-24:
Testing with `dtdebug=1` in config.txt gives me the following VC debug logs:

<details>
<summary>new tc358743.dtbo</summary>

```
004076.133: arasan: arasan_emmc_open
004229.842: brfs: File read: /mfs/sd/config.txt
004232.847: brfs: File read: 678 bytes
004286.513: HDMI1:EDID error reading EDID block 0 attempt 0
004296.029: HDMI1:EDID error reading EDID block 0 attempt 1
004305.536: HDMI1:EDID error reading EDID block 0 attempt 2
004315.051: HDMI1:EDID error reading EDID block 0 attempt 3
004324.561: HDMI1:EDID error reading EDID block 0 attempt 4
004334.075: HDMI1:EDID error reading EDID block 0 attempt 5
004343.582: HDMI1:EDID error reading EDID block 0 attempt 6
004353.096: HDMI1:EDID error reading EDID block 0 attempt 7
004362.606: HDMI1:EDID error reading EDID block 0 attempt 8
004372.120: HDMI1:EDID error reading EDID block 0 attempt 9
004376.618: HDMI1:EDID giving up on reading EDID block 0
004382.742: brfs: File read: /mfs/sd/config.txt
004387.498: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
004394.198: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
004422.335: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
004426.965: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
004434.705: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
004442.156: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
004998.421: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
005004.060: *** Restart logging
005006.704: brfs: File read: 678 bytes
005011.350: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
005025.028: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
005035.068: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
005045.098: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
005055.136: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
005065.169: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
005075.208: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
005085.238: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
005095.276: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
005105.309: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
005115.347: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
005120.369: hdmi: HDMI1:EDID giving up on reading EDID block 0
005130.992: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
005141.031: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
005151.061: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
005161.099: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
005171.132: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
005181.170: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
005191.201: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
005201.239: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
005211.272: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
005221.310: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
005226.331: hdmi: HDMI1:EDID giving up on reading EDID block 0
005231.929: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
005240.694: HDMI0: hdmi_pixel_encoding: 300000000
005246.161: HDMI1: hdmi_pixel_encoding: 300000000
005252.106: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
005263.618: dtb_file 'bcm2711-rpi-cm4.dtb'
005265.643: Trying Device Tree file 'bcm2711-rpi-cm4.dtb'
005277.635: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
005280.879: Loading 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xcbd8
005297.637: dtdebug: delete_node(/__local_fixups__)
005303.519: brfs: File read: 52184 bytes
005310.528: dtdebug: using platform 'bcm2711'
005313.892: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
005320.127: dtdebug: overlay map loaded
005325.329: dtdebug: /aliases:i2c_vc=i2c0
005333.788: dtdebug: /__symbols__:i2c_vc=i2c0
005339.616: dtdebug: /__overrides__:i2c_vc=i2c0
005349.333: dtdebug: /__overrides__:i2c_vc_baudrate=i2c0_baudrate
[... 466 lines cut ...]
006535.322: dtdebug:   +prop(status)
006540.199: dtdebug: merge_fragment() end
006556.264: brfs: File read: 7238 bytes
006558.830: brfs: File read: /mfs/sd/cmdline.txt
006563.434: Read command line from file 'cmdline.txt':
006569.312: 'earlycon=pl011,mmio32,0xfe201000 console=ttyAMA0,115200 elevator=deadline init=/sbin/overlay-init'
006653.228: dtparam: arm_freq=1500000000
006658.042: dtdebug: found override arm_freq
006660.274: dtparam: core_freq=500000000
006667.937: dtdebug: /__overrides__ has no core_freq property
006671.621: Unknown dtparam 'core_freq' - ignored
006689.613: dtdebug: delete_node(/hat)
006697.438: brfs: File read: 98 bytes
006738.297: brfs: File read: /mfs/sd/u-boot.bin
006740.762: Loading 'u-boot.bin' to 0x8000 size 0x6d25c
006746.763: Device tree loaded to 0x2eff2c00 (size 0xd374)
006755.337: uart: Set PL011 baud rate to 103448.300000 Hz
006762.049: uart: Set PL011 baud rate to 103448.300000 Hz
006762.133: uart: Baud rate change done...
006764.190: uart: Baud rate change done...
006769.705: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
014100.136: vchiq_core: vchiq_init_state: slot_zero = 0xec100000, is_master = 1
```

</details>

JakubVanek, 2021-12-28:
Found it! The `cam1_clk` node still had `status = "disabled"` for some reason. Adding that to the overlay seems to have fixed the problem.

pelwell, 2021-12-28:
Cool. We're still very much in holiday mode, but this looks like an easy fix.

closed 2021-12-28 by pelwell

BillTec-CyberSec, 2023-03-09:
I trying to use a raspberry pi 4 model B running LineageOS 18.1 (Android 11) with HDMI to CSI-2 (C790) but when i add dtoverlay=tc358743,4lane and dtoverlay=tc358743-audio in /boot/config.txt the camera application of LineageOS disappears and the raspberry pi 4 no more recognize the C790

The output of dmesg dont have nothing about tc358743

When i remove dtoverlay=tc358743,4lane and dtoverlay=tc358743-audio the raspberry recognize one camera device (/dev/video0) but the driver dont work

Anyone can help me to solve this?

Sorry about my english i am brasilian...

And sorry if i'm in the wrong channel

6by9, 2023-03-09:
Posting on a closed issue is not the best way to get support.
You've also commented on the forums (again on a partially unrelated thread) - please use that thread https://forums.raspberrypi.com/viewtopic.php?p=2088614

BillTec-CyberSec, 2023-03-09:
> Postar em um problema fechado não é a melhor maneira de obter suporte. Você também comentou nos fóruns (novamente em um tópico parcialmente não relacionado) - use esse tópico https://forums.raspberrypi.com/viewtopic.php?p=2088614

So sorry, I'm new to the forums

PR #4793 (2021-12-28): dtoverlays: Enable cam1_clock when using tc358743
This fixes a regression introduced in 131f1322039284932ccb601a5cffdd9ca5d36d96 (see also https://github.com/raspberrypi/linux/issues/4791). The tc358743 driver refused to bind to the device.

The new unified cam1_clk node that represents the fixed on-board oscillator is marked as disabled by default. The tc358743 overlay didn't expect this and so the clock node was stuck in disabled state.

This commit just adds the required status = "okay" line. Other sensor drivers do this too.

Fixes: #4791