raspberrypi/linux #4798: memory corruption when using memset32() on RPi Zero

npitre, 2021-12-31:
**Describe the bug**

Any configuration enabling CONFIG_BCM2835_FAST_MEMCPY pulls in
an alternative `memset()` implementation to be found in
arch/arm/lib/memset_rpi.S.

This alternative implementation aliases `__memset32()` and `__memset64()`
to `memset()` which is totally wrong, resulting in actual memory corruption
whenever `memset32()` and/or `memset64()` is used.

**Expected behaviour**

`memset32()` is used to fill memory with 32-bit values.
Likewise for `memset64()`.

**Actual behaviour**

The low 8 bits of provided value are repeated like `memset()` should do.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

PiZeroW

* Which OS and version (`cat /etc/rpi-issue`)?

All of them.

* Which firmware version (`vcgencmd version`)?

N/A

* Which kernel version (`uname -a`)?

All kernels since commit 385b89c3588d.

**Additional context**

I think the inclusion of an alternative `memset()` in a commit titled
"Improve __copy_to_user and __copy_from_user performance" was wrong
in the first place. There is nothing really that I can see to justify
an alternative implementation for ARMv6 here either, even if it hadn't
broken `memset32()` which is unacceptable.

marcone, 2022-01-10:
The `count` parameter for memset32 and memset64 is also the number of uint32 or uint64 to store, not the number of bytes, so by aliasing it to memset it's not even going to fill the entire block of memory.
This seems like it could be a security issue.

marcone, 2022-01-10:
~~The count parameter for memset32 and memset64 is also the number of uint32 or uint64 to store, not the number of bytes, so by aliasing it to memset it's not even going to fill the entire block of memory.~~
(scratch that, this is already accounted for when memset32 calls __memset32)
Still seems like filling memory with the wrong data could be a security issue though.

pelwell, 2022-01-11:
If you are in a position to build your own kernel, have a look at https://github.com/raspberrypi/linux/pull/4820 which should fix the problem. Tested with a hacked Kconfig on Pi 4, but should work as-is on Pi Zero etc.

popcornmix, 2022-01-24:
@npitre okay to close?

closed 2022-01-24 by npitre