raspberrypi/linux #4803: [kms] no audio in Kodi when booted with HDTV switched off

graysky2, 2022-01-03:
**Describe the bug**
Booting with the attached HDTV switched off results in no audio when attempting to playback in Kodi

**To reproduce**
1. Turn HDTV off
2. Reboot
3. Load Kodi
4. Turn HDTV on
5. Try to playback a video file

**Expected behaviour**
Audio output options should be detected regardless of HDTV's power state when booting

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi4B Rev 1.4
* Which OS and version? Arch ARM aarch64
* Which firmware version (`vcgencmd version`)?
```
Dec 22 2021 14:23:44 
Copyright (c) 2012 Broadcom
version 720889ee7c970afe516868d20515a73892f9c127 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux submyth 5.10.89-1-rpi-ARCH #1 SMP PREEMPT Fri Dec 31 18:16:21 UTC 2021 aarch64 GNU/Linux
```

**Logs**
[dmesg](https://github.com/raspberrypi/linux/files/7801877/dmesg.txt)
[kodi.log](https://github.com/raspberrypi/linux/files/7801878/kodi.log.txt)

**Additional context**
Here is my `/boot/config.txt`:
```
# Run as fast as firmware / board allows
arm_boost=1

# Kodi
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1
```

I am booting with the TV's extracted edid placed in `/usr/lib/firmware/edid.dat` and with this in my `/boot/cmdline.txt`:
```
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 drm.edid_firmware=edid.dat video=HDMI-A-1:D
```

For the record:
```
% base64 /usr/lib/firmware/edid.dat
AP///////wAebQEAAQEBAQEZAQOAoFp4Cu6Ro1RMmSYPUFShCAAxQEVAYUBxQIGAAQEBAQEBAjqA
GHE4LUBYLEUAQIRjAAAeZiFQsFEAGzBAcDYAQIRjAAAeAAAA/QA6Ph5TEAAKICAgICAgAAAA/ABM
RyBUVgogICAgICAgAeYCAxnxSJAiIAUEAwIBIwlXB2cDDAAgAIAeAjqAGHE4LUBYLAQFQIRjAAAe
AR2AGHEcFiBYLCUAQIRjAACeAR0AclHQHiBuKFUAQIRjAAAejArQiiDgLRAQPpYAQIRjAAAYAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA+A==
```

HiassofT, 2022-01-03:
Can you give the "drm/vc4: hdmi: Fix HDMI monitor detection in polled mode" commit from #4794 a try?

This reads a bit like RPi4 might also run into the hpd issue I noticed on RPi3

graysky2, 2022-01-03:
@HiassofT - I built https://github.com/raspberrypi/linux/commit/1d6957db7f835469ceab856ef255bdbdf16ddb5e patched with https://github.com/raspberrypi/linux/pull/4794, booting into it fixes this bug.  Thank you for pointing it out :+1:

popcornmix, 2022-01-04:
Fix in now in rpi-update kernel.

graysky2, 2022-01-04:
Thanks again, guys :beers:

closed 2022-01-04 by graysky2