raspberrypi/linux #4804: Bulseye background tear.

audetto, 2022-01-03:
**Describe the bug**
Bullseye background tear.

**To reproduce**
Just install bullseye, update, no other changes.
Move a window around and you will notice a little tear on the Desktop Background picture especially on the top right corner, extending 1-2 cm outside the window.

**Expected behaviour**
No tear.

**Actual behaviour**
Tear

**System**
System Information
------------------

Raspberry Pi 400 Rev 1.0\0
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage4

Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
Revision	: c03130
Serial		: 10000000dc6bd0c6
Model		: Raspberry Pi 400 Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Nov 18 2021 16:16:49 
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       14989480  4841748   9485856  34% /
devtmpfs         1800564        0   1800564   0% /dev
tmpfs            1965428    29004   1936424   2% /dev/shm
tmpfs             786172      988    785184   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    258095    49267    208828  20% /boot
tmpfs             393084       24    393060   1% /run/user/1000
/dev/sda2       14987608 11828180   2451176  83% /media/pi/rootfs
/dev/sda1         258095    49284    208812  20% /media/pi/boot

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20211123
raspberrypi-sys-mods:
  Installed: 20211005+bullseye
openbox:
  Installed: 3.6.1-9+rpt1+deb11u1
lxpanel:
  Installed: 0.10.1-2+rpt11
[... 696 lines cut ...]
[   14.346800] Bluetooth: RFCOMM ver 1.11
[   14.366124] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.399249] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   64.166695] usb 1-1.1: new high-speed USB device number 5 using xhci_hcd
[   64.301728] usb 1-1.1: New USB device found, idVendor=05e3, idProduct=0723, bcdDevice=94.51
[   64.301751] usb 1-1.1: New USB device strings: Mfr=3, Product=4, SerialNumber=2
[   64.301770] usb 1-1.1: Product: USB Storage
[   64.301788] usb 1-1.1: Manufacturer: Generic 
[   64.301805] usb 1-1.1: SerialNumber: 000000009451
[   64.305993] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[   64.306773] usb-storage 1-1.1:1.0: Quirks match for vid 05e3 pid 0723: 8000
[   64.307015] scsi host0: usb-storage 1-1.1:1.0
[   65.358424] scsi 0:0:0:0: Direct-Access     Generic  STORAGE DEVICE   9451 PQ: 0 ANSI: 0
[   65.396638] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   65.549379] sd 0:0:0:0: [sda] 31116288 512-byte logical blocks: (15.9 GB/14.8 GiB)
[   65.550200] sd 0:0:0:0: [sda] Write Protect is off
[   65.550207] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[   65.551045] sd 0:0:0:0: [sda] No Caching mode page found
[   65.551052] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   65.636752]  sda: sda1 sda2
[   65.645342] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   66.142394] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: errors=remount-ro
[  171.590456] v3d fec00000.v3d: MMU error from client CLE (4) at 0x7501000, pte invalid
[  613.673471] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[  613.676596] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19

audetto, 2022-01-03:
If you look carefully, just above the mouse pointer you can see a lighter blue on the left and then a dark line just below the status bar.

The lighter part IS the tear, the dark is normal. It will flicker when the window is moved and the it normally settles.

[image]

audetto, 2022-01-04:
And just to be clear, I have never seen it with Buster on the same hardware.

popcornmix, 2022-01-04:
This won't be a linux kernel issue (what this repo is about).
Can you create the issue [here](https://github.com/RPi-Distro/repo/issues) which may be seen by people who know more about desktop composition issues.

closed 2022-01-04 by popcornmix