raspberrypi/linux #4805: SPI CS always active high

nunojpg, 2022-01-04:
I notice that on rpi-5.16.y SPI CS is always active high. This appears to be the case after 5e31ba0c0543a04483b53151eb5b7413efece94c, as the this level control is suppose do be done at pinctrl, but it's not the case.

pelwell, 2022-01-04:
Always active-high? Have you got an example of something trying to use active-low and failing?

nunojpg, 2022-01-04:
**Let me state again this is rpi-5.16.y.**

I am using a display like this:

```
&spi {
	pinctrl-names = "default";
	pinctrl-0 = <&spi0_gpio7 &display_gpio1_31>;
	status = "okay";
	display: display@0 {
		compatible = "vot,v220hf01a-t", "ilitek,ili9225";
		reg = <0>;
		spi-max-frequency = <32000000>;
		rotation = <90>;
		rs-gpios = <&gpio 31 GPIO_ACTIVE_HIGH>;
		reset-gpios = <&gpio 1 GPIO_ACTIVE_HIGH>;
	};
};
```

To get it to work I need this minimum patch:

```
+++ b/drivers/spi/spi-bcm2835.c
@@ -1309,7 +1309,7 @@ static int bcm2835_spi_setup(struct spi_device *spi)
 
 	spi->cs_gpiod = gpiochip_request_own_desc(chip, 8 - spi->chip_select,
 						  DRV_NAME,
-						  GPIO_LOOKUP_FLAGS_DEFAULT,
+						  GPIO_ACTIVE_LOW | GPIO_PERSISTENT,
```

pelwell, 2022-01-05:
Before 5e31ba0c0543a04483b53151eb5b7413efece94c the spi-bcm2835 driver looked for an spi-cs-high on each client device, but now the system relies on the CS GPIOs being declared to be active-high (which makes far more sense). The DT declaration of spi0 includes `cs-gpios = <&gpio 8 1>, <&gpio 7 1>;`, showing that both CS lines are active-low.

So I can understand the failure, without the patch how do you think the kernel is supposed to know that the CS should be active-low? I don't see an spi-cs-high property in your DT.

nunojpg, 2022-01-05:
Ok, I was using mainline trees (bcm2837) which don't have proper CS nodes. Sorry and thanks

closed 2022-01-05 by nunojpg