raspberrypi/linux #4814: Pi4 vc4-kms-v3d for 2xHDMI + DPI doesn't work

6by9, 2022-01-07:
### Describe the bug

Note for myself / @mripard so I don't forget this.

Testing on Pi4 with HDMI-0 and DPI (HyperPixel4 800x480@60 using #4812) works fine. Plug in HDMI-1 and it fails with flip done timeouts.

2xHDMI + DSI1 (7" touch panel) works fine. DSI1 can be fed from any of the HVS channels though, whereas DPI is channel 0 / output 0 only.

### Steps to reproduce the behaviour

Build with 4812.
Add dtoverlay=vc4-kms-dpi-hyperpixel4.
Boot with just HDMI-0 connected and all OK.
Boot with both HDMIs connected and get no display.
Boot with just HDMI-0 and then connect HDMI-1, and HDMI-1 does not come up. At next update it will produce flip timeout.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Checked on both 5.10.90 and 5.15.15, 32bit kernels.

aBUGSworstnightmare-rpi, 2022-01-08:
Hi Dave, dual HDMI+DPI has never worked for me on Bullseye. Reported on 11th here https://forums.raspberrypi.com/viewtopic.php?p=1936459&hilit=Bullseye#p1936471

6by9, 2022-09-10:
Should be resolved now through #5121. I've certainly had dual DSI and single HDMI0 working.

There was insufficient COB assigned to HVS channel 2, and the interrupt masking was wrong for 2711 leading to the interrupt being disabled by the core.

closed 2022-09-10 by 6by9