raspberrypi/linux #4816: Kingston DC500M SSD unrecognized

ayo-x, 2022-01-09:
### Describe the bug

The Pi doesn't recognize this SSD. It sees the usb adapter(`lsusb`), but not the SSD itself. I've tested with a couple of adapters, both of which haven't given me any issues before. I connected the drive to an already booted Pi and ran `lsblk`. It only shows the SSD/SD card it booted from, even after rebooting the Pi.
I tried to use it as a boot drive firstly, but got this screen.
[image]

### Steps to reproduce the behaviour

Connect the SSD via usb adapter, with the intent to either boot from it or simply as storage extension.

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS: `Dietpi armv8`. Also tested with `Raspberry Pi OS Lite`

`vcgencmd version`
Nov 18 2021 16:17:04
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start_cd)

`uname -a`
5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux

`vcgencmd bootloader_version`
2021/12/02 11:08:03
version 78ec57468d6bb1dd7051698539ec814d22a98c64 (release)
timestamp 1638443283
update-time 1641718003
capabilities 0x0000007f

`vcgencmd bootloader_config`
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=1
BOOT_ORDER=0xf14

`/etc/default/rpi-eeprom-update`
FIRMWARE_RELEASE_STATUS="stable"

### Additional context

Output of `lsusb -vvv` when connected to an already booted Pi

```Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x174c ASMedia Technology Inc.
  idProduct          0x225c
  bcdDevice            1.00
  iManufacturer           2 Ugreen
  iProduct                3 Ugreen Storage Device
  iSerial                 1 20D11E80B118
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
[... 503 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

timg236, 2022-01-10:
Duplicate of https://github.com/raspberrypi/rpi-eeprom/issues/396

pelwell, 2022-01-10:
As @timg236 said in the rpi-eeprom repo:
> It's best to ask about hardware interop questions on the Raspberry Pi forums
https://forums.raspberrypi.com/viewforum.php?f=63

closed 2022-01-10 by pelwell