raspberrypi/linux #4817: Latest master 5.15 stops Kodi playback

A1RF0RCE1, 2022-01-09:
### Describe the bug

Inappropriate ioctl for device.
5 jan it was still working,.

### Steps to reproduce the behaviour

Build libreelec with master 5.15

### Device (s)

Raspberry Pi 4 Mod. B

### System

Libreelec

popcornmix, 2022-01-09:
I've just built LE Matrix for pI4 with latest 5.15 kernel.
```
2022-01-09 14:11:51.461 T:954      INFO <general>: Starting Kodi (19.3 (19.3.0) Git:19.3-Matrix). Platform: Linux ARM 32-bit
2022-01-09 14:11:51.462 T:954      INFO <general>: Using Release Kodi x32
2022-01-09 14:11:51.462 T:954      INFO <general>: Kodi compiled 2022-01-09 by GCC 10.2.0 for Linux ARM 32-bit version 5.15.13 (331533)
2022-01-09 14:11:51.462 T:954      INFO <general>: Running on BCM2835 with LibreELEC (popcornmix): devel-20220109140052-645764d 10.0, kernel: Linux ARM 64-bit version
 5.15.13
2022-01-09 14:11:51.462 T:954      INFO <general>: FFmpeg version/source: 4.3.2-Kodi
```
```
LibreELEC5:~ # uname -a
Linux LibreELEC5 5.15.13 #1 SMP Sun Jan 9 14:02:15 GMT 2022 aarch64 GNU/Linux
```
And also LE Maser
```
022-01-09 15:14:33.451 T:893      INFO <general>: Starting Kodi (20.0-ALPHA1 (19.90.101) Git:5ad708b518e6bb3e8ac31a9c7b5a8908573a409f). Platform: Linux ARM 32-bit
2022-01-09 15:14:33.451 T:893      INFO <general>: Using Release Kodi x32
2022-01-09 15:14:33.451 T:893      INFO <general>: Kodi compiled 2022-01-09 by GCC 10.3.0 for Linux ARM 32-bit version 5.15.13 (331533)
2022-01-09 15:14:33.451 T:893      INFO <general>: Running on BCM2835 with LibreELEC (popcornmix): devel-20220109142734-11a6c94 11.0, kernel: Linux ARM 64-bit version
 5.15.13
```
```
LibreELEC5:~ # uname -a
Linux LibreELEC5 5.15.13 #1 SMP Sun Jan 9 15:02:00 GMT 2022 aarch64 GNU/Linux
```

Both run fine and play videos (hevc, h264 and sw decoded).
Can you try a clean build and/or test the [url=https://test.libreelec.tv/]official nightly[/url].
If you still have problems, then report complete dmesg log, and kodi.log (assuming kodi launches).

A1RF0RCE1, 2022-01-09:
The official nightly works fine ,, 
Latest libreelec with latest master kodi (Nessus) with latest kernel 5.15 fails on playback
Latest libreelec with latest master kodi with kernel version of nightly(678d5d31f2723046c255038f79c39c7736655ea6 )works fine …

popcornmix, 2022-01-10:
> If you still have problems, then report complete dmsg log, and debug kodi.log (assuming kodi launches).

A1RF0RCE1, 2022-01-11:
> > If you still have problems, then report complete dmsg log, and debug kodi.log (assuming kodi launches).

Today I created a fresh build with the latest masters and no problems occur... weird :-)

A1RF0RCE1, 2022-01-11:
can't reproduce the error with the latest build.

closed 2022-01-11 by A1RF0RCE1